coolsnowwolf/lede #6045: 小米/红米AC2100的闭源驱动要如何才能跑到800M？

NagaseKouichi, 2020-12-17:
用大雕的源代码编译的固件，使用的闭源驱动，5G开启160MHZ，测速上传只能跑到不到500M，下载只有300M

我看大雕在telegram上说AC2100是可以跑到800M的

https://www.right.com.cn/forum/thread-4043514-1-1.html
用这个大佬的固件也确实可以跑到800M

那么自己编译的固件怎么才也能跑上800M呢，是需要什么额外的设置吗？

kzw200015, 2020-12-19:
似乎是现在的HWNAT不支持无线

NagaseKouichi, 2020-12-19:
> 似乎是现在的HWNAT不支持无线

那现在用lean的源码编译就没法跑到800M吗？L大说可以跑到800M也是基于237大佬的固件？

kzw200015, 2020-12-19:
> > 似乎是现在的HWNAT不支持无线
> 
> 那现在用lean的源码编译就没法跑到800M吗？L大说可以跑到800M也是基于237大佬的固件？

不太清楚，我用有线测速，CPU基本不动的，hwnat应该是起作用了，用无线跑100M，CPU 0占用已经30%+了，估计是没启用。5.4内核的HWNAT是低版本移植上来的，估计移植不完美吧。我也只是猜测，等个大佬解答

caoyipeng2008, 2020-12-21:
借问一下，ssr，我刷最近的固件，不管是passwall 还是ssrplus配置v2ray.节点，只要一开负载就很高。lz有没有这样的问题？

NagaseKouichi, 2020-12-22:
> 借问一下，ssr，我刷最近的固件，不管是passwall 还是ssrplus配置v2ray.节点，只要一开负载就很高。lz有没有这样的问题？

我主路由是X86软路由，没有这个问题。而且SSR用了套娃的方式，实际跑v2ray的是在PVE服务器上用lxc跑的openwrt。AC2100拿来刷机玩的

fishtorres, 2021-01-27:
所以这玩意根本跑不满 2*2 160mhz吗

closed 2021-06-20 by coolsnowwolf