coolsnowwolf/lede #6047: img镜像大小疑问

xiaocenter, 2020-12-18:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

 目前编译的X86版本，使用没有问题。有个疑问，使用大大源码编译的固件大小和lienol的不一样  同样的16+160默认配置 大大的img镜像是176.5MB 而lienol的img只有39MB  这不是固件出错了 都正常使用  听说是lienol的启用了镜像固实压缩  但是没找到设置的地方  这个很有用比如设置大容量分区  大大的固件很占空间   是否可以压缩固件？ 或者告知设置方法   万分感谢  

### (2) 路由器型号和固件版本

A：X86  最新版

### (3) 详细日志

A：无

EnnawYang, 2020-12-18:
```
Target Images  --->  [*] GZip images
```

xiaocenter, 2020-12-18:
> ```
> Target Images  --->  [*] GZip images
> ```
这个只是把 img镜像打包为压缩文件   而不是镜像本身压缩了

hyird, 2020-12-18:
lienol的 你没开填充 编译出来的 几十MB的img 写盘 overlay空间是几乎为0的 只能用来升级

xiaocenter, 2020-12-18:
> lienol的 你没开填充 编译出来的 几十MB的img 写盘 overlay空间是几乎为0的 只能用来升级

如何取消填充，不需要  求教

AmadeusGhost, 2020-12-18:
overlay 空间几乎为0的话容易丢配置

xiaocenter, 2020-12-19:
> overlay 空间几乎为0的话容易丢配置

取消填充咋弄

LoogCP, 2021-01-08:
> ```
> Target Images  --->  [*] GZip images
> ```

我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏

EnnawYang, 2021-01-08:
> > ```
> > Target Images  --->  [*] GZip images
> > ```
> 
> 我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏

没有问题 一直在用

LoogCP, 2021-01-08:
> > > ```
> > > Target Images  --->  [*] GZip images
> > > ```
> > 
> > 
> > 我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏
> 
> 没有问题 一直在用

你是解压后刷的还是直接刷的

EnnawYang, 2021-01-08:
> > > > ```
> > > > Target Images  --->  [*] GZip images
> > > > ```
> > > 
> > > 
> > > 我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏
> > 
> > 
> > 没有问题 一直在用
> 
> 你是解压后刷的还是直接刷的

不解压 直接用，解压也没问题

LoogCP, 2021-01-08:
> > > > > ```
> > > > > Target Images  --->  [*] GZip images
> > > > > ```
> > > > 
> > > > 
> > > > 我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏
> > > 
> > > 
> > > 没有问题 一直在用
> > 
> > 
> > 你是解压后刷的还是直接刷的
> 
> 不解压 直接用，解压也没问题

我啥时候有空再试试

closed 2021-06-20 by coolsnowwolf

drw2005, 2022-02-24:
> lienol的 你没开填充 编译出来的 几十MB的img 写盘 overlay空间是几乎为0的 只能用来升级

请教，这个如何操作？

Jefffish09, 2022-03-24:
> > > > > ```
> 
> > > > > Target Images  --->  [*] GZip images
> 
> > > > > ```
> 
> > > > 
> 
> > > > 
> 
> > > > 我记得lean大佬的GZip好像有问题，编译出来的不能用，提示损坏
> 
> > > 
> 
> > > 
> 
> > > 没有问题 一直在用
> 
> > 
> 
> > 你是解压后刷的还是直接刷的
> 
> 
> 
> 不解压 直接用，解压也没问题

想请教下，我编译出来.img.gz，在win下用7zip解压失败，会提示“显示有效数据外包含额外数据”。
在ubuntu用gizp命令解压，能解压出img文件，但会提示“decompression OK, trailing garbage ignored”，这样正常吗？能不能用这个img来刷机？怕出问题