coolsnowwolf/lede #6056: k2p 硬改32 刷12月5日-12月18日固件，重启无法启动，长亮红灯。

ltactor, 2020-12-19:
## 1.关于你要提交的问题
k2p 硬改32 刷11月15日-12月18日固件，重启无法启动。
11月15日编译固件可以正常启动，从12月5日后不断编译固件全部重启后长亮红灯。

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
选择k2p 硬改32M  默认配置+  v2ray+   wiregurad。
编译未报错。
breed 选择斐讯专区刷入。
第一次可以启动，恢复配置信息重启后，长亮红灯。

11月15日编译固件可以正常启动，从12月5日后不断编译固件全部重启后长亮红灯。
### (2) 路由器型号和固件版本

A：k2p 硬改32M   12月5日-12月18日固件全部编译测试不成功。

### (3) 详细日志

A：无法获取。

AmadeusGhost, 2020-12-19:
先 `make clean` 试试，不行就只能用 ttl 获取启动日志了，没日志解决不了问题的

ltactor, 2020-12-19:
> 先 `make clean` 试试，不行就只能用 ttl 获取启动日志了，没日志解决不了问题的

感谢回复。
我都是进行新的clone。  ttl不方便。

qjiang888, 2020-12-19:
k2p 我测试了好几台，新版源码是有问题，不断的插拔电源才有几率启动。就是一个插件也不选 也启动不了. breed升级最新也不行

hcym, 2020-12-19:
老毛子做ap完事了，折腾这个能登陆月饼

SilverHunterLib, 2020-12-20:
我的K2P原版，没有硬改，刷的12.19最新源码编译的固件，刷完后无法连接路由器，电脑无法分配到IP，无线也搜索不到，跟之前12.13源码编译的固件问题一样，应该是一直没解决。现在刷回了11.15号编译的固件，暂时没问题，等等看看有没有大佬解决了

AmadeusGhost, 2020-12-20:
试试这个 commit 能不能解决问题（需要 make clean）：https://github.com/AmadeusGhost/lede/commit/eff7c560e7cec8c5ef4ba592ac4227d75fe83f48

ltactor, 2020-12-21:
> 试试这个 commit 能不能解决问题（需要 make clean）：[AmadeusGhost@eff7c56](https://github.com/AmadeusGhost/lede/commit/eff7c560e7cec8c5ef4ba592ac4227d75fe83f48)

感谢回复，今晚我试试。

SilverHunterLib, 2020-12-22:
> > 试试这个 commit 能不能解决问题（需要 make clean）：[AmadeusGhost@eff7c56](https://github.com/AmadeusGhost/lede/commit/eff7c560e7cec8c5ef4ba592ac4227d75fe83f48)
> 
> 感谢回复，今晚我试试。
> 已经编译成功，时间原因，只能今晚刷机测试。

请问大佬试过以后正常吗？我的K2P最近固件都用不了，重启后无法连接路由器

ltactor, 2020-12-22:
> 试试这个 commit 能不能解决问题（需要 make clean）：[AmadeusGhost@eff7c56](https://github.com/AmadeusGhost/lede/commit/eff7c560e7cec8c5ef4ba592ac4227d75fe83f48)

 感谢回复，编译测试，可以重启，wifi无法启用。     别的暂时正常。

PR #6085 (2020-12-23): ramips: fix k2p 32m reboot problems
fix #6031 , fix #6056 , fix #6077 .
closed 2020-12-23 by coolsnowwolf