coolsnowwolf/lede #6059: youku L1 只能拔电源重启，不能软重启

qinfeng8848, 2020-12-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：youku L1 只能拔电源重启，不能软重启

### (2) 路由器型号和固件版本

A：youku L1  OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：

qjiang888, 2020-12-19:
k2p 也是一样的！！

yulinsoft, 2020-12-21:
5.4的内核没有添加重启补丁，建议编译4.19的先用着，等大神添加补丁。

AmadeusGhost, 2020-12-23:
重启补丁只有 4.14 内核需要

yulinsoft, 2020-12-23:
youku l1的软重启问题能解决吗？

AmadeusGhost, 2020-12-23:
> youku l1的软重启问题能解决吗？

有 ttl 日志吗？

DanielChen008, 2020-12-23:
> 5.4的内核没有添加重启补丁，建议编译4.19的先用着，等大神添加补丁。

您好，请教编译时如何选择内核版本？

yulinsoft, 2020-12-24:
> > 5.4的内核没有添加重启补丁，建议编译4.19的先用着，等大神添加补丁。
> 
> 您好，请教编译时如何选择内核版本？

lede/target/linux/ramips/Makefile中的5.4都改成4.14就行了，我就是这样弄的，可以用。

chenliangxin, 2021-03-06:
> > > 5.4的内核没有添加重启补丁，建议编译4.19的先用着，等大神添加补丁。
> > 
> > 
> > 您好，请教编译时如何选择内核版本？
> 
> lede/target/linux/ramips/Makefile中的5.4都改成4.14就行了，我就是这样弄的，可以用。

我已经改了，但选择4.14内核时编译一直报错。。

lon91ong, 2021-04-17:
> > > > 5.4的内核没有添加重启补丁，建议编译4.19的先用着，等大神添加补丁。
> > > 
> > > 
> > > 您好，请教编译时如何选择内核版本？
> > 
> > 
> > lede/target/linux/ramips/Makefile中的5.4都改成4.14就行了，我就是这样弄的，可以用。
> 
> 我已经改了，但选择4.14内核时编译一直报错。。

换了[lede-17.01分支](https://github.com/coolsnowwolf/openwrt)编译就好了

我用GitHub Acitons编译了几次, 最接近成功的一次, 刷入后, 拔电源重启, 成功启动但是wifi丢失, 其它几次都是卡在启动阶段, 反复重启
包括刷入[别人编译的OP系统](https://www.right.com.cn/forum/thread-4050369-1-1.html)也是反复自动重启
刷[padavan](https://breed.hackpascal.net/)正常, 但是也需要拔电源重启, breed自动重启无效

lon91ong, 2021-04-18:
breed的锅, 最后刷入了[**1169**](https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=4057201&pid=10551353)版的可以软重启了

closed 2021-06-20 by coolsnowwolf