coolsnowwolf/lede #6060: 小米R3G v1 - 编译出的 sysupgrade 是坏的

kmahyyg, 2020-12-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：保持默认设置，增加了 luci-app-openclash 到 package/lean 下，选中，编译成功。产生的 kernel.bin 可以从breed刷入，刷入后发现软件包列表页面存储可用为0b，尝试从breed刷入 sysupgrade 提示无法识别的固件类型，尝试从openwrt luci刷入sysupgrade，软砖。

### (2) 路由器型号和固件版本

A：小米R3G v1，最新开发版固件后自己刷了breed，目前等待刷入openwrt

### (3) 详细日志

A：无法确认相关日志所在位置，故无法提供。

kmahyyg, 2020-12-20:
[build_config_20201219_r3g.txt](https://github.com/coolsnowwolf/lede/files/5720393/build_config_20201219_r3g.txt)

Build Config 见上。

jxph1123, 2020-12-20:
我R3G是这样刷的，breed先备份之前的。要刷新的恢复进去后
mtd write /tmp/openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-kernel1.bin kernel0
mtd write /tmp/openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-rootfs0.bin rootfs0
这样刷的。
不知道V1是不是也是这样？

kmahyyg, 2020-12-20:
> 我R3G是这样刷的，breed先备份之前的。要刷新的恢复进去后
> mtd write /tmp/openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-kernel1.bin kernel0
> mtd write /tmp/openwrt-ramips-mt7621-xiaomi_mir3g-squashfs-rootfs0.bin rootfs0
> 这样刷的。
> 不知道V1是不是也是这样？

Sorry, It's not a correct way.

I'm so sorry that I left this issue opened after my problem solved.

Firstly, this "trouble" is a feature, not a bug. It's an intended behaviour. The man who added R3G support to OpenWRT just doesn't care about the bootloader-specific behaviour and released the support of this model. So you have to flash the initramfs one first. The sysupgrade one is NOT the flash layout we normally use, it's a u-boot based firmware, not for breed.

Secondly, According to [hackpascal][2], the man who built Breed(A bootloader so called never-die bootloader), XiaoMI just built a kernel0 for recovery purpose, kernel0 will be used for recovery from udisk when u're in a big trouble. So you flashed the kernel1 by default. But the default behaviour of breed on this model only boots kernel0, which lead to failure. [Original thread][1] can be found by Google in site right.com.cn.

@jxph1123  so bro, please do not flash third-party firmware to kernel0, which might save your router a life. I currently just hope you have a backup. And thanks to XiaoMI for this fail-safe switch.

The correct operation to do here is:

- (Optional, but might be helpful if the second step is not working)From "XiaoMI R3G Settings" in breed, remove the content of `normal_firmware_md5`, but do not remove the whole line.
- From `Environment Variables` page, Set a envvar: `xiaomi.r3g.bootfw`=`2`, then all things go fine. Works correctly now.

[1]:https://www.right.com.cn/forum/thread-987254-1-1.html
[2]:https://www.right.com.cn/forum/forum.php?mod=redirect&goto=findpost&ptid=267455&pid=2314800

closed 2020-12-20 by kmahyyg