coolsnowwolf/lede #6061: 【BUG】MT7620设备 无线网卡同时配置Client+AP，信道配置为auto时，AP无法正常初始化

NickLiu635, 2020-12-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
MT7620设备 无线网卡同时配置Client+AP，信道配置为auto时，AP无法正常初始化
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
MT7620设备 无线网卡同时配置Client+AP，信道配置为auto时，AP无法正常初始化

信道配置为Client的信道时，AP可以正常初始化
推测原因为
网卡在搜索上级AP信道时，自身AP初始化时没有正确的信道导致初始化失败，等到
搜索并连接上上级AP时，自身AP又不会重新初始化

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：路由器为斐讯E1（就是那个信号中继器） 固件选择的PSG1208，
固件版本为 3ed02932ae80 (git hash)(本issue提交前一个小时左右pull并编译的)

### (3) 详细日志

A：

```
Sun Dec 20 11:44:20 2020 kern.notice kernel: [    0.000000] Linux version 5.4.83 (yuzu@yuzu-VirtualBox) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2970-c06f6662d)) #0 Sun Dec 20 01:55:22 2020
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Board has DDR2
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Analog PMU set to hw control
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Digital PMU set to hw control
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] SoC Type: MediaTek MT7620A ver:2 eco:6
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] printk: bootconsole [early0] enabled
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] CPU0 revision is: 00019650 (MIPS 24KEc)
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] MIPS: machine is Phicomm PSG1208
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Initrd not found or empty - disabling initrd
Sun Dec 20 11:44:20 2020 kern.warn kernel: [    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
Sun Dec 20 11:44:20 2020 kern.warn kernel: [    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Zone ranges:
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000]   Normal   [mem 0x0000000000000000-0x0000000003ffffff]
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Movable zone start for each node
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Early memory node ranges
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000]   node   0: [mem 0x0000000000000000-0x0000000003ffffff]
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000003ffffff]
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000] On node 0 totalpages: 16384
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000]   Normal zone: 144 pages used for memmap
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000]   Normal zone: 0 pages reserved
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000]   Normal zone: 16384 pages, LIFO batch:3
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
Sun Dec 20 11:44:20 2020 kern.debug kernel: [    0.000000] pcpu-alloc: [0] 0
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 16240
Sun Dec 20 11:44:20 2020 kern.notice kernel: [    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes, linear)
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes, linear)
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Writing ErrCtl register=00000c02
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Readback ErrCtl register=00000c02
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] Memory: 57116K/65536K available (4975K kernel code, 205K rwdata, 1096K rodata, 1204K init, 209K bss, 8420K reserved, 0K cma-reserved)
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] NR_IRQS: 256
Sun Dec 20 11:44:20 2020 kern.notice kernel: [    0.000000] random: get_random_bytes called from start_kernel+0x32c/0x518 with crng_init=0
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] CPU Clock: 580MHz
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] clocksource: systick: mask: 0xffff max_cycles: 0xffff, max_idle_ns: 583261500 ns
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] systick: enable autosleep mode
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] systick: running - mult: 214748, shift: 32
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 6590553264 ns
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.000010] sched_clock: 32 bits at 290MHz, resolution 3ns, wraps every 7405115902ns
Sun Dec 20 11:44:20 2020 kern.info kernel: [    0.015472] Calibrating delay loop... 385.02 BogoMIPS (lpj=770048)
[... 342 lines cut ...]
Sun Dec 20 11:45:24 2020 daemon.err hostapd: Interface initialization failed
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: wlan0-1: interface state ACS->DISABLED
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: wlan0-1: AP-DISABLED
Sun Dec 20 11:45:24 2020 daemon.err hostapd: ACS: Failed to start
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: wlan0-1: AP-DISABLED
Sun Dec 20 11:45:24 2020 daemon.err hostapd: hostapd_free_hapd_data: Interface wlan0-1 wasn't started
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: nl80211: deinit ifname=wlan0-1 disabled_11b_rates=0
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: nl80211: Failed to remove interface wlan0-1 from bridge br-lan: Invalid argument
Sun Dec 20 11:45:24 2020 daemon.notice hostapd: wlan0-1: interface state DISABLED->DISABLED
Sun Dec 20 11:45:26 2020 daemon.notice wpa_supplicant[2823]: wlan0: SME: Trying to authenticate with b4:0f:3b:49:eb:bc (SSID='404' freq=5200 MHz)
Sun Dec 20 11:45:26 2020 kern.info kernel: [   64.797876] wlan0: authenticate with b4:0f:3b:49:eb:bc
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.850540] wlan0: send auth to b4:0f:3b:49:eb:bc (try 1/3)
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.863885] wlan0: authenticated
Sun Dec 20 11:45:27 2020 daemon.notice wpa_supplicant[2823]: wlan0: Trying to associate with b4:0f:3b:49:eb:bc (SSID='404' freq=5200 MHz)
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.872512] wlan0: associate with b4:0f:3b:49:eb:bc (try 1/3)
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.897016] wlan0: associate with b4:0f:3b:49:eb:bc (try 2/3)
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.921044] wlan0: associate with b4:0f:3b:49:eb:bc (try 3/3)
Sun Dec 20 11:45:27 2020 kern.info kernel: [   64.945653] wlan0: association with b4:0f:3b:49:eb:bc timed out
Sun Dec 20 11:45:44 2020 daemon.info hostapd: wlan1: STA 60:f6:77:c5:81:e5 IEEE 802.11: authenticated
Sun Dec 20 11:45:44 2020 daemon.info hostapd: wlan1: STA 60:f6:77:c5:81:e5 IEEE 802.11: associated (aid 1)
Sun Dec 20 11:45:44 2020 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 60:f6:77:c5:81:e5
Sun Dec 20 11:45:44 2020 daemon.info hostapd: wlan1: STA 60:f6:77:c5:81:e5 RADIUS: starting accounting session A7C65D12BD2E717C
Sun Dec 20 11:45:44 2020 daemon.info hostapd: wlan1: STA 60:f6:77:c5:81:e5 WPA: pairwise key handshake completed (RSN)

```

yulinsoft, 2020-12-22:
7620ap和client好像只能用同一个信道，只能使用中继路由上级路由的信道。

NickLiu635, 2020-12-22:
> 7620ap和client好像只能用同一个信道，只能使用中继路由上级路由的信道。

因为你不知道上级路由会不会切换信道，所以client不能配置一个准确的信道，只能自动搜索上级ap信道，然后自身ap也用这个信道

closed 2021-06-20 by coolsnowwolf