coolsnowwolf/lede #6063: 关于Lan接口DHCP 配置咨询

Guxing0321, 2020-12-20:
Hi Team,

最近有个需求，需要设置23位掩码，我配置了下lan接口，目前是没有报错，需要帮我确认下，我的配置，谢谢
[image]

BR
Robin

Dimitri2020007, 2020-12-20:
配置这么多地址空间用来干嘛？

192.168.3.0/23

起始ip是192.168.3.1

广播ip是192.168.4.255

客户端ip范围：192.168.3.2~192.168.4.254

可以设置510个客户端的ip

Guxing0321, 2020-12-20:
Hi  Dimitri，

我按照上面的配置，把广播修改成 192.168.4.255
但是，我手机连上的时候，还是获取了192.168.2.73
不是，我设置的那个3网段
[image]

这个是为何，我哪边配置有问题？

Dimitri2020007, 2020-12-20:
ip分配是从最低位置开始的

你子网掩码已经被设置成了vlsm/cidr（可变长度/超网聚合）

如果设置成192.168.3.0/24的长度，最低位置就是3

Guxing0321, 2020-12-21:
好的，谢谢

closed 2020-12-21 by Guxing0321