coolsnowwolf/lede #6064: IPQ40xx 超频到 896MHz 稳定性调查反馈

coolsnowwolf, 2020-12-20:
新引入了一个超频的补丁，针对 IPQ40xx 的架构

[image]

运行了2天还是挺稳定的，跑分也超过了 1万分，XXR Plus 性能居然也同步增加了 25% ，xray 现在超频arch17 也能 180Mbps+ 了

大家使用该补丁感觉如何？ 欢迎反馈

luckyyyyy, 2020-12-20:
手上有一台精斗云，丢掉wifi的话就可以刷和这个了。

star-o0O, 2020-12-20:
用了一天，感觉还是挺稳的，只是状态页的两个温度不显示了，也不知道发生了什么，刚刷上的时候还是有的。

yinzhefuyun, 2020-12-20:
> 用了一天，感觉还是挺稳的，只是状态页的两个温度不显示了，也不知道发生了什么，刚刷上的时候还是有的。

2.4Gwifi和5Gwifi打开一个显示一个温度

gw826943555, 2020-12-20:
> 用了一天，感觉还是挺稳的，只是状态页的两个温度不显示了，也不知道发生了什么，刚刷上的时候还是有的。

无线打开了吗 无线打开才会显示温度

hcym, 2020-12-21:
ac58试试看看，应该可以吧？回头反馈信息

mcucore, 2020-12-21:
建议别超频了，体质有差异，路由设备稳定第一！需求更高换硬件！

NagaseKouichi, 2020-12-21:
IPQ40xx有硬件NAT加速么

hcym, 2020-12-21:
温度高，刷原厂固件做ap了😂

mcucore, 2020-12-21:
> 温度高，刷原厂固件做ap了😂

为啥原厂，雕多啦不是挺稳定！

Cathgao, 2020-12-21:
用了两天，目前没发现问题，也没觉得有什么体验改进，因为本来性能就够用了

zhang5389, 2020-12-23:
用了一天，功能正常，温度稳定。

carrneverrun, 2020-12-23:
> 手上有一台精斗云，丢掉wifi的话就可以刷和这个了。

筋斗云刷这个系统会使WiFi无效么?

yinzhefuyun, 2020-12-23:
> > 手上有一台精斗云，丢掉wifi的话就可以刷和这个了。
> 
> 筋斗云刷这个系统会使WiFi无效么?

并不会，他只是不想用无线，所以去掉了WiFi（关闭）

xfmly, 2020-12-25:
4.14.202，竞斗云43.0°C 44.0°C
5.4，星际宝盒79.0°C 80.0°C
稳定使用

hzjnet, 2020-12-25:
> 4.14.202，竞斗云43.0°C 44.0°C
> 5.4，星际宝盒79.0°C 80.0°C
> 稳定使用

补充下吧：
主机型号 | ASUS RT-ACRH17 (CpuMark : 9704.174311 Scores)
架构        | ARMv7 Processor rev 5 (v7l) x 4 (896MHz, 59.0°C 59.0°C)
内核版本 | 5.4.79

不知道4.4版本如何弄？要是哪位出个补丁就好了

au3300, 2020-12-25:
正常使用，稳定！！！
[image]

xfmly, 2020-12-25:
> > 4.14.202，竞斗云43.0°C 44.0°C
> > 5.4，星际宝盒79.0°C 80.0°C
> > 稳定使用
> 
> 补充下吧：
> 主机型号 | ASUS RT-ACRH17 (CpuMark : 9704.174311 Scores)
> 架构 | ARMv7 Processor rev 5 (v7l) x 4 (896MHz, 59.0°C 59.0°C)
> 内核版本 | 5.4.79
> 
> 不知道4.4版本如何弄？要是哪位出个补丁就好了

P&W R619AC 128M (CpuMark : 10225.929122 Scores)
ARMv7 Processor rev 5 (v7l) x 4 (896MHz, 45.0°C 44.0°C)

coolsnowwolf, 2020-12-25:
看来还是竞斗云散热好啊

asushugo, 2020-12-25:
> 看来还是竞斗云散热好啊

[image]
我适配的这个 AP 也很稳定哦

ZGNCZMY, 2020-12-26:
使用一个星期，稳如狗~

ZGNCZMY, 2020-12-26:
忘了说设备acrh17

kilo1220, 2020-12-31:
该功能如何启用？为啥我的acrh17还是显示717MHz

asushugo, 2021-01-01:
> 该功能如何启用？为啥我的acrh17还是显示717MHz

[image]

ZGNCZMY, 2021-01-01:
> 
> 
> 该功能如何启用？为啥我的acrh17还是显示717MHz

有老兄给你截图了

kilo1220, 2021-01-01:
> > 该功能如何启用？为啥我的acrh17还是显示717MHz
> 
> [image]

感谢~~

kilo1220, 2021-01-01:
> > 该功能如何启用？为啥我的acrh17还是显示717MHz
> 
> 有老兄给你截图了

感谢~~

moqingzhu1, 2021-01-02:
现在编译怎么是4.19版，是我编译的问题还是代码问题，同步的大神源码，新手不懂就问，勿喷

OPA2604, 2021-01-08:
[image]

现在室内16度。

OPA2604, 2021-01-08:
[image]

lenghun, 2021-01-16:
别折腾超频了,路由稳定优先,如果有需要多这点性能的情况说明是很高负载超频了反而会出各种问题,退一万步讲你真的需要那进breed超不好吗,

AmadeusGhost, 2021-01-17:
> 别折腾超频了,路由稳定优先,如果有需要多这点性能的情况说明是很高负载超频了反而会出各种问题,退一万步讲你真的需要那进breed超不好吗,

arm 平台的 CPU 工作频率是可以手动更改的，本身固件也是默认频率启动的，稳得一批。再说了 arm 平台哪来的 breed ，，

yinzhefuyun, 2021-01-18:
> 别折腾超频了,路由稳定优先,如果有需要多这点性能的情况说明是很高负载超频了反而会出各种问题,退一万步讲你真的需要那进breed超不好吗,

默认启动是默认频率，你要进系统，在cpu控制那里再调高频率才可以

Marcotso, 2021-01-21:
> 看来还是竞斗云散热好啊

老大,铁盒不是盖的

467815891a, 2021-01-24:
[image]
星际宝盒散热真的捉鸡，散热片换了还这样。

sgsgsg520, 2021-05-08:
能不能超个1.2 g

Cathgao, 2021-05-08:
[image]
空调房，铁盒子摸起来不温，温度没导到外壳上

closed 2021-06-20 by coolsnowwolf