coolsnowwolf/lede #6083: 酸酸乳启用没多久分流就失效，国内变成走代理

hyyz17200, 2020-12-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
[x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译今日的新版以后升级上去，刚应用时，GFW和IP模式分流都正常。但过上一两分钟就会变成国内地址全部走代理。例如GFW模式时会变成这样：

[image]

### (2) 路由器型号和固件版本

A：x86_64 
OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：

系统日志里有大量dnsmasq的信息，不知道是否有关

Wed Dec 23 03:20:15 2020 cron.err crond[5032]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:20:16 2020 daemon.info dnsmasq[5395]: Connected to system UBus
Wed Dec 23 03:21:03 2020 daemon.info dnsmasq[5697]: Connected to system UBus
Wed Dec 23 03:21:03 2020 cron.err crond[5732]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:21:05 2020 daemon.info dnsmasq[6095]: Connected to system UBus
Wed Dec 23 03:22:00 2020 daemon.info dnsmasq[6464]: Connected to system UBus
Wed Dec 23 03:22:40 2020 daemon.info dnsmasq[6702]: Connected to system UBus
Wed Dec 23 03:22:40 2020 cron.err crond[6737]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:22:41 2020 daemon.info dnsmasq[7100]: Connected to system UBus
Wed Dec 23 03:26:05 2020 daemon.info dnsmasq[7904]: Connected to system UBus
Wed Dec 23 03:26:05 2020 cron.err crond[7939]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:26:07 2020 daemon.info dnsmasq[8302]: Connected to system UBus
Wed Dec 23 03:26:28 2020 daemon.info dnsmasq[8548]: Connected to system UBus
Wed Dec 23 03:26:28 2020 cron.err crond[8583]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:26:29 2020 daemon.info dnsmasq[8946]: Connected to system UBus
Wed Dec 23 03:26:47 2020 daemon.info dnsmasq[9192]: Connected to system UBus
Wed Dec 23 03:26:47 2020 cron.err crond[9227]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:26:48 2020 daemon.info dnsmasq[9591]: Connected to system UBus
Wed Dec 23 03:27:35 2020 daemon.info dnsmasq[9852]: Connected to system UBus
Wed Dec 23 03:27:35 2020 cron.err crond[9887]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:27:36 2020 daemon.info dnsmasq[10249]: Connected to system UBus
Wed Dec 23 03:28:26 2020 daemon.info dnsmasq[10559]: Connected to system UBus
Wed Dec 23 03:28:26 2020 cron.err crond[10594]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:28:28 2020 daemon.info dnsmasq[10956]: Connected to system UBus
Wed Dec 23 03:28:33 2020 daemon.info dnsmasq[11182]: Connected to system UBus
Wed Dec 23 03:28:33 2020 cron.err crond[11217]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:28:34 2020 daemon.info dnsmasq[11579]: Connected to system UBus
Wed Dec 23 03:37:12 2020 daemon.info dnsmasq[14036]: Connected to system UBus
Wed Dec 23 03:37:12 2020 cron.err crond[14071]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:37:14 2020 daemon.info dnsmasq[14433]: Connected to system UBus
Wed Dec 23 03:37:19 2020 daemon.info dnsmasq[14649]: Connected to system UBus
Wed Dec 23 03:37:19 2020 cron.err crond[14684]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:37:21 2020 daemon.info dnsmasq[15046]: Connected to system UBus
Wed Dec 23 03:37:34 2020 daemon.info dnsmasq[15318]: Connected to system UBus
Wed Dec 23 03:37:34 2020 cron.err crond[15353]: crond (busybox 1.31.1) started, log level 8
Wed Dec 23 03:37:35 2020 daemon.info dnsmasq[15715]: Connected to system UBus
Wed Dec 23 03:37:40 2020 daemon.info dnsmasq[15979]: Connected to system UBus
Wed Dec 23 03:37:40 2020 cron.err crond[16014]: crond (busybox 1.31.1) started, log level 8

hyyz17200, 2020-12-22:
测试了一下，保留配置降级，退回旧版本
OpenWrt R20.9.15 / LuCI Master (git-20.256.12360-1a54222)
一切正常

Dimitri2020007, 2020-12-26:
ssr+启用gfw模式是不是一开始用都解析失败啊，要等很长时间才反应过来?

hyyz17200, 2020-12-26:
> ssr+启用gfw模式是不是一开始用都解析失败啊，要等很长时间才反应过来?

老固件都正常的

hyyz17200, 2021-01-20:
三天前试了编译最新的源码，症状一样，最后还是退回去用了。

closed 2021-06-20 by coolsnowwolf