coolsnowwolf/lede #6088: 最新源码编译错误，softether

badgv, 2020-12-23:
[image]

make[3]: Entering directory '/compile/lede/feeds/packages/net/softethervpn'
. /compile/lede/include/shell.sh; gzip -dc /compile/lede/dl/softether-src-v4.34-9745-rtm.tar.gz | tar -C /compile/lede/build_dir/hostpkg/softethervpn/v4.34-9745/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/* /compile/lede/build_dir/hostpkg/softethervpn/v4.34-9745

Applying ./patches/103-noeucjp.patch using plaintext: 
patching file src/Mayaqua/Internat.c

Applying ./patches/110-no-m64.patch using plaintext: 
patching file src/makefiles/linux_64bit.mak

Applying ./patches/120-openssl-deprecated.patch using plaintext: 
patching file src/Mayaqua/Encrypt.c
Hunk #3 succeeded at 629 (offset 2 lines).
Hunk #4 succeeded at 928 (offset 2 lines).
Hunk #5 succeeded at 942 (offset 2 lines).
Hunk #6 succeeded at 960 (offset 2 lines).
Hunk #7 succeeded at 979 (offset 2 lines).
Hunk #8 succeeded at 1911 (offset 2 lines).
Hunk #9 succeeded at 2053 (offset 2 lines).
Hunk #15 FAILED at 4177.
1 out of 15 hunks FAILED -- saving rejects to file src/Mayaqua/Encrypt.c.rej
Patch failed!  Please fix ./patches/120-openssl-deprecated.patch!
make[3]: *** [Makefile:204: /compile/lede/build_dir/hostpkg/softethervpn/v4.34-9745/.preparede68aeaae44c2a4e009134cf85f492fb5_6664517399ebbbc92a37c5bb081b5c53] Error 1
make[3]: Leaving directory '/compile/lede/feeds/packages/net/softethervpn'
time: package/feeds/packages/softethervpn/host-compile#1.37#0.67#1.71
make[2]: *** [package/Makefile:111: package/feeds/packages/softethervpn/host/compile] Error 2
make[2]: Leaving directory '/compile/lede'
make[1]: *** [package/Makefile:107: /compile/lede/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/compile/lede'
make: *** [/compile/lede/include/toplevel.mk:224: world] Error 2

bennyzhou88, 2020-12-27:
v4.34 base这个包有问题。menuconfig换v5的softether包完美跳过

badgv, 2020-12-28:
> v4.34 base这个包有问题。menuconfig换v5的softether包完美跳过

V5的softether，SSTP有问题，还有一些其它的问题，所以我一直用的4.X版本的

sxf5487, 2021-01-26:
我用softether的SSTP，如果是假设我同一路由器下有两个设备，拨号到softether服务器，这两个设备分配到的ip是同一个（设备名使用当前外网ip第一位命名），导致先拨号的设备无法连接。你有这个情况吗？

sxf5487, 2021-01-26:
> [image]
>
> make[3]: Entering directory '/compile/lede/feeds/packages/net/softethervpn'
> . /compile/lede/include/shell.sh; gzip -dc /compile/lede/dl/softether-src-v4.34-9745-rtm.tar.gz | tar -C [...]

我用softether的SSTP，如果是假设我同一路由器下有两个设备，拨号到softether服务器，这两个设备分配到的ip是同一个（设备名使用当前外网ip第一位命名），导致先拨号的设备无法连接。你有这个情况吗？

closed 2021-06-20 by coolsnowwolf