coolsnowwolf/lede #6089: CPU软中断不平衡问题导致CPU瞬间负载过高，造成在大流量下eth网口drop包和overrun,表现为ping>1000ms

galaxyskyknight, 2020-12-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：CPU软中断不平衡问题导致CPU瞬间负载过高，造成在大流量下eth网口drop包和overrun,表现为ping>1000ms
参考文章：
https://www.jianshu.com/p/461d660f2ddc

### (2) 路由器型号和固件版本

A：见图
[image]

### (3) 详细日志

A：
ping突发高延时图（晚间延时可达2000-3000ms), eth4上挂了两个玩客云，上下载峰值可达80-90M，但average load并不大（<15%),系统AL见下图
[image]
[image]

[image]
如上图eth0是我的wan口（双拨）eth1-eth5为br-lan, 可以看到中断基本集中在cpu0, 从netdata监控看，每次有大量softirq来的时候上述的ping高延时便会发生，各个网口的io便会断流。上面的ping截图是（有线千兆直连）
[image]
如上图，可以看到有eth0和eth4有大量的drop和overrun.

请看固件里是否可以按参考文章的配置一下中断的core之间的平衡？

neavo, 2020-12-23:
我有时候在使用百度云下载时也有类似问题，不知道是不是同一个原因

ZqinKing, 2020-12-23:
你这个应该不是你说的问题，更有可能是负载太大，导致的软中断切换导致的 ping延时高。可以把中断动态调整直接关闭。蒋eth4网卡的软中断绑定cpu1,其余网口的软中断绑在cpu0上。

galaxyskyknight, 2020-12-23:
> 你这个应该不是你说的问题，更有可能是负载太大，导致的软中断切换导致的 ping延时高。可以把中断动态调整直接关闭。将eth4网卡的软中断绑定cpu1,其余网口的软中断绑在cpu0上。软中断切换会导致内核的cache miss。

多谢，我听上去你分析的有道理，具体怎么操作哈？能给个命令指引不？

galaxyskyknight, 2020-12-23:
刚试了几个方法，
1. 仓库下载irqbalance, 无论启动不启动，均无效
2. 关闭irqbalance,把eth4/eth0的rx_0绑定到cpu1,affanity 设置为cpu1 无效
3. 设置rps_flow_cnt 为16384 （2个队列，rps_sock_flow_entries设置为32768） 无效
以上配了后均无效，方法2，又看到softirq上往cpu1上转移，但丢包drop统计和瞬间ping高的问题依旧
感觉找错方向了，请大牛救命....

galaxyskyknight, 2020-12-24:
问题看上去暂时解决了，经过逐个排查启动项里的进程，最终锁定wpad这个无线认证程序有毒，好像对网卡驱动有冲突还是怎么着，总之禁用后，目前观察好像恢复正常了，请参考。

643406834, 2020-12-27:
> > 你这个应该不是你说的问题，可能可能是负载太大，导致软中断切换导致的ping连续高。可以把中断动态调整直接关闭。将eth4网卡的软中断绑定cpu1，其余网口的软中断绑在cpu0上。软中断切换会导致内核的缓存未命中。
> 
> 多谢，我听上去你分析的有道理，具体怎么操作哈？能给个命令指引不？

把负载高的应用kill掉，或者改用netdata，里面有个监控丢包，双核CPU，负载不应该超2

closed 2021-06-20 by coolsnowwolf