coolsnowwolf/lede #6090: K2P A1 A2最新源码基本可以软重启了！

qjiang888, 2020-12-23:
但是还是有概率起不来，拔电就行， 无线信道无法修改，不知道是不是驱动的问题！！

SilverHunterLib, 2020-12-23:
请教一下，最新源码K2P刷完以后还有重启无法连上路由器的问题吗？

qjiang888, 2020-12-24:
make clean 一下再编译看看！

qifly, 2020-12-25:
> 请教一下，最新源码K2P刷完以后还有重启无法连上路由器的问题吗？我的12.19源码编译的固件，重启无法连接路由器，不分配地址，拔电重启都不好使

昨天编译的，同无法连接路由器，不分配地址，拔电重启都不好使

ltactor, 2020-12-25:
> > 请教一下，最新源码K2P刷完以后还有重启无法连上路由器的问题吗？我的12.19源码编译的固件，重启无法连接路由器，不分配地址，拔电重启都不好使
> 
> 昨天编译的，在8月24日OpenWrt R20.7.20(5.4.51)上面升级后重启，无法连接路由器，不分配地址，拔电重启都不好使。在breed里面清空配置，重新刷入后可以启动，但比8月24日编译的启动慢，大约1分30秒才能获取ip地址，之前40多秒就可以。配置WAN，无线，Turbo ACC 开启FLOW，BBR，HWNAT加速后重启，又无法连接路由器，不分配地址，拔电重启都不好使。

我硬改32M，也出现此问题。

SilverHunterLib, 2020-12-25:
> > 请教一下，最新源码K2P刷完以后还有重启无法连上路由器的问题吗？我的12.19源码编译的固件，重启无法连接路由器，不分配地址，拔电重启都不好使
> 
> 昨天编译的，在8月24日OpenWrt R20.7.20(5.4.51)上面升级后重启，无法连接路由器，不分配地址，拔电重启都不好使。在breed里面清空配置，重新刷入后可以启动，但比8月24日编译的启动慢，大约1分30秒才能获取ip地址，之前40多秒就可以。配置WAN，无线，Turbo ACC 开启FLOW，BBR，HWNAT加速后重启，又无法连接路由器，不分配地址，拔电重启都不好使。

算了，我还是不刷了，现在用的11.15的。最近固件问题比较多，这个重启无法连接路由器的问题一直都没解决，我从12.13刷到12.20，都不行，估计也没人解决了

AmadeusGhost, 2020-12-26:
> 估计也没人解决了

没人贴出问题日志解决啥？

ltactor, 2020-12-27:
> > 估计也没人解决了
> 
> 没人贴出问题日志解决啥？

请问怎样找日志。我在var\log没找到启动日志。
我是k2p 32M硬改。 12月26日编译固件，第一次可以启动，配置后，进行重启不可以启动，亮蓝灯不闪烁。  断电后有1/3的机会可以启动成功，启动成功蓝灯闪烁。  没有查询到日志。
之前手动更改k2p 内存地址没有出现此问题。
如有需要我可以进行k2p测试工作。感谢大神。

AmadeusGhost, 2020-12-27:
> 请问怎样找日志。我在var\log没找到启动日志。

重启失败的时候的 ttl 启动日志

ltactor, 2020-12-27:
> > 请问怎样找日志。我在var\log没找到启动日志。
> 
> 重启失败的时候的 ttl 启动日志

感谢回复，最晚元旦假期，可以导出ttl日志。  感谢。

qjiang888, 2020-12-28:
大佬怎么提取TTL 日志啊？ 我有工具，不知道焊接哪里？ 今天git pull 刷了下 还是不能重启！！！！！！

qifly, 2020-12-29:
> 大佬怎么提取TTL 日志啊？ 我有工具，不知道焊接哪里？ 今天git pull 刷了下 还是不能重启！！！！！！
这里有个视频，你可以参照下。
https://www.bilibili.com/s/video/BV1AE411L7fn

qifly, 2020-12-30:
> > 估计也没人解决了
> 
> 没人贴出问题日志解决啥？

今天make clean，rm .config，重新配置之后编译了最新版，在breed里面清除配置之后刷了，可以启动。
刷最新版和第一次启动的日志如下：
[flash_1st_boot.txt](https://github.com/coolsnowwolf/lede/files/5754247/flash_1st_boot.txt)
启动之后，登录luci管理，直接点了重启，就无法启动了，这个日志没保存到。之后拔电硬重启，可以启动。日志如下：
[hardreboot.txt](https://github.com/coolsnowwolf/lede/files/5754249/hardreboot.txt)
启动起来后，登录luci和往常一样配置，日志如下：
[config.txt](https://github.com/coolsnowwolf/lede/files/5754245/config.txt)
之后又测试了在luci管理里面重启，无法启动，不同时间的两次的日志：
[softreboot.txt](https://github.com/coolsnowwolf/lede/files/5754251/softreboot.txt)
[softreboot2.txt](https://github.com/coolsnowwolf/lede/files/5754252/softreboot2.txt)
配置完之后重启起来的日志：
[ready.txt](https://github.com/coolsnowwolf/lede/files/5754250/ready.txt)

希望能有用。

ltactor, 2020-12-31:
> > 请问怎样找日志。我在var\log没找到启动日志。
> 
> 重启失败的时候的 ttl 启动日志

您好，已经有人上传日志，请问还需要日志文件吗？

AmadeusGhost, 2021-01-01:
> 您好，已经有人上传日志，请问还需要日志文件吗？

不用了，看样子还是去掉 32M 版本的支持吧

ltactor, 2021-01-01:
> > 您好，已经有人上传日志，请问还需要日志文件吗？
> 
> 不用了，看样子还是去掉 32M 版本的支持吧

我是硬改32M，之前是通过修改文件可以顺畅的支持。希望支持32M，我可以负责测试。感谢回复。

AmadeusGhost, 2021-01-01:
@qifly @ltactor 试试这样子改了后能不能正常重启（需要 `make clean`）: https://github.com/AmadeusGhost/lede/commit/7658518358a2c99010522483095d1526e90ecf03

ltactor, 2021-01-01:
> @qifly @ltactor 试试这样子改了后能不能正常重启（需要 `make clean`）: [AmadeusGhost@7658518](https://github.com/AmadeusGhost/lede/commit/7658518358a2c99010522483095d1526e90ecf03)

感谢回复。我现在就编译。感谢。

ltactor, 2021-01-02:
> @qifly @ltactor 试试这样子改了后能不能正常重启（需要 `make clean`）: [AmadeusGhost@7658518](https://github.com/AmadeusGhost/lede/commit/7658518358a2c99010522483095d1526e90ecf03)

感谢大神，已经编译成功。
使用全新git glong
选择curl ssmp iphelp 等插件。
刷机成功后直接软起没问题。
回复备份文件软起没问题。
第一次启动需要2分钟。
我是硬改32m。
暂时没有问题。感谢。

qjiang888, 2021-01-02:
普通k2 我重新克隆编译刷了还是不行

ltactor, 2021-01-02:
> > @qifly @ltactor 试试这样子改了后能不能正常重启（需要 `make clean`）: [AmadeusGhost@7658518](https://github.com/AmadeusGhost/lede/commit/7658518358a2c99010522483095d1526e90ecf03)
> 
> 感谢大神，已经编译成功。
> 使用全新git glong
> 选择curl ssmp iphelp 等插件。
> 刷机成功后直接软起没问题。
> 恢复备份文件软起没问题。
> 第一次启动需要2分钟。
> 我是硬改k2p 32m。
> 暂时没有问题。感谢。

我又进行一次断电启动，也没有问题。感谢修复。

qjiang888, 2021-01-05:
请问大佬，普通K2P 也修改上面那三个文件吗？？

ltactor, 2021-01-05:
> 请问大佬，普通K2P 也修改上面那三个文件吗？？

我觉得不用吧，配置选择k2p就可以吧。
记得切换到提交版本。

qjiang888, 2021-01-05:
不行，我删掉目录重新克隆编译 ，普通k2p是重启不了的

qjiang888, 2021-01-05:
我克隆了 modified  这个分支编译完 K2P16M版本还是没法启动！！！

SilverHunterLib, 2021-01-05:
> > 请问大佬，普通K2P 也修改上面那三个文件吗？？
> 
> 我觉得不用吧，配置选择k2p就可以吧。
> 记得切换到提交版本。

不好意思，请问用什么指令切换到提交版本？刚接触编译，我用git reset 加commit id提示找不到，是参数不对吗？

closed 2021-01-13 by coolsnowwolf

lphgor, 2021-01-13:
楼主能正常重启了吗？
去掉超频补丁以后，启动的问题应该解决了，我现在可以正常启动了。
但是重启还是有问题，ssh里面reboot，或者网页里面执行重启，基本上都会死机，没有反应，只能断电。难得有一次能重启的。

qjiang888, 2021-01-15:
> 楼主能正常重启了吗？
> 去掉超频补丁以后，启动的问题应该解决了，我现在可以正常启动了。
> 但是重启还是有问题，ssh里面reboot，或者网页里面执行重启，基本上都会死机，没有反应，只能断电。 难得有一次能重启的。

可以了 ，不行你删掉lede目录 重新git编译下看看

lphgor, 2021-01-15:
> 可以了 ，不行你删掉lede目录 重新git编译下看看

我这边是云编译的，应该每次编译都是全新的代码，但还是不能正常重启。可能要拆机上TTL看日志了

ltactor, 2021-01-16:
> > 可以了 ，不行你删掉lede目录 重新git编译下看看
> 
> 我这边是云编译的，应该每次编译都是全新的代码，但还是不能正常重启，启动完全靠运气，没法日常使用。可能要拆机上TTL看日志了

我是32M硬改，本来使用AmadeusGhost@7658518  这个版本解决了启动问题。  现在使用更新版本，还是无法软起动。  也没有单独32M选项。

PR #6190 (2021-01-13): Revert "ramips: mt7621 default overclock to 1000Mhz"
建议还是用 Breed 来超频更为合适
Fix: #6090 , Fix: #6166 , Fix: #6172