coolsnowwolf/lede #6091: 树莓派 4 首页信息无法加载问题

SuLingGG, 2020-12-24:
树莓派 4 的 OpenWrt 在运行一段时间后，常会出现首页信息加载不出来的问题，具体表现在时间，CPU 负载/温度，架构等信息无法正常加载:

[image]

经过排查，发现这些信息由 `/cgi-bin/luci/admin/status/overview?status=1` 提供，故手动访问此链接，经过漫长等待，uhttpd 返回了 `The process did not produce any response` 的响应。

于是通过 htop 查看进程情况，发现此时负载居高不下，并且发现进程:

`cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq`

在读取 CPU 频率时无法得到回应:

[image]

手动获取 CPU 频率，同样被卡住:

[image]

故现在基本可以确定是因为 autocore-arm 在尝试读取 CPU 频率时内核 未响应/未返回数据 造成的问题。

经重启后此问题消失:

[image]

[image]

[image]

但经多人测试，此问题在重启后仍不时出现，且可以确定只有树莓派 4 存在这个问题，树莓派 3 以及其他设备未出现。

hcym, 2020-12-24:
一天重启一下，目前为止没有任何事情

AmadeusGhost, 2020-12-24:
> `cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq`
> 在读取 CPU 频率时无法得到回应

没有响应的时候试试 `cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` 或者 `vcgencmd get_config arm_freq`

SuLingGG, 2020-12-24:
在卡住的情况下用这两种方法是能拿到数据的，但是 `cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq` 拿到的数据看起来不太对，`vcgencmd get_config arm_freq` 的数据是对的。

```
# root @ OpenWrt in ~ [15:27:27] 
$ time cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq                                                            
1000000
cat /sys/devices/system/cpu/cpu0/cpufreq/scaling_cur_freq  0.00s user 0.00s system 79% cpu 0.002 total

# root @ OpenWrt in ~ [15:27:42] 
$ time vcgencmd get_config arm_freq                                                         
arm_freq=1500
vcgencmd get_config arm_freq  0.00s user 0.00s system 82% cpu 0.002 total

# root @ OpenWrt in ~ [15:27:58] 
$ time cat /sys/devices/system/cpu/cpufreq/policy0/cpuinfo_cur_freq ## 这里没响应，说明卡住了

```

31415926pai, 2020-12-24:
我的树莓派4B是作的旁路，用的时候才开，没有发现这个问题。

hcym, 2020-12-24:
注意了一下，早上4点至今看上去正常，明天早上还是要重启的

closed 2020-12-25 by coolsnowwolf

deathsatan, 2020-12-30:
我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。

SuLingGG, 2020-12-30:
> 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。

试试：

```
curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
```

但是执行后会阉割 CPU 频率显示。自行斟酌吧。

deathsatan, 2020-12-30:
> > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> 
> 试试：
> 
> ```
> curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> ```
> 
> 但是执行后会阉割 CPU 频率显示。自行斟酌吧。

试了，依然高负载。

SuLingGG, 2020-12-30:
> > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > 
> > 
> > 试试：
> > ```
> > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > ```
> > 
> > 
> > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> 
> 试了，依然高负载。

需要重启。

deathsatan, 2020-12-30:
> > > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > > 
> > > 
> > > 试试：
> > > ```
> > > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > > ```
> > > 
> > > 
> > > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> > 
> > 
> > 试了，依然高负载。
> 
> 需要重启。

重启了

deathsatan, 2020-12-30:
> > > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > > 
> > > 
> > > 试试：
> > > ```
> > > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > > ```
> > > 
> > > 
> > > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> > 
> > 
> > 试了，依然高负载。
> 
> 需要重启。

重启了，信息可以显示了，负载也没那么高了，但是还是在1左右徘徊。

SuLingGG, 2020-12-30:
> > > > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > > > 
> > > > 
> > > > 试试：
> > > > ```
> > > > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > > > ```
> > > > 
> > > > 
> > > > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> > > 
> > > 
> > > 试了，依然高负载。
> > 
> > 
> > 需要重启。
> 
> 重启了，信息可以显示了，负载也没那么高了，但是还是在 1 左右徘徊。

能提供下内核日志吗？

deathsatan, 2020-12-31:
> > > > > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > > > > 
> > > > > 
> > > > > 试试：
> > > > > ```
> > > > > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > > > > ```
> > > > > 
> > > > > 
> > > > > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> > > > 
> > > > 
> > > > 试了，依然高负载。
> > > 
> > > 
> > > 需要重启。
> > 
> > 
> > 重启了，信息可以显示了，负载也没那么高了，但是还是在 1 左右徘徊。
> 
> 能提供下内核日志吗？

昨天又重新编译了一版，负载是不高了，但是还卡，流量稍微高一点就卡的连luci都打不开。
日志如下：

`Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.972097] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.977144] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.985715] br-lan: port 1(eth0) entered disabled state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.991133] device eth0 entered promiscuous mode
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.996829] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   11.002086] br-lan: port 1(eth0) entered forwarding state
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: bridge 'br-lan' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'eth0' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'lo' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: radio0 (1078): command failed: No error information (-524)
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Successfully dropped root privileges.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: avahi-daemon 0.8 starting up.
Wed Dec 30 19:40:16 2020 daemon.warn avahi-daemon[1179]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: No service file found in /etc/avahi/services.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.2.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface br-lan.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv6 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Network interface enumeration completed.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 192.168.2.2 on br-lan.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for ::1 on lo.*.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 127.0.0.1 on lo.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering HINFO record with values 'AARCH64'/'LINUX'.
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.404466] fast-classifier (PBR safe v2.1.4a): starting up
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.410122] fast-classifier: registered
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: net.ipv4.tcp_congestion_control = cubic
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: started, v1.31.1
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: sending discover
Wed Dec 30 19:40:16 2020 user.info adblock-4.0.6[807]: download of 'adaway' failed, url: https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt, rule: /^127\.0\.0\.1[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories: -, rc: 1, log:  12/30 19:40:16 [[1;32mNOTICE[0m] Downloading 1 item(s)  12/30 19:40:16 [[1;31mERROR[0m] CUID#7 - Download aborted. URI=https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt Exception: [AbstractCommand.cc:856] errorCode=1 Failed to establish connection, cause: Connection refused  12/30 19:40:16 [[1;32mNOTICE[0m] Download GID#5bbb27d9917559b5 not complete: .//tmp/tmp.MAhAAI/tmp.jogdln.adaway.load  Download Results: gid   |stat|avg speed  |path/URI ======+====+===========+======================================================= 5bbb27|ERR |       0B/s|.//tmp/tmp.MAhAAI/tmp. [... cut ...]
Thu Dec 31 09:59:54 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 09:59:56 2020 cron.err crond[19892]: crond (busybox 1.31.1) started, log level 8
Thu Dec 31 09:59:57 2020 daemon.info dnsmasq[19857]: Connected to system UBus
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 10:00:07 2020 daemon.info dnsmasq[20413]: Connected to system UBus
`

deathsatan, 2020-12-31:
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.972097] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.977144] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.985715] br-lan: port 1(eth0) entered disabled state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.991133] device eth0 entered promiscuous mode
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.996829] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   11.002086] br-lan: port 1(eth0) entered forwarding state
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: bridge 'br-lan' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'eth0' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'lo' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: radio0 (1078): command failed: No error information (-524)
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Successfully dropped root privileges.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: avahi-daemon 0.8 starting up.
Wed Dec 30 19:40:16 2020 daemon.warn avahi-daemon[1179]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: No service file found in /etc/avahi/services.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.2.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface br-lan.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv6 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Network interface enumeration completed.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 192.168.2.2 on br-lan.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for ::1 on lo.*.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 127.0.0.1 on lo.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering HINFO record with values 'AARCH64'/'LINUX'.
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.404466] fast-classifier (PBR safe v2.1.4a): starting up
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.410122] fast-classifier: registered
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: net.ipv4.tcp_congestion_control = cubic
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: started, v1.31.1
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: sending discover
Wed Dec 30 19:40:16 2020 user.info adblock-4.0.6[807]: download of 'adaway' failed, url: https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt, rule: /^127\.0\.0\.1[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories: -, rc: 1, log:  12/30 19:40:16 [[1;32mNOTICE[0m] Downloading 1 item(s)  12/30 19:40:16 [[1;31mERROR[0m] CUID#7 - Download aborted. URI=https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt Exception: [AbstractCommand.cc:856] errorCode=1 Failed to establish connection, cause: Connection refused  12/30 19:40:16 [[1;32mNOTICE[0m] Download GID#5bbb27d9917559b5 not complete: .//tmp/tmp.MAhAAI/tmp.jogdln.adaway.load  Download Results: gid   |stat|avg speed  |path/URI ======+====+===========+======================================================= 5bbb27|ERR |       0B/s|.//tmp/tmp.MAhAAI/tmp.jogdln.adaway.load  Status Legend: (ERR):error occurred.  aria2 will resume download if the transfer is restarted. If there are any error
Wed Dec 30 19:40:17 2020 kern.info kernel: [   11.999604] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Wed Dec 30 19:40:17 2020 authpriv.info dropbear[1411]: Not backgrounding
[... cut ...]
Thu Dec 31 09:59:54 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 09:59:56 2020 cron.err crond[19892]: crond (busybox 1.31.1) started, log level 8
Thu Dec 31 09:59:57 2020 daemon.info dnsmasq[19857]: Connected to system UBus
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 10:00:07 2020 daemon.info dnsmasq[20413]: Connected to system UBus

deathsatan, 2020-12-31:
负载又上来了：
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.972097] bcmgenet fd580000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.977144] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.985715] br-lan: port 1(eth0) entered disabled state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.991133] device eth0 entered promiscuous mode
Wed Dec 30 19:40:16 2020 kern.info kernel: [   10.996829] br-lan: port 1(eth0) entered blocking state
Wed Dec 30 19:40:16 2020 kern.info kernel: [   11.002086] br-lan: port 1(eth0) entered forwarding state
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: bridge 'br-lan' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'lan' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is enabled
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is setting up now
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' is now up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'eth0' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Network device 'lo' link is up
Wed Dec 30 19:40:16 2020 daemon.notice netifd: Interface 'loopback' has link connectivity
Wed Dec 30 19:40:16 2020 daemon.notice netifd: radio0 (1078): command failed: No error information (-524)
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90kms: KMS Server has started.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Found user 'nobody' (UID 65534) and group 'nogroup' (GID 65534).
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Successfully dropped root privileges.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: avahi-daemon 0.8 starting up.
Wed Dec 30 19:40:16 2020 daemon.warn avahi-daemon[1179]: WARNING: No NSS support for mDNS detected, consider installing nss-mdns!
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: No service file found in /etc/avahi/services.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface br-lan.IPv4 with address 192.168.2.2.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface br-lan.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv6 with address ::1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv6 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Joining mDNS multicast group on interface lo.IPv4 with address 127.0.0.1.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: New relevant interface lo.IPv4 for mDNS.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Network interface enumeration completed.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 192.168.2.2 on br-lan.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for ::1 on lo.*.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering new address record for 127.0.0.1 on lo.IPv4.
Wed Dec 30 19:40:16 2020 daemon.info avahi-daemon[1179]: Registering HINFO record with values 'AARCH64'/'LINUX'.
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.404466] fast-classifier (PBR safe v2.1.4a): starting up
Wed Dec 30 19:40:16 2020 kern.alert kernel: [   11.410122] fast-classifier: registered
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: net.ipv4.tcp_congestion_control = cubic
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Wed Dec 30 19:40:16 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: started, v1.31.1
Wed Dec 30 19:40:16 2020 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: sending discover
Wed Dec 30 19:40:16 2020 user.info adblock-4.0.6[807]: download of 'adaway' failed, url: https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt, rule: /^127\.0\.0\.1[[:space:]]+([[:alnum:]_-]{1,63}\.)+[[:alpha:]]+([[:space:]]|$)/{print tolower($2)}, categories: -, rc: 1, log:  12/30 19:40:16 [[1;32mNOTICE[0m] Downloading 1 item(s)  12/30 19:40:16 [[1;31mERROR[0m] CUID#7 - Download aborted. URI=https://raw.githubusercontent.com/AdAway/adaway.github.io/master/hosts.txt Exception: [AbstractCommand.cc:856] errorCode=1 Failed to establish connection, cause: Connection refused  12/30 19:40:16 [[1;32mNOTICE[0m] Download GID#5bbb27d9917559b5 not complete: .//tmp/tmp.MAhAAI/tmp.jogdln.adaway.load  Download Results: gid   |stat|avg speed  |path/URI ======+====+===========+======================================================= 5bbb27|ERR |       0B/s|.//tmp/tmp.MAhAAI/tmp.jogdln.adaway.load  Status Legend: (ERR):error occurred.  aria2 will resume download if the transfer is restarted. If there are any error
Wed Dec 30 19:40:17 2020 kern.info kernel: [   11.999604] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
Wed Dec 30 19:40:17 2020 authpriv.info dropbear[1411]: Not backgrounding
[... cut ...]
Thu Dec 31 09:59:54 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 09:59:56 2020 cron.err crond[19892]: crond (busybox 1.31.1) started, log level 8
Thu Dec 31 09:59:57 2020 daemon.info dnsmasq[19857]: Connected to system UBus
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Thu Dec 31 10:00:03 2020 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Thu Dec 31 10:00:07 2020 daemon.info dnsmasq[20413]: Connected to system UBus

SuLingGG, 2020-12-31:
> > > > > > > 我天天有这个问题，问题出现时网络变的极不稳定，所以网络不好时去看负载都很高，实际又没干啥。
> > > > > > 
> > > > > > 
> > > > > > 试试：
> > > > > > ```
> > > > > > curl -fsSL https://file.mlapp.cn/patch/fix-rpi4-overview.sh | bash
> > > > > > ```
> > > > > > 
> > > > > > 
> > > > > > 但是执行后会阉割 CPU 频率显示。自行斟酌吧。
> > > > > 
> > > > > 
> > > > > 试了，依然高负载。
> > > > 
> > > > 
> > > > 需要重启。
> > > 
> > > 
> > > 重启了，信息可以显示了，负载也没那么高了，但是还是在 1 左右徘徊。
> > 
> > 
> > 能提供下内核日志吗？
> 
> 昨天又重新编译了一版，负载是不高了，但是还卡。流量稍微高一点就卡的连 luci 都打不开，然后不一会流量就会降下来。
> 日志如下：

这是系统日志吧，提供下内核日志我看看。

deathsatan, 2020-12-31:
> 这是系统日志吧，提供下内核日志我看看。

[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.4.85 (runner@fv-az56-539) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2999-81a53ead5)) #0 SMP Tue Dec 29 14:28:04 2020
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 3792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 242688 pages, LIFO batch:63
[    0.000000]   DMA32 zone: 4096 pages used for memmap
[    0.000000]   DMA32 zone: 262144 pages, LIFO batch:63
[    0.000000] percpu: Embedded 21 pages/cpu s45144 r8192 d32680 u86016
[    0.000000] pcpu-alloc: s45144 r8192 d32680 u86016 alloc=21*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: EL2 vector hardening
[    0.000000] ARM_SMCCC_ARCH_WORKAROUND_1 missing from firmware
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 496944
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=DC:A6:32:E4:1B:A8 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyAMA0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=squashfs,ext4 rootwait
[    0.000000] Dentry cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Inode-cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x37400000-0x3b400000] (64MB)
[    0.000000] Memory: 1837540K/2019328K available (6654K kernel code, 408K rwdata, 1720K rodata, 512K init, 527K bss, 116252K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: 	CONFIG_RCU_FANOUT set to non-default value of 32.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x2dc/0x3dc with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000005] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000157] Console: colour dummy device 80x25
[    0.000391] printk: console [tty1] enabled
[    0.000425] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000447] pid_max: default: 32768 minimum: 301
[    0.000607] Mount-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.000658] Mountpoint-cache hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.001314] Disabling memory control group subsystem
[    0.002047] ASID allocator initialised with 32768 entries
[    0.002137] rcu: Hierarchical SRCU implementation.
[    0.002769] smp: Bringing up secondary CPUs ...
[    0.003170] Detected PIPT I-cache on CPU1
[    0.003219] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.003667] Detected PIPT I-cache on CPU2
[    0.003699] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.004205] Detected PIPT I-cache on CPU3
[    0.004238] CPU3: Booted secondary processor 0x0000000003 [0x410fd083]
[    0.004307] smp: Brought up 1 node, 4 CPUs
[    0.004334] SMP: Total of 4 processors activated.
[    0.004347] CPU features: detected: 32-bit EL0 Support
[    0.004360] CPU features: detected: CRC32 instructions
[    0.004415] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.004432] CPU: All CPU(s) started at EL2
[    0.004461] alternatives: patching kernel code
[    0.011986] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[    0.012027] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.013758] pinctrl core: initialized pinctrl subsystem
[    0.014501] NET: Registered protocol family 16
[    0.015883] DMA: preallocated 1024 KiB pool for atomic allocations
[    0.016423] cpuidle: using governor ladder
[    0.016453] cpuidle: using governor menu
[    0.016723] Serial: AMBA PL011 UART driver
[    0.018189] bcm2835-mbox fe00b880.mailbox: mailbox enabled
[    0.028038] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-04-15 11:41, variant start
[    0.032041] raspberrypi-firmware soc:firmware: Firmware hash is 1103ec16df2571274600b377411499bf5aed6d0d
[... 190 lines cut ...]
[    9.061045] PPP generic driver version 2.4.2
[    9.065758] PPP MPPE Compression module registered
[    9.070904] NET: Registered protocol family 24
[    9.086018] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    9.330118] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[... cut ...]
[   10.968756] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   11.336549] fast-classifier (PBR safe v2.1.4a): starting up
[   11.342220] fast-classifier: registered
[   13.276702] br-lan: port 2(wlan0) entered blocking state
[   13.282046] br-lan: port 2(wlan0) entered disabled state
[   13.287549] device wlan0 entered promiscuous mode
[   17.389617] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   17.396080] br-lan: port 2(wlan0) entered blocking state
[   17.401395] br-lan: port 2(wlan0) entered forwarding state

PR #6095 (2020-12-25): autocore: avoid resource busy issue on rpi4
fix #6091