coolsnowwolf/lede #6102: mt7615 还有必要使用闭源版本驱动吗？

jinglei207, 2020-12-25:
今天去openwrt 官网刷个最新版的 ac2100固件，发现无线能跑到600M+， 在远距离测试基本和闭源没区别。
开源的更安全，支持wpa3， mesh。。。感觉现在闭源完全没必要存在了啊！
还是我漏了什么？

ZeaKyX, 2020-12-26:
#5002 
（话说大雕为什么不试一下GitHub的Discussions呢 @coolsnowwolf ）

NagaseKouichi, 2020-12-26:
我用闭源驱动，无线只能跑到500m上传，300m下载。用237大佬的固件可以跑到800m

jinglei207, 2020-12-26:
谁是237？ 给个链接去瞧瞧。

NagaseKouichi, 2020-12-27:
> 谁是237？ 给个链接去瞧瞧。

https://www.right.com.cn/forum/thread-4043514-1-1.html
这个固件

coolsnowwolf, 2020-12-28:
主要是为了预留给无线 HWNAT 用的。只有闭源驱动才能和 HWNAT 配合

NagaseKouichi, 2020-12-28:
> 主要是为了预留给无线 HWNAT 用的。只有闭源驱动才能和 HWNAT 配合

有计划支持无线的HWNAT？

hyird, 2020-12-28:
> 我用闭源驱动，无线只能跑到500m上传，300m下载。用237大佬的固件可以跑到800m

hwnat 目前只能4.4内核~~~~

hyird, 2020-12-28:
> 主要是为了预留给无线 HWNAT 用的。只有闭源驱动才能和 HWNAT 配合

目前的闭源驱动源码 hwnat 不是只能4.4内核跑么 大雕准备 弄到5.4内核上？

AmadeusGhost, 2020-12-31:
有的，像 k2p 这种设备还是得用闭源驱动，开源驱动虽然能够正常驱动了，大部分人估计并不懂得怎么设置（luci 还不能很好的进行控制，存在一些参数显示问题）。
其实使用闭源驱动最重要的一点是稳定，君不见今年上半年还在用开源驱动的时候，本库一堆 mtk 无线有问题的 issues，我自己也因为回退了/升了 mt76 的版本被喷了。🌚

AmadeusGhost, 2020-12-31:
> 有计划支持无线的HWNAT？

7月那时候就有计划了，然后鸽了🌚

NagaseKouichi, 2020-12-31:
> > 有计划支持无线的HWNAT？
> 
> 7月那时候就有计划了，然后鸽了🌚

期待一下，希望性能和237大佬的固件一样，可以跑到800M的5G NAT

1715173329, 2021-01-01:
> > 有计划支持无线的HWNAT？
> 
> 7月那时候就有计划了，然后鸽了🌚

据 Chen Mingqiang 大佬所述他的 natflow 似乎已经实现了无线的 HWNAT？

meoow, 2021-03-17:
闭源不支持802.11r，或者我实在没找到怎么配置

closed 2021-06-20 by coolsnowwolf