coolsnowwolf/lede #6106: make error

saltfishh, 2020-12-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
编译的时候报错
A：

### (2) 路由器型号和固件版本
编译的x86版本
A：

### (3) 详细日志
make[3]: Leaving directory '/home/wind/Project/github/lede/feeds/passwall/trojan-go'
time: package/feeds/passwall/trojan-go/compile#42.26#5.40#34.65
make[2]: Leaving directory '/home/wind/Project/github/lede'
make[1]: *** [package/Makefile:107: /home/wind/Project/github/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wind/Project/github/lede'
make: *** [/home/wind/Project/github/lede/include/toplevel.mk:224：world] 错误 2

A：
搜了一下,issue里貌似没有,才来提交新的issue.
有遇到过这个的吗

AmadeusGhost, 2020-12-26:
`make V=s -j1` 跑一遍看具体报错再说

saltfishh, 2020-12-26:
> make[3]: Leaving directory '/home/wind/Project/github/lede/package/lean/amule'
time: package/lean/amule/compile#0.34#0.04#0.38
make[2]: *** [package/Makefile:113: package/lean/amule/compile] Error 2
make[2]: Leaving directory '/home/wind/Project/github/lede'
make[1]: *** [package/Makefile:107: /home/wind/Project/github/lede/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/wind/Project/github/lede'
make: *** [/home/wind/Project/github/lede/include/toplevel.mk:224：world] 错误 2

还是这样,也不知道那里出问题了

唉

AmadeusGhost, 2020-12-26:
amule 编译出错了
`make package/amule/{clean,compile} V=s -j1` 看看出错

jinglei207, 2020-12-26:
> 还是这样,也不知道那里出问题了
>
> 唉

删了，从新弄。

saltfishh, 2020-12-27:
> amule 编译出错了
> `make package/amule/{clean,compile} V=s -j1` 看看报错

`make package/lean/amule/{clean,compile} V=s -j1`
> /home/wind/Project/github/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: skipping incompatible /usr/lib/libm.so when searching for -lm
/home/wind/Project/github/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: skipping incompatible /usr/lib/libm.a when searching for -lm
/home/wind/Project/github/lede/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/lib/gcc/aarch64-openwrt-linux-musl/8.4.0/../../../../aarch64-openwrt-linux-musl/bin/ld: skipping incompatible /usr/lib/libgcc_s.so.1 when searching for libgcc_s.so.1
collect2: error: ld returned 1 exit status
make[7]: *** [Makefile:515: amuleweb] Error 1
make[6]: *** [Makefile:440: all-recursive] Error 1
make[5]: *** [Makefile:3413: all-recursive] Error 1
make[4]: *** [Makefile:590: all-recursive] Error 1
make[3]: *** [Makefile:497: all] Error 2
make[3]: Leaving directory '/home/wind/Project/github/lede/build_dir/target-aarch64_cortex-a72_musl/aMule-2.3.2'
make[2]: *** [Makefile:113: /home/wind/Project/github/lede/build_dir/target-aarch64_cortex-a72_musl/aMule-2.3.2/.built] Error 2
make[2]: Leaving directory '/home/wind/Project/github/lede/package/lean/amule'
time: package/lean/amule/compile#104.63#9.18#111.43
make[1]: *** [package/Makefile:113: package/lean/amule/compile] Error 2
make[1]: Leaving directory '/home/wind/Project/github/lede'
make: *** [/home/wind/Project/github/lede/include/toplevel.mk:224：package/lean/amule/compile] 错误 2

看起来确实是amule这个出错了.

saltfishh, 2020-12-27:
> amule 编译出错了
> `make package/amule/{clean,compile} V=s -j1` 看看报错

不使用电驴就通过了,谢谢啦~

closed 2020-12-27 by saltfishh