coolsnowwolf/lede #6109: 友华wr1200js 接口桥接不生效问题

maxmib, 2020-12-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：路由接在主路由下做ap，原wan口连接上级路由lan口，设置有两个接口，1个guest，1个lan，guest接口里wan.3（vlan id 3）和访客无线、lan1（未设置vlan）桥接，实际使用访客无线可以正确获取ip，但lan1口接笔记本无法获取ip。

### (2) 路由器型号和固件版本

A：友华wr1200js  固件版本 OpenWrt R20.10.20 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：

jinglei207, 2020-12-26:
安装luci relayd.  
https://openwrt.org/docs/guide-user/network/wifi/relay_configuration

maxmib, 2020-12-27:
> 安装luci relayd.
> https://openwrt.org/docs/guide-user/network/wifi/relay_configuration

我这个情况和这文章里的不一样，文章里的是ap无线连接上级路由lan不能获取主路由分配的ip，我的配置是ap有线连接上级路由通过vlan扩展主路由的主wifi和访客wifi，目前无线部分是生效的，但是我想把几个lan口也加进访客网络，就出现了上面说的情况不能获取ip，笔记本手动设置固定ip也没法上网，搞不清楚问题在哪，wr1200js不是通过vlan划分网口，每个网口都虚拟了一个网卡，按道理在接口配置里像配置访客无线一样把要用的网口桥接到wan.3就能用。

LGA1150, 2020-12-28:
这个需要等 luci 实现 VLAN filtering 设置
或者你可以单独给 wan.3 建一个网桥，然后把这个网桥再桥接到 br-lan

maxmib, 2021-01-01:
> 这个需要等 luci 实现 VLAN filtering 设置
> 
> 或者你可以单独给 wan.3 建一个网桥，然后把这个网桥再桥接到 br-lan

意思是再单独建一个接口，单独桥接wan.3和需要的网口？

maxmib, 2021-02-23:
> 这个需要等 luci 实现 VLAN filtering 设置
> 或者你可以单独给 wan.3 建一个网桥，然后把这个网桥再桥接到 br-lan
这么搞了一下，还是不行。新建了一个接口，物理设置里选了桥接接口，选了wan.3和lan1，还是不能获取到访客网络的ip

LGA1150, 2021-02-24:
> 这么搞了一下，还是不行。新建了一个接口，物理设置里选了桥接接口，选了wan.3和lan1，还是不能获取到访客网络的ip

只选择 `wan.3`，再去编辑 LAN，把你新建的接口勾上

maxmib, 2021-02-24:
> > 这么搞了一下，还是不行。新建了一个接口，物理设置里选了桥接接口，选了wan.3和lan1，还是不能获取到访客网络的ip
> 
> 只选择 `wan.3`，再去编辑 LAN，把你新建的接口勾上

[image]

这样吗，因为luci界面不会显示自建的接口，是否是使用自定义接口输入自建接口名称来指定到自建的接口

LGA1150, 2021-02-25:
自定义接口填写 `@逻辑接口名`

maxmib, 2021-05-01:
> 自定义接口填写 `@逻辑接口名`

填写wan3还是@wan3还是@wan.3

closed 2021-06-20 by coolsnowwolf