coolsnowwolf/lede #6114: 日志里面不断打irq中断告警是什么原因？

galaxyskyknight, 2020-12-27:
i3865u,挂了两个玩客云，网络流量比较大，最新的5.x内核的固件，不断地系统日志打下面的玩意，啥意思？

型号：Intel(R) Celeron(R) CPU 3865U @ 1.80GHz : 2 Core 2 Thread (CpuMark : 26964.852999 Scores)
版本：OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)
内核：5.4.84

Sun Dec 27 19:52:28 2020 kern.warn kernel: [125731.849689] ------------[ cut here ]------------Sun Dec 27 19:52:28 2020 kern.warn kernel: [125731.854565] WARNING: CPU: 1 PID: 0 at 0xffffffffa189022aSun Dec 27 19:52:28 2020 kern.warn kernel: [125731.860167] Modules linked in: fast_classifier xt_FULLCONENAT ath9k iwl4965 iwl3945 ath9k_htc ath9k_common xt_connlimit wl18xx wl12xx rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb rt2500pci rt2400pci qcserial pppoe ppp_async option nf_conncount mt76x0u mt76x0e mt76x0_common l2tp_ppp iwlmvm iwlegacy iwldvm ipw2200 ipw2100 iptable_nat gspca_m5602 gspca_konica gspca_jeilinj gspca_gl860 gspca_finepix gspca_etoms gspca_conex cdc_mbim carl9170 btcoexist brcmsmac brcmfmac b43legacy b43 ath9k_hw ath6kl_usb ath6kl_core ath5k ath10k_pci ath10k_core ath adm8211 zd1211rw xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_CT wlcore uvcvideo usb8xxx usb_wwan ti_usb_3410_5052 sr9700 snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codecSun Dec 27 19:52:28 2020 kern.warn kernel: [125731.860192]  smsc95xx sierra_net rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl8812au rtl818x_pci rtl8187 rt73usb rt61pci rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rsi_usb rsi_91x rndis_host qmi_wwan pwc p p t p pppox ppp_mppe ppp_generic plusb p54usb p54pci p54common orinoco_pci orinoco nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_p p t p nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_p p t p nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mwl8k mwifiex_pcie mwifiex mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615e mt7615_common mt7603e mt7601u mt76_usb mt76 mcs7830 mac80211 libipw libertas kalmia iwlwifi ipt_REJECT huawei_cdc_ncm gspca_zc3xx gspca_vc032x gspca_tv8532 gspca_t613 gspca_sunplus gspca_stv06xx gspca_stk014 gspca_sq905c gspca_sq905 gspca_spca561 gspca_spca508Sun Dec 27 19:52:28 2020 kern.warn kernel: [125731.949934]  gspca_spca506 gspca_spca505 gspca_spca501 gspca_spca500 gspca_sonixj gspca_sonixb gspca_sn9c20x gspca_se401 gspca_pac7311 gspca_pac207 gspca_ov534_9 gspca_ov534 gspca_ov519 gspca_mr97310a gspca_mars gspca_main dm9601 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a ath9k_pci_owl_loader asix amdgpu xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_esp xt_ecn xt_dscp xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY xr_usb_serial_common wlcore_sdio wireguard videobuf2_v4l2 videobuf2_common via_velocity via_rhine usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm solos_pci snd_usbmidi_lib snd_intel_nhlt snd_hda_core slhc sky2 skge sis900 sis190 sch_cake rtl8150 regmap_core radeon r8712u(C) r8169 r8168 r8152 r8125 r6040Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.039168]  pegasus pcnet32 niu nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core macvlan lzo lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kvm_intel kvm kaweth irqbypass iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables i40e hso gpu_sched forcedeth exfat ethoc et131x e1000e e100 dmx_usb crc8 crc7 crc_itu_t crc_ccitt cpia2 cordic compat cdc_wdm cdc_acm brcmutil br_netfilter bnx2x bnx2 be2net atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs videobuf2_vmalloc videobuf2_memops videodev evdev i2c_dev spi_ks8995 siit cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_macSun Dec 27 19:52:28 2020 kern.warn kernel: [125732.128928]  ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink rtl8306 mvsw61xx ip17xx nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig nfsv4 nfsv3 nfsd nfs bonding 3c59x ip_gre gre snd_via82xx snd_intel8x0 snd_mpu401_uart ixgbevf ixgbe igbvf igb e1000 snd_ac97_codec ac97_bus ifb dummy mdio l2tp_netlink l2tp_core udp_tunnel ip6_udp_tunnel ipcomp6 xfrm6_tunnel esp6 ah6 xfrm4_tunnel ipcomp esp4 ah4 tunnel6 tunnel4 ip_tunnel hfcpci hfcmulti rpcsec_gss_krb5 auth_rpcgss oid_registry veth tun snd_rawmidi snd_seq_device snd_pcm_oss snd_pcm snd_timer snd_mixer_oss snd_hwdep snd_compress snd soundcore nbd mISDN_dsp l1oip mISDN_core xfrm_user xfrm_ipcomp af_key xfrm_algo vfat fat lockd sunrpc grace cifs dns_resolver dm_mirror dm_region_hash dm_log dm_crypt dm_mod dax br2684 atm bcma raid456 async_raid6_recov async_pqSun Dec 27 19:52:28 2020 kern.warn kernel: [125732.221255]  async_xor async_memcpy async_tx raid10 raid1 raid0 md_mod nls_utf8 nls_iso8859_1 nls_cp437 natsemi eeprom_93cx6 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha256_ssse3 sha256_generic libsha256 sha1_ssse3 sha1_generic seqiv jitterentropy_rng drbg michael_mic md5 md4 hmac ghash_generic gcm echainiv des_generic libdes deflate cts ctr ccm cbc authenc arc4 crypto_acompress uas sdhci_pltfm xhci_plat_hcd fsl_mph_dr_of ehci_platform ehci_fsl btrfs zstd_decompress zstd_compress xxhash xor raid6_pq lzo_decompress lzo_compress libcrc32c antfs tg3 b44 ssb ptp realtek broadcom bcm_phy_lib pps_core mii libphySun Dec 27 19:52:28 2020 kern.warn kernel: [125732.370706] CPU: 1 PID: 0 Comm: swapper/1 Tainted: G        WC        5.4.84 #0Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.378368] Hardware name: Default string Default string/Default string, BIOS 5.12 09/15/2019Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.387275] RIP: 0010:0xffffffffa189022aSun Dec 27 19:52:28 2020 kern.warn kernel: [125732.391446] Code: b8 89 41 08 48 3b 31 74 b0 8b 46 40 85 c0 74 2d 8d 78 01 48 8d 56 40 f0 0f b1 3a 75 11 48 89 f7 48 87 39 e8 98 d6 ef df eb 8f <0f> 0b eb 8b 85 c0 74 0b 8d 78 01 f0 0f b1 3a 74 e2 eb f1 0f 0b ebSun Dec 27 19:52:28 2020 kern.warn kernel: [125732.410800] RSP: 0018:ffffc9000003c9e0 EFLAGS: 00010286Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.416290] RAX: 0000000000000058 RBX: 0000000000000001 RCX: ffff88825b254dd8Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.423760] RDX: ffffc9000003ca30 RSI: ffff888261c56948 RDI: ffff888258607280Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.431240] RBP: ffffc9000003c9e0 R08: 0000000000000000 R09: ffffffffa0914480Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.438722] R10: ffff888265d2ca80 R11: 0000000000000020 R12: 0000000000000002Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.446219] R13: ffff888260a87100 R14: ffff888243c0da00 R15: ffffc9000003ca30Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.453707] FS:  0000000000000000(0000) GS:ffff888266d00000(0000) knlGS:0000000000000000Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.462149] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.468206] CR2: 000000000041b14f CR3: 0000000260bc0001 CR4: 00000000003606e0Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.475659] Call Trace:Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.478332]  <IRQ>Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.480561]  nf_hook_slow+0x42/0xc0Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.484269]  br_nf_hook_thresh+0x6f6/0x12a0 [br_netfilter]Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.490019]  ? br_nf_hook_thresh+0xd0/0x12a0 [br_netfilte [... cut ...] 20 kern.warn kernel: [125732.707436]  call_cpuidle+0x1e/0x40Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.711167]  do_idle+0xf4/0x1c0Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.714522]  cpu_startup_entry+0x18/0x20Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.718701]  start_secondary+0x145/0x170Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.722868]  secondary_startup_64+0xa4/0xb0Sun Dec 27 19:52:28 2020 kern.warn kernel: [125732.727307] ---[ end trace cd52dde72e8fd38b ]---
--

Dimitri2020007, 2020-12-27:
这个问题是sfe和flowoffload起冲突了

建议sfe和flowoffload二选一

我现在只用sfe

galaxyskyknight, 2020-12-27:
> 这个问题是sfe和flowoffload起冲突了
> 
> 建议sfe和flowoffload二选一
> 
> 我现在只用sfe

具体哪里操作？sfe和flowoffload是什么？

galaxyskyknight, 2020-12-27:
好像我的只有turbo sfe没有flowoffload哈。。。不过我有单线双拨倒是真的。但我关了SFE甚至关了Fullcone-NAT，这个日志还是大不停，而且跟你现象一样，偶尔会重启（watchcat已经关掉），丢包很高, 见我反应的另一个issue #6089. net Rx的softirq软中断很多，有drop和overrun.

Dimitri2020007, 2020-12-27:
make menuconfig>luci>application下保留或者取消luci-app-sfe/flowoffload

drop这个我是启动约1天左右会出现

这个留心下内核日志

不过我sfe和flowoffload二选一之后就没有刷过屏了

若你用的是x86取消掉这个就不会有问题

closed 2021-06-20 by coolsnowwolf