coolsnowwolf/lede #6116: openwrt都无法识别出512MB

nianququ22, 2020-12-27:
大佬你好，我有台华硕acrh17升级了512MB内存。但刷了很多版本的openwrt都无法识别出512MB，请问该怎么弄才能支持512MB，谢谢

AmadeusGhost, 2020-12-27:
估计需要换成 pbboot，然后看看 pbboot 能不能识别到你换的 512M 内存

jinglei207, 2020-12-27:
没看懂。
你说的是内存无法识别，还是能识别，但是无法识别到512mb？

AmadeusGhost, 2020-12-27:
后者

jinglei207, 2020-12-27:
> 后者

https://blog.csdn.net/soar0603/article/details/89531913
可以试试修改 rqm大小

LGA1150, 2020-12-28:
因为 dts 里写死了 RAM 大小
https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-rt-acrh17.dts#L27-L30
直接去掉就可以了，内核会自动检测
@coolsnowwolf @AmadeusGhost

nianququ22, 2020-12-28:
> 因为 dts 里写死了 RAM 大小
> https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-rt-acrh17.dts#L27-L30
> 
>
> 直接去掉就可以了，内核会自动检测
> @coolsnowwolf @AmadeusGhost
 您好，我按照您的指点删掉相关代码后编译出的固件并刷入，openwrt还是只能识别出256MB

[image]

AmadeusGhost, 2020-12-28:
引导换成 pbboot 试试，看看 pbboot 的页面能不能认到 512M 内存

closed 2020-12-31 by nianququ22