coolsnowwolf/lede #6117: 如何设置pppoe为wan口默认协议，并设置账号与密码？

axlexan, 2020-12-27:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
需要编译多个路由器的固件，需要将wan口设置为pppoe协议，且写入账号与密码
尝试过files大法，不行，试了两次会无法连接路由（不分配ip,自己手动设置ip ping路由也没反应,最后是刷回原厂固件才救回的)
A：

### (2) 路由器型号和固件版本

A：红米AC2100 & 新三

### (3) 详细日志

A：

LGA1150, 2020-12-28:
编辑`package/base-files/files/lib/functions/uci-defaults.sh`找到这一行
https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/package/base-files/files/lib/functions/uci-defaults.sh#L76
把`dhcp`改为`pppoe`

编辑`package/base-files/files/bin/config_generate`找到这两行
https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/package/base-files/files/bin/config_generate#L135-L136
把等号后面的`username` `password`分别替换为你的账号和密码

hyird, 2020-12-28:
> 编辑`package/base-files/files/lib/functions/uci-defaults.sh`找到这一行 [...]

改源码不推荐吧 还不如 在大雕的 default-settings 里面 写uci命令

zwillhill, 2020-12-29:
笨办法：将正常使用的network文件放到/package/base-files/files/etc/config目录下。

zxcvbxzb, 2020-12-29:
> > 编辑`package/base-files/files/lib/functions/uci-defaults.sh`找到这一行
> > https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/package/base-files/files/lib/functions/uci-defaults.sh#L76
> > 
> > 把`dhcp`对划线`pppoe`
> > 编辑`package/base-files/files/bin/config_generate`找到这两行
> > https://github.com/coolsnowwolf/lede/blob/f8ef6220d66faefd17f720ed8ed8806188ab60c7/package/base-files/files/bin/config_generate#L135-L136
> > 
> > 把等号后面的`username` `password`分别替换为你的账号和密码
> 
> 改原始码不推荐吧还不如在大雕的default-settings里面写uci命令

uci命令有参考吗，不会写啊

hyird, 2020-12-29:
> uci命令有参考吗，不会写啊

uci set network.wan.proto='pppoe'
uci set network.wan.username='账号'
uci set network.wan.password='密码'

试试

closed 2021-06-20 by coolsnowwolf

wangweitung, 2021-06-23:
> 
> 
> 笨办法：将正常使用的network文件放到/package/base-files/files/etc/config目录下。

我用files大法没编译成功。

这个方法好，用这个成功了。