coolsnowwolf/lede #6126: ksoftirqd进程吃掉全部cpu导致断流，求一个解决思路

LOVE2CMOL, 2020-12-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：ksoftirqd吃掉全部cpu导致断流，过段时间会自动恢复
A：随机发生 （存疑: 貌似是手机看b站的概率会高一丢丢？！ 或者频繁打开网页之类的？！）
A：已经找过很多资料了
A：内核 系统 日志之类的都没有打印奇怪的东西
A：尝试过增加neigh.default.gc_thresh
A：SFE已经关闭（恢复会快点）

### (2) 路由器型号和固件版本
固件版本 OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)
内核版本 5.4.84
### (3) 详细日志

A：
系统日志
[image]

内核日志
[image]

cat /proc/interrupts
[image]

top的话卡的时候连命令都输不了 没法看
附一张恢复过程的
[image]

LOVE2CMOL, 2020-12-28:
还有个小细节
就是 出现最频繁的时间是00:00-05:00 
已经排查过内网所有服务器的自动服务

minlang112, 2020-12-29:
ksoftirqd吃掉全部cpu，说明你的网卡已经处于极高的负载之下？

LOVE2CMOL, 2020-12-29:
> ksoftirqd吃掉全部cpu，说明你的网卡已经处于极高的负载之下？

但是问题是，这是虚拟网卡，负载能跑很高的，我另外尝试连接数压力测试，跑到16000左右，ksoftirqd也就2%，但是这个卡住能跑到炸机

LOVE2CMOL, 2020-12-29:
因为我安装的pve是6.3版本的，尝试更新qemu-ga(5.0.0-1 to 5.0.0-4) virtio-console-helper(5.0.0-1 to 5.0.0-4) 先作一个记录，有结果再反馈

思路：qemu-ga virtio-console-helper 这两玩意是在其他linux系统有带驱动，推断可能是由于驱动版本的问题导致的？！

LOVE2CMOL, 2020-12-29:
> 因为我安装的pve是6.3版本的，尝试更新qemu-ga(5.0.0-1 to 5.0.0-4) virtio-console-helper(5.0.0-1 to 5.0.0-4) 先作一个记录，有结果再反馈
> 
> 思路：qemu-ga virtio-console-helper 这两玩意是在其他linux系统有带驱动，推断可能是由于驱动版本的问题导致的？！

还是一样，该炸的还是炸

LOVE2CMOL, 2020-12-31:
排查出来了FullCone-NAT的问题

具体诱发方法 当有效连接数达到某个阈值（我的是15000左右），就会开始拥堵ksoftirqd上升，一直飙到2万+(最高看见的数值，已经卡住了)，直到负载爆表

关闭后测试有效连接数连接数到达3w+都没问题，负载在5%-45%波动

我是PVE下跑的op，网卡是virtio模式（E1000下也是这种情况）

在此给各位遇到同样问题的朋友指明下方向

closed 2020-12-31 by LOVE2CMOL

568875138, 2021-01-29:
遇到同样的问题了，但是也没找出原因。

xydche, 2021-12-24:
我昨天编译的版本也出现了这个问题，开机差不多5个小时cpu就会满，路由网页都打开非常慢，无奈回退了

rheatin, 2023-08-30:
> > 排查出来了FullCone-NAT的问题
> > 具体诱发方法 当有效连接数达到某个阈值（我的是15000左右），就会开始拥堵ksoftirqd上升，一直飙到2万+(最高看见的数值，已经卡住了)，直到负载爆表
> > 关闭后测试有效连接数连接数到达3w+都没问题，负载在5%-45%波动
> > 我是PVE下跑的op，网卡是virtio模式（E1000下也是这种情况）
> > 在此给各位遇到同样问题的朋友指明下方向
> 
> 我是直通物理网卡，跑网心云连接数到1.2w的时候单核因为ksoftirqd满载，查看软中断的亲和性都是全核的，但是因为单核满载导致ssh都连不上、网页打不开。看到你的这个方法关掉fullconenat，现在即使是2w的连接都没问题，单核也没有满载了。

好兄弟们，有解决的办法吗，我也是关闭了fullcone就好了，但是fullcone要用的呀

GreatMichaelLee, 2023-10-14:
> 好兄弟们，有解决的办法吗，我也是关闭了fullcone就好了，但是fullcone要用的呀

没啥好办法，这问题困扰我三年了，一直在。

rheatin, 2023-10-16:
> > > > 排查出来了FullCone-NAT的问题
> > > > 具体诱发方法 当有效连接数达到某个阈值（我的是15000左右），就会开始拥堵ksoftirqd上升，一直飙到2万+(最高看见的数值，已经卡住了)，直到负载爆表
> > > > 关闭后测试有效连接数连接数到达3w+都没问题，负载在5%-45%波动
> > > > 我是PVE下跑的op，网卡是virtio模式（E1000下也是这种情况）
> > > > 在此给各位遇到同样问题的朋友指明下方向
> > > 
> > > 
> > > 我是直通物理网卡，跑网心云连接数到1.2w的时候单核因为ksoftirqd满载，查看软中断的亲和性都是全核的，但是因为单核满载导致ssh都连不上、网页打不开。看到你的这个方法关掉fullconenat，现在即使是2w的连接都没问题，单核也没有满载了。
> > 
> > 
> > 好兄弟们，有解决的办法吗，我也是关闭了fullcone就好了，但是fullcone要用的呀
> 
> 没啥好办法，这问题困扰我三年了，一直在，也没人定位和重视

诶，我也是手贱，没事升级pve和openwrt（之前稳定两年了）

rheatin, 2023-11-14:
> 没啥好办法，这问题困扰我三年了，一直在，也没人定位和重视

这位大神编译的openwrt不知道为什么，fullcone正常的 https://github.com/stupidloud/nanopi-openwrt