coolsnowwolf/lede #6132: 更改固件大小，依然提示error: images are too big by 10219168 bytes

ljm1447669873, 2020-12-29:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：通过修改target/linux/ar71xx/image的tiny-tp-link.mk文件，将4m改成16m，但是无法生成sysupgrade.bin，在初次编译前更改，依然无法生成，是因为是tiny固件，还需要更改其他地方吗
`define Device/tl-wdr3320-v2
  $(Device/tplink-16mlzma)
  DEVICE_TITLE := TP-LINK TL-WDR3320 v2
  DEVICE_PACKAGES := kmod-usb2 kmod-usb-ledtrig-usbport
  BOARDNAME := TL-WDR3320-v2
  DEVICE_PROFILE := TLWDR3320V2
  TPLINK_HWID := 0x33200002
  TPLINK_HEADER_VERSION := 2
endef
TARGET_DEVICES += tl-wdr3320-v2`

### (2) 路由器型号和固件版本

A： TL-WDR3320-v2   最新版

### (3) 详细日志

A：`/home/user/lede/staging_dir/host/bin/mktplinkfw -H 0x33200002 -W 0x1 -F 16Mlzma -N OpenWrt -V r2998-f8ef6220d -m 2 -k /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tl-wdr3320-v2-kernel.bin -r /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin -o /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin.new -j -X 0x40000 -a 0x4  -s && mv /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin.new /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin || rm -f /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin
_[mktplinkfw] rootfs offset aligned to 0x1350208
[mktplinkfw] rootfs offset aligned to 0x1350208
[mktplinkfw] *** error: images are too big by 10377286 bytes
[mktplinkfw] *** error: images are too big by 10377286 bytes_
cp /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-factory.bin /home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-factory.bin
cp /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin /home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin
cp: cannot stat '/home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-factory.bin': No such file or directory
Makefile:69: recipe for target '/home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-factory.bin' failed
make[5]: [/home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-factory.bin] Error 1 (ignored)
cp: cannot stat '/home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/tmp/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin': No such file or directory
Makefile:69: recipe for target '/home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin' failed
make[5]: [/home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2-squashfs-sysupgrade.bin] Error 1 (ignored)
cp /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs-raw
cp /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs-64k
/home/user/lede/staging_dir/host/bin/padjffs2 /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs-64k 64
padding image to 01820000
/home/user/lede/staging_dir/host/bin/padjffs2 /home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs 4 8 16 64 128 256
padding image to 0181c000
padding image to 01820000
padding image to 01840000
dd if=/home/user/lede/build_dir/target-mips_24kc_musl/linux-ar71xx_tiny/root.squashfs of=/home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-root.squashfs bs=128k conv=sync
194+1 records in
195+0 records out
25559040 bytes (26 MB, 24 MiB) copied, 0.0163682 s, 1.6 GB/s
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/user/lede/build_dir/target-mips_24kc_musl/root.orig-ar71xx TMPDIR=/home/user/lede/build_dir/target-mips_24kc_musl/root.orig-ar71xx/tmp /home/user/lede/staging_dir/host/bin/opkg --offline-root /home/user/lede/build_dir/target-mips_24kc_musl/root.orig-ar71xx --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch mips_24kc:200 list-installed > /home/user/lede/bin/targets/ar71xx/tiny/openwrt-ar71xx-tiny-tl-wdr3320-v2.manifest
make[5]: Leaving directory '/home/user/lede/target/linux/ar71xx/image'
make[4]: Leaving directory '/home/user/lede/target/linux/ar71xx'
make[3]: Leaving directory '/home/user/lede/target/linux'
time: target/linux/install#502.68#384.10#166.49
make[2]: Leaving directory '/home/user/lede'
export MAKEFLAGS= ;make -w -r package/index
make[2]: Entering directory '/home/user/lede'`

xtwz, 2020-12-30:
虚拟机，修改配置 同样也会提示固件大小错误！

sxx1314, 2020-12-30:
你的squashfs都26M了怎么可能给你生产bin。。。自行删减组件。

closed 2020-12-30 by ljm1447669873