coolsnowwolf/lede #6137: 申请加入GL-inet的mt1300型号路由器支持

dreamcmi, 2020-12-30:
手里有一批mt1300的路由器，但是lede仓库内尚未支持该型号路由器，申请加入一下。
该路由器已经在openwrt官方源码中的mt7621下支持，请作者大大加入一下吧，很喜欢lede源码中的功能。
以下是该路由器型号参数
Model | GL-MT1300
-- | --
CPU | MTK MT7621A, Dual-Core@880MHz
Memory | DDR3L 256MB
Storage | 32MB Nor Flash
Interfaces | 1 WAN, 2 LANs 1 USB3.0, 1 Reset Button, 1 TF Card Slot (512GB Max.)
Frequency | 2.4GHz / 5GHz
Transmission Rate | 400Mbps(2.4GHz) + 867Mbps(5GHz)
Max. Tx Power | 20dBm
Protocol | 802.11 a/b/g/n/ac
External Drive Format Support | FAT32/NTFS/EXT4/EXT3/EXT2
Wi-Fi Antennas | 2 X Undetachable Antennas
Ethernet Port | 10/100/1000M
Power Input | 5V/3A Type-C
Power Consumption | <8W (to be updated)
Dimension, Weight | 118mmX85mmX30mm, 184g

sdf8057, 2020-12-31:
你应该给开发者寄个机器适配

dreamcmi, 2020-12-31:
> 你应该给开发者寄个机器适配

我自己移植了，就是想让lean大加入仓库

hzjnet, 2020-12-31:
> > 你应该给开发者寄个机器适配
> 
> 我自己移植了，就是想让lean大加入仓库

把你的适配提交给他就是三

minlang112, 2020-12-31:
> > 你应该给开发者寄个机器适配
> 
> 我自己移植了，就是想让lean大加入仓库

直接pull requests就好拉

closed 2021-01-02 by dreamcmi

angelfreedomv, 2021-02-19:
> > 你应该给开发者寄个机器适配
> 
> 我自己移植了，就是想让lean大加入仓库

移植的pull在哪？