coolsnowwolf/lede #6138: 请问nft-qos不生效是什么原因

WROIATE, 2020-12-30:
### (1) 具体问题
nft-qos和sqm无效
已关闭bbr以及转发加速
nft-qos静态动态均无效

### (2) 路由器型号和固件版本

Phicomm N1
ARMv8 Processor rev 4 (v8l) x 4
OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

sqm日志
```
start-sqm: Log for interface eth0.2: Thu Dec 31 01:45:27 CST 2020

Thu Dec 31 01:45:27 CST 2020: Starting.
Starting SQM script: piece_of_cake.qos on eth0.2, in: 0 Kbps, out: 0 Kbps
fn_exists: function candidate name: sqm_start
fn_exists: TYPE_OUTPUT: sqm_start: not found
fn_exists: return value: 1
Using generic sqm_start_default function.
fn_exists: function candidate name: sqm_prepare_script
fn_exists: TYPE_OUTPUT: sqm_prepare_script is a function
fn_exists: return value: 0
sqm_start_default: starting sqm_prepare_script
insmod: ERROR: could not load module act_ipt: No such file or directory
insmod: ERROR: could not load module sch_ingress: No such file or directory
cmd_wrapper: COMMAND: /sbin/ip link add name TMP_IFB_4_SQM type ifb
cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
cmd_wrapper: COMMAND: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
QDISC cake is useable.
cmd_wrapper: COMMAND: /sbin/ip link set dev TMP_IFB_4_SQM down
cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev TMP_IFB_4_SQM down
cmd_wrapper: COMMAND: /sbin/ip link delete TMP_IFB_4_SQM type ifb
cmd_wrapper: ip: SUCCESS: /sbin/ip link delete TMP_IFB_4_SQM type ifb
insmod: ERROR: could not load module act_ipt: No such file or directory
insmod: ERROR: could not load module sch_ingress: No such file or directory
cmd_wrapper: COMMAND: /sbin/ip link add name TMP_IFB_4_SQM type ifb
cmd_wrapper: ip: SUCCESS: /sbin/ip link add name TMP_IFB_4_SQM type ifb
cmd_wrapper: COMMAND: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
cmd_wrapper: tc: SUCCESS: /usr/sbin/tc qdisc replace dev TMP_IFB_4_SQM root cake
QDISC cake is useable.
cmd_wrapper: COMMAND: /sbin/ip link set dev TMP_IFB_4_SQM down
cmd_wrapper: ip: SUCCESS: /sbin/ip link set dev TMP_IFB_4_SQM down
cmd_wrapper: COMMAND: /sbin/ip link delete TMP_IFB_4_SQM type ifb
cmd_wrapper: ip: SUCCESS: /sbin/ip link delete TMP_IFB_4_SQM type ifb
sqm_start_default: Starting piece_of_cake.qos
ifb associated with interface eth0.2:
Currently no ifb is associated with eth0.2, this is normal during starting of the sqm system.
cmd_wrapper: COMMAND: /sbin/ip link add name ifb4eth0.2 type ifb
cmd_wrapper: ip: SUCCESS: /sbin/ip link add name ifb4eth0.2 type ifb
sqm_start_default: egress shaping deactivated
cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev eth0.2 root
cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev eth0.2 root
cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
sqm_start_default: ingress shaping deactivated
cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev ifb4eth0.2 root
cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev ifb4eth0.2 root
cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
cmd_wrapper: tc: invocation silenced by request, FAILURE either expected or acceptable.
cmd_wrapper: COMMAND: /usr/sbin/tc qdisc del dev eth0.2 ingress
cmd_wrapper: tc: FAILURE (2): /usr/sbin/tc qdisc del dev eth0.2 ingress
cmd_wrapper: tc: LAST ERROR: RTNETLINK answers: No such file or directory
piece_of_cake.qos was started on eth0.2 successfully
```

nft-qos日志
```
Error: Could not process rule: Resource busy
add chain inet nft-qos-monitor upload { type filter hook postrouting priority 0; }
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Error: Could not process rule: Resource busy
add chain inet nft-qos-monitor download { type filter hook prerouting priority 0; }
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
sh: dynamic: unknown operand
```

swxk521, 2020-12-31:
这是我用的，限制上传下载，没有问题。
https://github.com/swxk521/luci-app-eqos

WROIATE, 2020-12-31:
> 这是我用的，限制上传下载，没有问题。
> https://github.com/swxk521/luci-app-eqos
> 
> [image]

在有nat6的情况下是否能生效？我这测试无效

swxk521, 2021-01-01:
> > 这是我用的，限制上传下载，没有问题。
> > https://github.com/swxk521/luci-app-eqos
> > [image]
> 
> ~在有nat6的情况下是否能生效？我这测试无效~ 貌似原因是接口脚本里定死了是br-lan
我没开启IPV6，我也是小白，IPV4正常

jxph1123, 2021-01-05:
> > > 这是我用的，限制上传下载，没有问题。
> > > https://github.com/swxk521/luci-app-eqos
> > > [image]
> > 
> > 
> > ~在有nat6的情况下是否能生效？我这测试无效~ 貌似原因是接口脚本里定死了是br-lan
> > 我没开启IPV6，我也是小白，IPV4正常
开了Flow Offloading还能用吗？

WROIATE, 2021-05-08:
更新一下，ipv6没用是因为tc规则脚本里只写了v4的规则，若是需要v6可以手动添加v6规则，但是比较复杂，如果有大佬知道怎么使用mac来匹配的话就能一并解决了

closed 2021-05-08 by WROIATE