coolsnowwolf/lede #6141: 编译固件时出现大量undefined错误

mervynsword, 2020-12-31:
为R4S编译固件，基础包使用friendlyarm官方提供的，在 friendlywrt 下修改 feeds.conf.default 文件，添加对应的依赖库
将L大的 tools/upx 和 tools/ucl 拷贝到的 tools 目录下下
修改 tools 下的 Makefile 添加两行
```
tools-y += ucl upx
$(curdir)/upx/compile := $(curdir)/ucl/compile
```
linux下编译所需软件已经装全，且全部upgrade过
但是编译依然无限失败。错误提示如下

```
Building targets
google.golang.org/protobuf/internal/flags
golang.org/x/sys/internal/unsafeheader
go build golang.org/x/sys/internal/unsafeheader: no non-test Go files in /home/ubuntu/friendlywrt-rk3399/friendlywrt/dl/go-mod-cache/golang.org/x/sys@v0.0.0-20201107080550-4d91cf3a1aaf/internal/unsafeheader
github.com/lucas-clemente/quic-go/internal/logutils
go build github.com/lucas-clemente/quic-go/internal/logutils: no non-test Go files in /home/ubuntu/friendlywrt-rk3399/friendlywrt/dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/logutils
# google.golang.org/protobuf/internal/flags
../../../../../dl/go-mod-cache/google.golang.org/protobuf@v1.25.0/internal/flags/flags.go:17:21: undefined: protoLegacy
golang.org/x/crypto/poly1305
golang.org/x/crypto/chacha20
golang.org/x/crypto/sha3
golang.org/x/text/secure/bidirule
github.com/seiflotfy/cuckoofilter
github.com/lucas-clemente/quic-go/internal/protocol
# golang.org/x/crypto/poly1305
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/poly1305/poly1305.go:64:2: undefined: mac
# golang.org/x/text/secure/bidirule
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/secure/bidirule/bidirule.go:175:10: t.isFinal undefined (type Transformer has no field or method isFinal)
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/secure/bidirule/bidirule.go:184:10: t.isFinal undefined (type Transformer has no field or method isFinal)
../../../../../dl/go-mod-cache/golang.org/x/text@v0.3.3/secure/bidirule/bidirule.go:240:9: t.isFinal undefined (type *Transformer has no field or method isFinal)
# golang.org/x/crypto/chacha20
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:46:7: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:198:22: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:227:30: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:229:4: s.xorKeyStreamBlocks undefined (type *Cipher has no field or method xorKeyStreamBlocks)
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:235:23: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:237:12: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:239:16: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:249:12: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:251:4: s.xorKeyStreamBlocks undefined (type *Cipher has no field or method xorKeyStreamBlocks)
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:252:11: undefined: bufSize
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go:252:11: too many errors
golang.org/x/sys/cpu
golang.org/x/crypto/curve25519
# golang.org/x/crypto/sha3
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:19:10: undefined: new224Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:29:10: undefined: new256Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:39:10: undefined: new384Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/hashes.go:49:10: undefined: new512Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/sha3.go:43:10: undefined: storageBuf
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/sha3.go:85:3: undefined: xorIn
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/sha3.go:93:3: undefined: copyOut
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/sha3.go:121:2: undefined: copyOut
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/sha3.go:138:4: undefined: xorIn
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/shake.go:119:10: undefined: newShake128Asm
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/sha3/shake.go:119:10: too many errors
# github.com/seiflotfy/cuckoofilter
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:13:14: undefined: bucket
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:22:45: undefined: bucketSize
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:26:20: undefined: bucket
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:82:18: undefined: bucketSize
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:117:40: undefined: bucketSize
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:130:16: undefined: bucketSize
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:131:73: undefined: bucketSize
../../../../../dl/go-mod-cache/github.com/seiflotfy/cuckoofilter@v0.0.0-20201009151232-afb285a456ab/cuckoofilter.go:133:20: undefined: bucket
github.com/gorilla/websocket
# github.com/lucas-clemente/quic-go/internal/protocol
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:29:52: undefined: Perspective
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:37:8: undefined: PerspectiveClient
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:39:8: undefined: PerspectiveServer
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:44:8: undefined: PerspectiveClient
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:46:8: undefined: PerspectiveServer
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:57:33: undefined: Perspective
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:59:10: undefined: PerspectiveClient
../../../../../dl/go-mod-cache/github.com/lucas-clemente/quic-go@v0.19.2/internal/protocol/stream.go:61:9: undefined: PerspectiveServer
# golang.org/x/sys/cpu
../../../../../dl/go-mod-cache/golang.org/x/sys@v0.0.0-20201107080550-4d91cf3a1aaf/cpu/cpu_arm64.go:65:28: undefined: getisar0
../../../../../dl/go-mod-cache/golang.org/x/sys@v0.0.0-20201107080550-4d91cf3a1aaf/cpu/cpu_arm64.go:65:40: undefined: getisar1
../../../../../dl/go-mod-cache/golang.org/x/sys@v0.0.0-20201107080550-4d91cf3a1aaf/cpu/cpu_arm64.go:65:52: undefined: getpfr0
# golang.org/x/crypto/curve25519
../../../../../dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/curve25519/curve25519.go:21:2: undefined: scalarMult
# github.com/gorilla/websocket
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/client.go:308:10: undefined: cloneTLSConfig
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/client.go:317:10: undefined: doHandshakeWithTrace
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/conn.go:393:10: c.writeBufs undefined (type *Conn has no field or method writeBufs)

Makefile:151: recipe for target '/home/ubuntu/friendlywrt-rk3399/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.33.0/.built' failed
make[3]: *** [/home/ubuntu/friendlywrt-rk3399/friendlywrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-core-4.33.0/.built] Error 2
make[3]: Leaving directory '/home/ubuntu/friendlywrt-rk3399/friendlywrt/feeds/passwall/v2ray'
time: package/feeds/passwall/v2ray/compile#1.20#0.46#0.46
package/Makefile:111: recipe for target 'package/feeds/passwall/v2ray/compile' failed
make[2]: *** [package/feeds/passwall/v2ray/compile] Error 2
```
还望有大神能解惑

coolsnowwolf, 2020-12-31:
你需要问下  friendlywrt  那边了

mervynsword, 2020-12-31:
> 你需要问下 friendlywrt 那边了

仔细检查了源码，好像是内核部分没有集成openssl？不是很懂编译，我猜测这些undefined错误是因为在编译过程中找不到对应的值，而且都是出现在加密功能部分，请问这里是依赖哪一部分进行编译的？可能的话我打算手动修复相应的依赖。

mervynsword, 2020-12-31:
比方说 `/dl/go-mod-cache/golang.org/x/crypto@v0.0.0-20201016220609-9e8e0b390897/chacha20/chacha_generic.go` 文件中，首先是
```
import (
        "crypto/cipher"
        "encoding/binary"
        "errors"
        "math/bits"

        "golang.org/x/crypto/internal/subtle"
)
```
这几个导入是从哪里导入的？就是说绝对路径

然后后面有如下代码

```
type Cipher struct {
        // The ChaCha20 state is 16 words: 4 constant, 8 of key, 1 of counter
        // (incremented after each block), and 3 of nonce.
        key     [8]uint32
        counter uint32
        nonce   [3]uint32

        // The last len bytes of buf are leftover key stream bytes from the previous
        // XORKeyStream invocation. The size of buf depends on how many blocks are
        // computed at a time by xorKeyStreamBlocks.
        buf [bufSize]byte
        len int

        // overflow is set when the counter overflowed, no more blocks can be
        // generated, and the next XORKeyStream call should panic.
        overflow bool

        // The counter-independent results of the first round are cached after they
        // are computed the first time.
        precompDone      bool
        p1, p5, p9, p13  uint32
        p2, p6, p10, p14 uint32
        p3, p7, p11, p15 uint32
}
```

其中 `buf [bufSize]byte` 就是第一个undefined错误，这里的注释说bufSize取决于xorKeyStreamBlocks，但是这个xorKeyStreamBlocks我也没找到……

mervynsword, 2020-12-31:
解决了，本问题终结，原因应该是golang相关组件没有更新到最新版，在更新了最新版feeds之后，一次性编译成功

closed 2020-12-31 by mervynsword