coolsnowwolf/lede #6144: rtl8156和rtl8156B 不能驱动，如绿联，比亚兹2.5G usb网卡

liuquan321, 2021-01-01:
1月1日最新源码编译的x86固件无法驱动rtl8156和8156b，具体表现为lsusb可以看到设备，但是没有生成eth设备。
`Bus 001 Device 002: ID 0781:5571 SanDisk Corp. Cruzer Fit
Bus 002 Device 003: ID 0bda:8156 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 05e3:0608 Genesys Logic, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub`

看上半年的issue说是因为和千兆的网卡有些冲突暂时关闭了，半年下来8156b的usb网卡性价比越来越高，京东自营已经跌到百元上下，完全可以替代千兆网卡了，是否可以考虑使用螃蟹10月中旬的新8152驱动或给出互斥的两个可选项呢？

liuquan321, 2021-01-02:
通过patch方式用螃蟹官方驱动替换内核自带驱动成功驱动8156b，大概思路如下
```
tar -xvf linux-5.4.85.tar.xz
cp linux-5.4.85_new
tar -xvf linux-5.4.85.tar.xz
tar -zxvf r8152.53.56-2.14.0.tar
cp -a r8152.53.56-2.14.0/r8152-2.14.0/50-usb-realtek-net.rules linux-5.4.85_new/drivers/net/usb/
cp -a r8152.53.56-2.14.0/r8152-2.14.0/compatibility.h linux-5.4.85_new/drivers/net/usb/
cp -a r8152.53.56-2.14.0/r8152-2.14.0/r8152.c linux-5.4.85_new/drivers/net/usb/

diff -uparN linux-5.4.85/* linux-5.4.85_new/* >988-r8152-2.14.0.patch

cp 988-r8152-2.14.0.patch ../lean/R20/lede/target/linux/generic/pending-5.4/
cd ../lean/R20/lede/ 
make
```

helianling, 2021-01-02:
8156可以参考8125的相同思路，另外最新的8125B驱动参数改一下可以支持RSS的，不支持8156支不支持。
[image]

liuquan321, 2021-01-04:
> 8156可以参考8125的相同思路，另外最新的8125B驱动参数改一下可以支持RSS的，不知道8156支不支持。
> [image]

我也参考过这个issue，两者稍有区别，8152的源码默认不在lede源码中而在linux kernel中，这导致两者虽然思路相同但是实际操作差异较大，最终参考 https://forum.openwrt.org/t/2-5g-usb-ethernet-kmod-usb-net-rtl8152/72154/57 完成了编译时替换。目前这个方法存在较大弊端内核版本更新需要重新生成patch。

liuquan321, 2021-01-04:
> 8156可以参考8125的相同思路，另外最新的8125B驱动参数改一下可以支持RSS的，不知道8156支不支持。
> [image]

closed 2021-01-09 by liuquan321

longdahai, 2021-01-13:
期待集成

yi-ge, 2021-02-21:
非常期待集成

one02free, 2021-04-11:
> 通过patch方式用螃蟹官方驱动替换内核自带驱动成功驱动8156b，大概思路如下
> 
> ```
> tar -xvf linux-5.4.85.tar.xz
> cp linux-5.4.85_new
> tar -xvf linux-5.4.85.tar.xz
> tar -zxvf r8152.53.56-2.14.0.tar
> cp -a r8152.53.56-2.14.0/r8152-2.14.0/50-usb-realtek-net.rules linux-5.4.85_new/drivers/net/usb/
> cp -a r8152.53.56-2.14.0/r8152-2.14.0/compatibility.h linux-5.4.85_new/drivers/net/usb/
> cp -a r8152.53.56-2.14.0/r8152-2.14.0/r8152.c linux-5.4.85_new/drivers/net/usb/
> 
> diff -uparN linux-5.4.85/* linux-5.4.85_new/* >988-r8152-2.14.0.patch
> 
> cp 988-r8152-2.14.0.patch ../lean/R20/lede/target/linux/generic/pending-5.4/
> cd ../lean/R20/lede/ 
> make
> ```
> 
> 方式比较low，有更优雅实现方式的勿忘告知。。。
> 
> 参考: https://forum.openwrt.org/t/2-5g-usb-ethernet-kmod-usb-net-rtl8152/72154/57

你好，我按照你上述提供的方法基于5.4.109生成了patch包，最后要怎样才能编译到Openwrt上？是不是把patch文件放进*/target/linux/generic/pending-5.4/后正常编译就可以了？还需要其他操作吗？

liuquan321, 2021-04-11:
> 你好，我按照你上述提供的方法基于5.4.109生成了patch包，最后要怎样才能编译到Openwrt上？是不是把patch文件放进*/target/linux/generic/pending-5.4/后正常编译就可以了？还需要其他操作吗？

编译选项中选择8152就可以，新版源码中已经集成了8156驱动不用这么折腾了。

one02free, 2021-04-14:
> > > 通过patch方式用螃蟹官方驱动替换内核自带驱动成功驱动8156b，大概思路如下
> > > ```
> > > tar -xvf linux-5.4.85.tar.xz
> > > cp linux-5.4.85_new
> > > tar -xvf linux-5.4.85.tar.xz
> > > tar -zxvf r8152.53.56-2.14.0.tar
> > > cp -a r8152.53.56-2.14.0/r8152-2.14.0/50-usb-realtek-net.rules linux-5.4.85_new/drivers/net/usb/
> > > cp -a r8152.53.56-2.14.0/r8152-2.14.0/compatibility.h linux-5.4.85_new/drivers/net/usb/
> > > cp -a r8152.53.56-2.14.0/r8152-2.14.0/r8152.c linux-5.4.85_new/drivers/net/usb/
> > > 
> > > diff -uparN linux-5.4.85/* linux-5.4.85_new/* >988-r8152-2.14.0.patch
> > > 
> > > cp 988-r8152-2.14.0.patch ../lean/R20/lede/target/linux/generic/pending-5.4/
> > > cd ../lean/R20/lede/ 
> > > make
> > > ```
> > > 
> > > 
> > > 方式比较low，有更优雅实现方式的勿忘告知。。。
> > > 参考: https://forum.openwrt.org/t/2-5g-usb-ethernet-kmod-usb-net-rtl8152/72154/57
> > 
> > 
> > 你好，我按照你上述提供的方法基于5.4.109生成了patch包，最后要怎样才能编译到Openwrt上？是不是把patch文件放进*/target/linux/generic/pending-5.4/后正常编译就可以了？还需要其他操作吗？
> 
> 编译选项中取消kmod-usb-net-rtl8152勾选就可以。
> 其实新版源码中已经集成了8156驱动不用这么折腾了，直接如下操作：
> 编译选项中取消`kmod-usb-net-rtl8152`
> 勾选`kmod-usb-net-rtl8152-vendor`
> 编译`make`

你好，我按照你的方法选中kmod-usb-net-rtl8152-vendor编译了新包，但是接口处并没有显示【2500Mb/s全双工】这样的字眼，请问是正常的吗？有什么方法可以检查2.5G正常被识别和使用了呢？

liuquan321, 2021-05-26:
> 你好，我按照你的方法选中kmod-usb-net-rtl8152-vendor编译了新包，但是接口处并没有显示【2500Mb/s全双工】这样的字眼，请问是正常的吗？有什么方法可以检查2.5G正常被识别和使用了呢？

网卡连上2.5g设备后ethtool 看下设备的实际工作速率