coolsnowwolf/lede #6147: 挂载的ext4硬盘，使用一段时间后变为只读，无法写入数据

zfh135072, 2021-01-02:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：挂载的ext4格式硬盘，重启Openwrt后的一段时间是可以正常读写的，然后一段时间没操作后，硬盘就变成只读了，无法写入数据

### (2) 路由器型号和固件版本

A：
路由型号：Intel(R) Core(TM) i5-7200U CPU @ 2.50GHz : 1 Core 4 Thread (CpuMark : 68552.821377 Scores)
固件版本：OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

[    0.000000] Linux version 5.4.85 (runner@fv-az32-410) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r2999-81a53ead5)) #0 SMP Tue Dec 29 14:28:04 2020
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz root=PARTUUID=f008e9f3-02 rootwait console=tty0 console=ttyS0,115200n8 noinitrd
[    0.000000] Disabled fast string operations
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x008: 'MPX bounds registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x010: 'MPX CSR'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: xstate_offset[3]:  832, xstate_sizes[3]:   64
[    0.000000] x86/fpu: xstate_offset[4]:  896, xstate_sizes[4]:   64
[    0.000000] x86/fpu: Enabled xstate features 0x1f, context size is 960 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ebff] usable
[    0.000000] BIOS-e820: [mem 0x000000000009ec00-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000000dc000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000007fedffff] usable
[    0.000000] BIOS-e820: [mem 0x000000007fee0000-0x000000007fefefff] ACPI data
[    0.000000] BIOS-e820: [mem 0x000000007feff000-0x000000007fefffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000007ff00000-0x000000007fffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000f0000000-0x00000000f7ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fec00000-0x00000000fec0ffff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fee00000-0x00000000fee00fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fffe0000-0x00000000ffffffff] reserved
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] SMBIOS 2.7 present.
[    0.000000] DMI: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 12/12/2018
[    0.000000] vmware: hypercall mode: 0x00
[    0.000000] Hypervisor detected: VMware
[    0.000000] vmware: TSC freq read from hypervisor : 2712.000 MHz
[    0.000000] vmware: Host bus clock speed read from hypervisor : 66000000 Hz
[    0.000000] vmware: using sched offset of 2404946076 ns
[    0.000001] tsc: Detected 2712.000 MHz processor
[    0.003829] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.003830] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.003833] last_pfn = 0x80000 max_arch_pfn = 0x400000000
[    0.003847] MTRR default type: uncachable
[    0.003847] MTRR fixed ranges enabled:
[    0.003848]   00000-9FFFF write-back
[    0.003849]   A0000-BFFFF uncachable
[    0.003849]   C0000-CBFFF write-protect
[    0.003849]   CC000-EFFFF uncachable
[    0.003850]   F0000-FFFFF write-protect
[    0.003850] MTRR variable ranges enabled:
[    0.003851]   0 base 00000000000 mask 7FF00000000 write-back
[    0.003852]   1 base 000C0000000 mask 7FFC0000000 uncachable
[    0.003852]   2 disabled
[    0.003853]   3 disabled
[    0.003853]   4 disabled
[... 1308 lines cut ...]
[21623.189975] Buffer I/O error on device sdb1, logical block 96190496
[21623.190121] Buffer I/O error on device sdb1, logical block 96190497
[21623.190261] Buffer I/O error on device sdb1, logical block 96190498
[21623.190400] Buffer I/O error on device sdb1, logical block 96190499
[21623.190539] Buffer I/O error on device sdb1, logical block 96190500
[21623.190678] Buffer I/O error on device sdb1, logical block 96190501
[21623.190817] Buffer I/O error on device sdb1, logical block 96190502
[21623.190957] Buffer I/O error on device sdb1, logical block 96190503
[21623.191097] Buffer I/O error on device sdb1, logical block 96190504
[21623.191237] Buffer I/O error on device sdb1, logical block 96190505
[21628.543742] sd 2:0:1:0: [sdb] tag#53 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[21628.543990] sd 2:0:1:0: [sdb] tag#53 Sense Key : 0x4 [current] 
[21628.544126] sd 2:0:1:0: [sdb] tag#53 ASC=0x44 ASCQ=0x0 
[21628.544246] sd 2:0:1:0: [sdb] tag#53 CDB: opcode=0x2a 2a 00 25 87 d9 10 00 00 40 00
[21628.544418] blk_update_request: critical target error, dev sdb, sector 629659920 op 0x1:(WRITE) flags 0x800 phys_seg 8 prio class 0
[21628.545380] Aborting journal on device sdb1-8.
[23423.106966] EXT4-fs error (device sdb1): ext4_journal_check_start:61: Detected aborted journal
[23423.107180] EXT4-fs (sdb1): Remounting filesystem read-only
[23431.141451] sd 2:0:1:0: [sdb] tag#7 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
[23431.141666] sd 2:0:1:0: [sdb] tag#7 Sense Key : 0x4 [current] 
[23431.141808] sd 2:0:1:0: [sdb] tag#7 ASC=0x44 ASCQ=0x0 
[23431.141935] sd 2:0:1:0: [sdb] tag#7 CDB: opcode=0x2a 2a 00 00 00 08 00 00 00 08 00
[23431.142107] blk_update_request: critical target error, dev sdb, sector 2048 op 0x1:(WRITE) flags 0x800 phys_seg 1 prio class 0
[23431.142359] Buffer I/O error on dev sdb1, logical block 0, lost sync page write
[23431.142542] EXT4-fs (sdb1): I/O error while writing superblock

caicaicai21, 2021-01-02:
这日志最后不是写着设备I/O错误吗？可能硬盘存在坏道或者供电不足吧，可以先尝试修复一下磁盘分区，如果还是会，可能是硬盘快要坏了。

zfh135072, 2021-01-02:
> 这日志最后不是写着设备I/O错误吗？可能硬盘存在坏道或者供电不足吧，可以先尝试修复一下磁盘分区，如果还是会，可能是硬盘快要坏了。

但是重启后又会好，用一段时间又会出现无法写的问题，这样也有可能是硬盘自身的问题吗，不大理解

Neverends4, 2021-01-02:
我遇到过，硬盘有坏道。可以插电脑上扫描下。

zfh135072, 2021-01-02:
> 我遇到过，硬盘有坏道。可以插电脑上扫描下。

好的，我试试，谢谢！

caicaicai21, 2021-01-03:
> > 这日志最后不是写着设备I/O错误吗？可能硬盘存在坏道或者供电不足吧，可以先尝试修复一下磁盘分区，如果还是会，可能是硬盘快要坏了。
> 
> 但是重启后又会好，用一段时间又会出现无法写的问题，这样也有可能是硬盘自身的问题吗，不大理解

我软路由也有类似的情况，还是系统盘、固态，某天突然出现I/O错误，被锁定只读，然后虚拟机里面的系统就全部死了，尝试修复过，但问题依旧。所以我可能觉得是硬盘坏了，得换了。

zfh135072, 2021-01-05:
e2fsck修复一下后，暂时好了，估计硬盘也快坏了。。。

closed 2021-01-05 by zfh135072