coolsnowwolf/lede #6149: 编译时添加了docker，NAT环回还是有问题

NagaseKouichi, 2021-01-03:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：今天编译的最新代码，编译时添加了docker，路由器上升级后，NAT环回还是有问题，无法用DDNS域名访问内网的服务。升级的时候保留了设置，是需要不保留设置升级才行吗？

### (2) 路由器型号和固件版本

A：X86软路由

### (3) 详细日志

A：

NagaseKouichi, 2021-01-03:
升级之前的版本没有编译docker，NAT环回没问题

jun129, 2021-01-03:
docker network create -d macvlan --subnet=192.168.1.0/24 --gateway=192.168.1.1 -o parent=br-lan -o macvlan_mode=bridge macvlan
docker run -d --restart=always --name="portainer" --net=macvlan --ip=192.168.1.100 -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce

jun129, 2021-01-03:
我是设置过这个后内网的Nat回环就通了 br-lan对应你的内网

jun129, 2021-01-03:
192.168.1.100改你docker地址

NagaseKouichi, 2021-01-03:
> docker network create -d macvlan --subnet=192.168.1.0/24 --gateway=192.168.1.1 -o parent=br-lan -o macvlan_mode=bridge macvlan
> docker run -d --restart=always --name="portainer" --net=macvlan [...]

这个是用了macvlan驱动，让容器有独立的IP吧，macvlan的性能好像比桥接或者host差一些

jun129, 2021-01-03:
> 这个是用了macvlan驱动，让容器有独立的IP吧，macvlan的性能好像比桥接或者host差一些

当初是也是试了好久 发现只有macvlan可以内网NAT回环，你要是试出别的办法记得告诉我一下。。。

NagaseKouichi, 2021-01-03:
> > > docker network create -d macvlan --subnet=192.168.1.0/24 --gateway=192.168.1.1 -o parent=br-lan -o macvlan_mode=bridge macvlan
> > > docker run -d --restart=always --name="portainer" --net=macvlan --ip=192.168.1.100 -v /var/run/docker.sock:/var/run/docker.sock -v portainer_data:/data portainer/portainer-ce
> > 
> > 
> > 这个是用了macvlan驱动，让容器有独立的IP吧，macvlan的性能好像比桥接或者host差一些
> 
> 当初也是试了好久 发现只有macvlan可以内网NAT回环，你要是试出别的办法记得告诉我一下。。。

前几天的更新，大雕说已经解决了docker和NAT回环的问题了：

> 修复开启 Docker 后 NAT loopback 回环失效问题

但是我测试还是不行，不知道是不是需要不保留设置升级才可以

jun129, 2021-01-03:
> 前几天的更新，大雕说已经解决了docker和NAT回环的问题了：
>
>
> 但是我测试还是不行，不知道是不是需要不保留设置升级才可以

你全新编译试试。。。

NagaseKouichi, 2021-01-03:
> 你全新编译试试。。。

我用的云编译，都是全新编译的

jun129, 2021-01-03:
> 我用的云编译，都是全新编译的

哦哦 那你全新安装试试。。我一般是用机器的双系统ubuntu编译

NagaseKouichi, 2021-01-04:
> 哦哦 那你全新安装试试。。我一般是用机器的双系统ubuntu编译

解决了，参考了https://github.com/coolsnowwolf/lede/commit/81a53ead5320df7ffd1be03962840ac473c385bc，应该是和默认的网桥防火墙有关，新版本默认关闭了网桥防火墙，全新安装的话应该没问题。

升级的时候如果保留了设置，应该是把之前的防火墙设置也保留了。可以修改/etc/sysctl.conf文件，添加下面几行：
net.bridge.bridge-nf-call-arptables = 0
net.bridge.bridge-nf-call-ip6tables = 0
net.bridge.bridge-nf-call-iptables = 0

然后运行 sysctl -p，NAT环回就可以用了

NagaseKouichi, 2021-01-08:
发现全新安装也不行，还是要改/etc/sysctl.conf文件才能NAT环回

hello--world, 2021-01-08:
全新安装还是不成功嘞

设置这个也不行，请问你是怎么弄好的？

net.bridge.bridge-nf-call-arptables = 0
net.bridge.bridge-nf-call-ip6tables = 0
net.bridge.bridge-nf-call-iptables = 0

NagaseKouichi, 2021-01-08:
> 全新安装还是不成功嘞
> 
> 设置这个也不行，请问你是怎么弄好的？
> 
> net.bridge.bridge-nf-call-arptables = 0
> net.bridge.bridge-nf-call-ip6tables = 0
> net.bridge.bridge-nf-call-iptables = 0

我就是在/etc/sysctl.conf文件里面加上这三行，然后重启就好了

hello--world, 2021-01-09:
我的不行，全新的编译，全新的安装，真奇怪。  但是前几天编译的又没问题，不知道是不是又更新了什么。

hello--world, 2021-01-09:
把luci-app-docker 取消就好了。。。

NagaseKouichi, 2021-01-09:
> 把luci-app-docker 取消就好了。。。

不编译docker当然没问题。要的是编译docker而且还NAT环回没问题

closed 2021-01-18 by coolsnowwolf

fhbyljj, 2021-05-22:
https://github.com/coolsnowwolf/lede/issues/6870

楼主，我怎么感觉，我的问题跟你的一样

fhbyljj, 2021-05-22:
今晚试试你这个解决方法

fhbyljj, 2021-05-22:
问题依旧

fhbyljj, 2021-05-22:
设置这个后要不要重启

pocari89, 2021-12-28:
唉，这个ddns 和nat 回环真是患疾呀，19年到现在我也没安装docker ，有偿寻求真有效办法

superliurui, 2023-02-15:
> 唉，这个ddns 和nat 回环真是患疾呀，19年到现在我也没安装docker ，有偿寻求真有效办法

用socat做端口映射可以