coolsnowwolf/lede #6151: 编译失败

EriTang, 2021-01-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
 [ *] 没有类似的issue

## 2. 详细叙述

make[5]: Entering directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.85'
7713
scripts/kconfig/conf  --syncconfig Kconfig
7714
net/sched/Kconfig:45: warning: menuconfig statement without prompt
7715
net/Kconfig:478:warning: multi-line strings not supported
7716
  CALL    scripts/checksyscalls.sh
7717
  CALL    scripts/atomic/check-atomics.sh
7718
  DESCEND  objtool
7719
  CHK     include/generated/compile.h
7720
  GEN     usr/initramfs_data.cpio
7721
sort: write failed: 'standard output': Broken pipe
7722
sort: write error
7723
  AS      usr/initramfs_data.o
7724
  AR      usr/built-in.a
7725
  GZIP    kernel/config_data.gz
7726
  CC [M]  kernel/configs.o
7727
  CC [M]  lib/zlib_inflate/inffast.o
7728
  CC [M]  lib/zlib_inflate/inflate.o
7729
  CC [M]  lib/zlib_inflate/infutil.o
7730
  CC [M]  lib/zlib_inflate/inftrees.o
7731
  CC [M]  lib/zlib_inflate/inflate_syms.o
7732
  LD [M]  lib/zlib_inflate/zlib_inflate.o
7733
  CC      lib/decompress.o
7734
  AR      lib/lib.a
7735
  EXPORTS lib/lib-ksyms.o
7736
  AR      lib/built-in.a
7737
  GEN     .version
7738
  CHK     include/generated/compile.h
7739
  LD      vmlinux.o
7740
  MODPOST vmlinux.o
7741
  MODINFO modules.builtin.modinfo
7742
  LD      .tmp_vmlinux.kallsyms1
7743
x86_64-openwrt-linux-musl-ld: kernel image bigger than KERNEL_IMAGE_SIZE
7744
Makefile:1106: recipe for target 'vmlinux' failed
7745
make[5]: *** [vmlinux] Error 1
7746
make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.85/.image] Error 2
7747
make[3]: *** [install] Error 2
7748
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.85'
7749
Makefile:29: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.85/.image' failed
7750
make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
7751
Makefile:13: recipe for target 'install' failed
7752
make[3]: Leaving directory '/workdir/openwrt/target/linux'
7753
make[2]: *** [target/linux/install] Error 2
7754
time: target/linux/install#104.66#40.62#163.26
7755
target/Makefile:23: recipe for target 'target/linux/install' failed
7756
make[2]: Leaving directory '/workdir/openwrt'
7757
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
7758
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install' failed
7759
make[1]: Leaving directory '/workdir/openwrt'
7760
make: *** [world] Error 2
7761
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed

### (1) 具体问题

A：编译失败

### (2) 路由器型号和固件版本

A：x86_64

minlang112, 2021-01-05:
kernel image bigger than KERNEL_IMAGE_SIZE

EriTang, 2021-01-05:
我看到了，可是不知道怎么解决

> kernel image bigger than KERNEL_IMAGE_SIZE

minlang112, 2021-01-05:
> 我看到了，可是不知道怎么解决
> 
> > kernel image bigger than KERNEL_IMAGE_SIZE

简言之，如果你是room空间有限的嵌入式设备，检查是不是加入了过多的组件；如果是标准target配置未做改动，就报这个错误，请提供具体信息。

EriTang, 2021-01-05:
需要贴出.config文件吗

> > 我看到了，可是不知道怎么解决
> > > kernel image bigger than KERNEL_IMAGE_SIZE
> 
> 简言之，如果你是room空间有限的嵌入式设备，检查是不是加入了过多的组件；如果是标准target配置未做改动，就报这个错误，请提供具体信息。

EriTang, 2021-01-06:
检查过config文件，没有更改过target配置

> > 我看到了，可是不知道怎么解决
> > > kernel image bigger than KERNEL_IMAGE_SIZE
> 
> 简言之，如果你是room空间有限的嵌入式设备，检查是不是加入了过多的组件；如果是标准target配置未做改动，就报这个错误，请提供具体信息。

closed 2021-06-20 by coolsnowwolf

han0x7300, 2022-04-29:
请问如何解决的呀？

Oumourin, 2022-05-15:
> 请问如何解决的呀？

同求解决方法

licsber, 2022-08-03:
取消ramdisk的选择即可解决 不明问题出处

WhateverMO, 2024-02-28:
同样的问题