coolsnowwolf/lede #6154: esxi7.0u1 平台运行 OpenWRT 网口不定时全部丢失

sky2016cn, 2021-01-05:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

Esxi7.0 只运行 OpenWrt，双网口一个 Wan 口和 Lan 口，网口不定时全部丢失，且无法登录管理页，Esxi 显示 CPU 被全部占用，只能强制关闭 OpenWrt，试过好几个固件都一样会丢失，为此困扰多时，特求助。

### (1) 具体问题

A：能启动正常使用，一段时间后（当天），cpu100% ，网络无法访问。重启OpenWrt虚拟机可以恢复

### (2) 路由器型号和固件版本

A： X86版本内核5.4.85 +sfe 开启docker

### (3) 详细日志

A：[19331.046777]  call_function_single_interrupt +0xf/0x20
[19331.046983]  </IRQ>
[19331.047288]   RIP: 0010 : native_safe_half +0x17/0x20
[19331.047282]  Code : 00 a8 08 75 c5 e9 77 ff ff ff 90 90 90 90 90 90 90 90 90 90 55
8b 05 f9 64 b1 00 48 89 e5 85 c0 7e 07 0f 00 2d 1b 4c 55 00 2d 1b 4c 55 00 fb f4 <5d> c3 0f 1f
80 00 00 00 55 8b 05 d9 64 b1 00 48 89 e5 85 c0 7e 
[19331.048001] RSP: 0018:f f f f f f f 82203e40   EFLAGS: 00000246 ORlG_RAX: f f f f f f f f f f
fff04
[19331.040301I RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000001
[19331.04859ZI RDX; 000000000487921a RSI : 0000000000000000 RDI: 0000000000000000
[19331.048876] RBP: f f f f f f f  82203e40 R08: f f f f80813b1de480  R08: 0000080000000000
[19331.049161] R10: 0000000000000001 Rl1: 0000000000000000   R12: f f f f f f f f8228d010
[19331.049452] R13: 0000000000000000 Rlq: 0000000000000000   R15: 0000000000000000
[19331.049739]  default_idle+0x9/0x10
[19331.049890]  arch_cpu_idle_0x10r0x20
[19331.050047]  default_idle_call +0x27/0x30
[19331.050217]  do_idle+0x19f/0x1c0
[19331.050363]  cpu__startup_entry+0x18/0x20
[19331.050532]  rest_init +0x_a4/0xa6
[19331.050677]  arch_call_rest_init +0x9/0xc
[19331.050846]  start_kernel +0x559/0x567
[19331.051005]  x86_64__start_reservations+0x29/0x2b
[19331.051199]  x86_64__start_kernel +0x71/0x74
[19331.051373]   secondary_startup_64 +0xa4/0xb0

zhangj3, 2021-01-06:
sfe跟docker好像又冲突...
你这套娃厉害啊
Esxi里套OP， OP里套docker.....
你不如玩unraid，将OP跟Docker分开

sky2016cn, 2021-01-06:
问题好像解决了，没再发生网口丢失，你说的对，谢谢！

NagaseKouichi, 2021-01-06:
现在都不敢开SFE了

lovelyOK, 2021-01-06:
开FLOW OFFLOAD就行

sky2016cn, 2021-01-06:
是的，谢谢你们

closed 2021-06-20 by coolsnowwolf