coolsnowwolf/lede #6159: 编译失败

zhengxiang311019, 2021-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

按照readme中描述的操作，今天下载了最新的代码，然后便已出现了报错，报错信息如下：

使用“make -j$(($(nproc) + 1)) V=s”命令报错如下：
make[3]: Leaving directory '/home/ra/lede/package/libs/uclibc++'
time: package/libs/uclibc++/compile#4.30#0.36#12.57
make[2]: Leaving directory '/home/ra/lede'
package/Makefile:107: recipe for target '/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ra/lede'
**Build failed - please re-run with -j1 to see the real error message**
/home/ra/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

使用“make -j1 V=s”命令报错如下：
make[3]: Entering directory '/home/ra/lede/package/lean/openwrt-fullconenat'
touch /home/ra/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.prepared_b17c8c6a639ce0886fd34eefec2f11ad_6664517399ebbbc92a37c5bb081b5c53_check
. /home/ra/lede/include/shell.sh; xzcat /home/ra/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz | tar -C /home/ra/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.. -xf -
tar: This does not look like a tar archive
tar: Exiting with failure status due to previous errors
Makefile:64: recipe for target '/home/ra/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.prepared_b17c8c6a639ce0886fd34eefec2f11ad_6664517399ebbbc92a37c5bb081b5c53' failed
make[3]: *** [/home/ra/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2019-10-21-0cf3b48f/.prepared_b17c8c6a639ce0886fd34eefec2f11ad_6664517399ebbbc92a37c5bb081b5c53] Error 2
make[3]: Leaving directory '/home/ra/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.03#0.02#0.29
package/Makefile:111: recipe for target 'package/lean/openwrt-fullconenat/compile' failed
make[2]: *** [package/lean/openwrt-fullconenat/compile] Error 2
make[2]: Leaving directory '/home/ra/lede'
package/Makefile:107: recipe for target '/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ra/lede'
/home/ra/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

lphgor, 2021-01-06:
tar: This does not look like a tar archive

/home/ra/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz 这个文件有问题，删掉来过

zhengxiang311019, 2021-01-07:
> tar: This does not look like a tar archive
> 
> /home/ra/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz 这个文件有问题，删掉来过

删除/home/ra/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz 之后，重新编译出现了如下错误：
Finding targets
cannot find module providing package github.com/cnsilvan/UnblockNeteaseMusic: working directory is not part of a module

Building targets
go: cannot find main module, but found .git/config in /home/ra/lede
        to create a module there, run:
        cd ../../../../.. && go mod init

Makefile:77: recipe for target '/home/ra/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built' failed
make[3]: *** [/home/ra/lede/build_dir/target-x86_64_musl/UnblockNeteaseMusicGo/.built] Error 1
make[3]: Leaving directory '/home/ra/lede/package/lean/UnblockNeteaseMusicGo'
time: package/lean/UnblockNeteaseMusicGo/compile#0.03#0.02#0.27
package/Makefile:111: recipe for target 'package/lean/UnblockNeteaseMusicGo/compile' failed
make[2]: *** [package/lean/UnblockNeteaseMusicGo/compile] Error 2
make[2]: Leaving directory '/home/ra/lede'
package/Makefile:107: recipe for target '/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/ra/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/ra/lede'
/home/ra/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

minlang112, 2021-01-07:
> 删除/home/ra/lede/dl/fullconenat-2019-10-21-0cf3b48f.tar.xz 之后，重新编译出现了如下错误：
> Finding targets
> cannot find module providing package github.com/cnsilvan/UnblockNeteaseMusic: working directory is not part [...]

你这个应该还是网络问题……

zhengxiang311019, 2021-01-12:
> 你这个应该还是网络问题……

我尝试重新下载代码编译还是有问题，我是可以正常访问github和google的，而且网速不差，按理说不应该存在网络原因的。

zhileifun, 2021-01-15:
我和你出现了一样的问题，还没解决。。。

la8896, 2021-01-19:
我也你出现了一样的问题!

closed 2021-06-20 by coolsnowwolf

zhangguanzhang, 2021-08-14:
伙计们怎么解决的

netlovehf, 2022-01-12:
我也出现这个问题，编译了N次，都是到这个地方出错
make[3]: Entering directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
mkdir -p /home/netlove/lede/dl
SHELL= flock /home/netlove/lede/tmp/.fullconenat-2021-12-04-7fe02604.tar.xz.flock -c '  	/home/netlove/lede/scripts/download.pl "/home/netlove/lede/dl" "fullconenat-2021-12-04-7fe02604.tar.xz" "857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" "" || (  /home/netlove/lede/scripts/dl_github_archive.py --dl-dir="/home/netlove/lede/dl" --url="https://github.com/llccd/netfilter-full-cone-nat.git" --version="7fe02604e073945a3b081c18efbb4d3c799a31f6" --subdir="fullconenat-2021-12-04-7fe02604" --source="fullconenat-2021-12-04-7fe02604.tar.xz" --hash="857062e07606d00ef9ce1b6c5d457ffc1ccfb7667694c1958bc59f1d7c4f08a4" || ( 	echo "Checking out files from the git repository..."; mkdir -p /home/netlove/lede/tmp/dl && cd /home/netlove/lede/tmp/dl && rm -rf fullconenat-2021-12-04-7fe02604 && [ \! -d fullconenat-2021-12-04-7fe02604 ] && git clone  https://github.com/llccd/netfilter-full-cone-nat.git fullconenat-2021-12-04-7fe02604 && (cd fullconenat-2021-12-04-7fe02604 && git checkout 7fe02604e073945a3b081c18efbb4d3c799a31f6 && git submodule update --init --recursive) && echo "Packing checkout..." && export TAR_TIMESTAMP=`cd fullconenat-2021-12-04-7fe02604 && git log -1 --format='\''@%ct'\''` && rm -rf fullconenat-2021-12-04-7fe02604/.git && 	tar --numeric-owner --owner=0 --group=0 --mode=a-s --sort=name ${TAR_TIMESTAMP:+--mtime="$TAR_TIMESTAMP"} -c fullconenat-2021-12-04-7fe02604 | 	xz -zc -7e > /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz && mv /home/netlove/lede/tmp/dl/fullconenat-2021-12-04-7fe02604.tar.xz /home/netlove/lede/dl/ && rm -rf fullconenat-2021-12-04-7fe02604; );  )    '
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53_check
. /home/netlove/lede/include/shell.sh; xzcat /home/netlove/lede/dl/fullconenat-2021-12-04-7fe02604.tar.xz | tar -C /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.. -xf -
[ ! -d ./src/ ] || cp -fpR ./src/. /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604
cp -fpR ./files/Makefile /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/
touch /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.prepared_42645540f7ac912202af65068e987065_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/.configured_*
rm -f /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.openwrt-fullconenat_installed
(cd /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/./; if [ -x ./configure ]; then find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.guess | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.guess; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r chmod u+w; find /home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604/ -name config.sub | xargs -r -n1 cp --remove-destination /home/netlove/lede/scripts/config.sub; AR="x86_64-openwrt-linux-musl-gcc-ar" AS="x86_64-openwrt-linux-musl-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=x86_64-openwrt-linux-musl-ld NM="x86_64-openwrt-linux-musl-gcc-nm" CC="x86_64-openwrt-linux-musl-gcc" GCC="x86_64-openwrt-linux-musl-gcc" CXX="x86_64-openwrt-linux-musl-g++" RANLIB="x86_64-openwrt-linux-musl-gcc-ranlib" STRIP=x86_64-openwrt-linux-musl-strip OBJCOPY=x86_64-openwrt-linux-musl-objcopy OBJDUMP=x86_64-openwrt-linux-musl-objdump SIZE=x86_64-openwrt-linux-musl-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -fmacro-prefix-map=/home/netlove/lede/build_dir/target-x86_64_musl/linux-x86_64/fullconenat-2021-12-04-7fe02604=fullconenat-2021-12-04-7fe02604 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/include -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/fortify -I/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include " LDFLAGS="-L/home/netlove/lede/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/usr/lib -L/home/netlove/lede/staging_dir/toolchain-x86_6 [... cut ...]
make[3]: Leaving directory '/home/netlove/lede/package/lean/openwrt-fullconenat'
time: package/lean/openwrt-fullconenat/compile#0.74#0.34#1.14
    ERROR: package/lean/openwrt-fullconenat failed to build.
make[2]: *** [package/Makefile:116: package/lean/openwrt-fullconenat/compile] Error 1
make[2]: Leaving directory '/home/netlove/lede'
make[1]: *** [package/Makefile:110: /home/netlove/lede/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/netlove/lede'
make: *** [/home/netlove/lede/include/toplevel.mk:230: world] Error 2