coolsnowwolf/lede #6161: 今天翻系统日志出现很多.err

aape01, 2021-01-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：
Intel(R) Celeron(R) J4125 CPU @ 2.00GHz : 4 Core 4 Thread (CpuMark : 59880.239521 Scores)
1312.612 MHz +27.0°C (high = +105.0°C, crit = +105.0°C)
OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)
5.4.79
15d 5h 36m 51s

### (3) 详细日志

A：系统日志
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:30 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:32 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:36 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:36 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:36 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:36 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:38 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:42 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:42 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:42 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:42 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:43 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:43 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:43 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:43 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:44 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:48 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:09:49 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:49 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:09:49 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:49 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:09:50 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:09:56 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:10:02 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:10:25 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:10:25 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:10:28 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:10:31 2020 kern.err /usr/bin/ssr-redir[1352]: server recv: Connection reset by peer
Tue Dec 29 22:10:31 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:10:31 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:10:34 2020 kern.err /usr/bin/ssr-redir[1356]: server recv: Connection reset by peer
Tue Dec 29 22:10:37 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:10:37 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:10:37 2020 kern.err /usr/bin/ssr-redir[1358]: server recv: Connection reset by peer
Tue Dec 29 22:10:40 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
Tue Dec 29 22:10:43 2020 kern.err /usr/bin/ssr-redir[1354]: server recv: Connection reset by peer
[... 625 lines cut ...]
Tue Jan  5 21:12:55 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Tue Jan  5 21:37:00 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Tue Jan  5 21:37:00 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Tue Jan  5 22:13:07 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Out of memory
Tue Jan  5 22:13:07 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: No buffer space available
Wed Jan  6 04:56:26 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Out of memory
Wed Jan  6 04:56:26 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: No buffer space available
Wed Jan  6 06:34:46 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 06:34:46 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 07:35:58 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Out of memory
Wed Jan  6 07:35:58 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: No buffer space available
Wed Jan  6 09:23:19 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Out of memory
Wed Jan  6 09:23:19 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: No buffer space available
Wed Jan  6 09:42:23 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 09:42:23 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 09:42:53 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 09:42:53 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 09:59:57 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 09:59:57 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 10:20:01 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 10:20:01 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 10:43:05 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Object busy
Wed Jan  6 10:43:05 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: I/O error
Wed Jan  6 19:11:45 2021 daemon.err nlbwmon[2548]: Netlink receive failure: Out of memory
Wed Jan  6 19:11:45 2021 daemon.err nlbwmon[2548]: Unable to dump conntrack: No buffer space available

skefy, 2021-01-06:
> kern.err /usr/bin/ssr-redir[11553]: server recv: Connection reset by peer

类似的错误，我也一大堆。但是并没有感觉影响使用。

keeno1982, 2021-01-14:
这个日志很烦，拼命刷存在感，导致想看的日志都看不到，全是它的！想知道怎么去掉这个日志

closed 2021-06-20 by coolsnowwolf