coolsnowwolf/lede #6164: 树莓派4B编译失败 cypress-fmac-v5.4.18-2020_0925 下载不了 链接全部失效

IvanSolis1989, 2021-01-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [X ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

编译失败

### (2) 路由器型号和固件版本

树莓派4B

### (3) 详细日志

SHELL= flock /workdir/openwrt/tmp/.cypress-fmac-v5.4.18-2020_0925.zip.flock -c '  	/workdir/openwrt/scripts/download.pl "/workdir/openwrt/dl" "cypress-fmac-v5.4.18-2020_0925.zip" "fb71c344e705f5bc9fdae3ce0fbfa299f0af0939ff3ec782aeca0308911d830d" "" "https://community.cypress.com/servlet/JiveServlet/download/21490-1-58574/"    '
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://community.cypress.com/servlet/JiveServlet/download/21490-1-58574/cypress-fmac-v5.4.18-2020_0925.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://sources.openwrt.org/cypress-fmac-v5.4.18-2020_0925.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.
+ curl -f --connect-timeout 20 --retry 5 --location --insecure https://mirror2.openwrt.org/sources/cypress-fmac-v5.4.18-2020_0925.zip
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed

  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
  0     0    0     0    0     0      0      0 --:--:-- --:--:-- --:--:--     0
curl: (22) The requested URL returned error: 404 Not Found
Download failed.

minlang112, 2021-01-07:
这几个地址确实都失效了

spypie, 2021-01-08:
我的是跟verysync.com有关的几个失效了

meishimeike, 2021-01-09:
新地址
https://community.cypress.com/gfawx74859/attachments/gfawx74859/resourcelibrary/1030/1/cypress-fmac-v5.4.18-2020_0925.zip

PR #6435 (2021-02-25): bcm27xx: sync upstream source code
fix: #6164
closed 2021-02-25 by coolsnowwolf