coolsnowwolf/lede #6166: k2p普通版编译完无法启动

286700234, 2021-01-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

qjiang888, 2021-01-08:
帮顶，哈哈，新内核K2P 都启动不了！！

q397333, 2021-01-08:
顶，16M无法启动

YorkPeng, 2021-01-09:
启动不了+1

AmadeusGhost, 2021-01-10:
>  帮顶，哈哈，新内核K2P 都启动不了！！ 

u1s1，跟新内核没关系，目前来看大概率是因为默认超频了 cpu mt7621 引起的

qjiang888, 2021-01-10:
> > 帮顶，哈哈，新内核K2P 都启动不了！！
> 
> u1s1，跟新内核没关系，目前来看大概率是因为默认超频了 cpu mt7621 引起的

大佬，我发了启动日志，有时间麻烦看看

qjiang888, 2021-01-10:
默认超频breed里面可以改吗？

qjiang888, 2021-01-10:
通过修改 \target\linux\ramips\patches-5.4\102-mt7621-fix-cpu-clk-add-clkdev.patch

`case 1:
-		fbdiv = ((rt_sysc_r32(0x648) >> 4) & 0x7F) + 1;
-		syscfg = rt_sysc_r32(SYSC_REG_SYSCFG);
-		xtal_mode = (syscfg >> 6) & 0x7;
-		if (xtal_mode >= 6) {
-			/* 25Mhz Xtal */
-			cpu_clk = 25 * fbdiv * 1000 * 1000;
-		} else if (xtal_mode >= 3) {
-			/* 40Mhz Xtal */
-			cpu_clk = 40 * fbdiv * 1000 * 1000;
-		} else {
-			/* 20Mhz Xtal */
-			cpu_clk = 20 * fbdiv * 1000 * 1000;
-		}
+		pll = rt_memc_r32(MEMC_REG_CPU_PLL);
+		if ((pll & 0x7f0) == 0x2b0) {
+			volatile u32 i;
+
+			pr_info("CPU Clock: 880MHz, start overclocking\n");
+			pll &= ~0x7ff;
+			pll |= 0x312;
+			rt_memc_w32(pll, MEMC_REG_CPU_PLL);
+			for (i = 0; i < 1000; i++);
+		}
+		fbdiv = (pll >> CPU_PLL_FBDIV_SHIFT) & CPU_PLL_FBDIV_MASK;
+		prediv = (pll >> CPU_PLL_PREDIV_SHIFT) & CPU_PLL_PREDIV_MASK;
+		cpu_clk = ((fbdiv + 1) * xtal_clk) >> prediv_tbl[prediv];
 		break;`

我把pll |= 0x312;改成0X2B2 make clean  重新编译刷了 好像可以重启了！！

qjiang888, 2021-01-10:
但是还是有几率无法启动！！别的固件都没有问题

q397333, 2021-01-10:
> > 帮顶，哈哈，新内核K2P 都启动不了！！
> 
> u1s1，跟新内核没关系，目前来看大概率是因为默认超频了 cpu mt7621 引起的

试了下 git revert -n 5be8131 果然解决了，可以正常重启

dengxin6, 2021-01-12:
web升级openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade后无法启动，Breed里刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel能正常启动

qjiang888, 2021-01-12:
是固件里面超频了，不超频就没事，等大佬去掉超频吧

PR #6190 (2021-01-13): Revert "ramips: mt7621 default overclock to 1000Mhz"
建议还是用 Breed 来超频更为合适
Fix: #6090 , Fix: #6166 , Fix: #6172
closed 2021-01-13 by coolsnowwolf