coolsnowwolf/lede #6168: IPsec 无法使用没有会处理的吗？lean也没个说法？

kkxkx, 2021-01-08:
Thu Dec 3 21:50:46 2020 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.8.4, Linux 5.4.75, x86_64)
Thu Dec 3 21:50:46 2020 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Thu Dec 3 21:50:46 2020 daemon.info : 00[KNL] failed to create TUN device
Thu Dec 3 21:50:46 2020 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Thu Dec 3 21:50:46 2020 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Thu Dec 3 21:50:46 2020 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loaded IKE secret for %any
Thu Dec 3 21:50:46 2020 daemon.info : 00[CFG] loaded EAP secret for lean
Thu Dec 3 21:50:46 2020 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Thu Dec 3 21:50:46 2020 daemon.info : 00[JOB] spawning 16 worker threads
Thu Dec 3 21:50:46 2020 daemon.info : 05[DMN] thread 5 received 11
Thu Dec 3 21:50:46 2020 daemon.info : 05[LIB] no support for capturing backtraces
Thu Dec 3 21:50:46 2020 daemon.info : 05[DMN] killing ourself, received critical signal
Thu Dec 3 21:50:46 2020 authpriv.info ipsec_starter[3425]: charon has died -- restart scheduled (5sec)
Thu Dec 3 21:50:46 2020 authpriv.info ipsec_starter[3425]: charon refused to be started

dolphinpaopao, 2021-01-08:
请问需要作者给个什么说法呢？这项目只是作者个人兴趣的开源项目，非商业化，既然是开源，那么有额外功能需求是可以自己修改或者添加代码来实现。没必要什么都要作者做这做那，有的bug也有可能是上游的问题。lean的这个有问题完全也可以切换到其他人的分支，不用这个就是。

NagaseKouichi, 2021-01-08:
要个说法……你以为维权呢 XD

kkxkx, 2021-01-09:
不好意思 表达失误！

closed 2021-01-09 by kkxkx

eearphon, 2021-07-31:
同样问题，劳驾大神看看？红米AC2100，最新源码Actions编译 492539792279dd38cac889288443324d63e40aca
log里循环这样：
Sat Jul 31 08:51:44 2021 daemon.info : 00[DMN] Starting IKE charon daemon (strongSwan 5.9.3, Linux 5.4.132, mips)
Sat Jul 31 08:51:44 2021 daemon.info : 00[LIB] failed to open /dev/net/tun: No such file or directory
Sat Jul 31 08:51:44 2021 daemon.info : 00[KNL] failed to create TUN device
Sat Jul 31 08:51:44 2021 daemon.info : 00[LIB] plugin 'kernel-libipsec': failed to load - kernel_libipsec_plugin_create returned NULL
Sat Jul 31 08:51:44 2021 daemon.info : 00[KNL] unable to create IPv4 routing table rule
Sat Jul 31 08:51:44 2021 daemon.info : 00[KNL] unable to create IPv6 routing table rule
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading ca certificates from '/etc/ipsec.d/cacerts'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading aa certificates from '/etc/ipsec.d/aacerts'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading ocsp signer certificates from '/etc/ipsec.d/ocspcerts'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading attribute certificates from '/etc/ipsec.d/acerts'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading crls from '/etc/ipsec.d/crls'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG] loading secrets from '/etc/ipsec.secrets'
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG]   loaded IKE secret for %any
Sat Jul 31 08:51:44 2021 daemon.info : 00[CFG]   loaded EAP secret for myuser
Sat Jul 31 08:51:44 2021 daemon.info : 00[LIB] loaded plugins: charon aes sha1 random nonce x509 pubkey gmp xcbc hmac kernel-netlink socket-default stroke updown xauth-generic
Sat Jul 31 08:51:44 2021 daemon.info : 00[JOB] spawning 16 worker threads
Sat Jul 31 08:51:44 2021 daemon.info : 05[DMN] thread 5 received 11
Sat Jul 31 08:51:44 2021 daemon.info : 05[LIB] no support for capturing backtraces
Sat Jul 31 08:51:44 2021 daemon.info : 07[DMN] thread 7 received 11
Sat Jul 31 08:51:44 2021 daemon.info : 05[DMN] killing ourself, received critical signal
Sat Jul 31 08:51:44 2021 authpriv.info ipsec_starter[9325]: charon has died -- restart scheduled (5sec)
Sat Jul 31 08:51:44 2021 authpriv.info ipsec_starter[9325]: charon refused to be started