coolsnowwolf/lede #6171: MT1300使用coolsnowwolf/lede最新代码编译OpenWrt-mt7621刷入后：内存提示 0.00 mb 。不知道怎么解决，求助

designer-fanfan, 2021-01-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用coolsnowwolf/lede最新代码编译OpenWrt-mt7621刷入后：内存提示 0.00 mb 。不知道怎么解决，求助

### (2) 路由器型号和固件版本

A：GL iNET MT1300

### (3) 详细日志

A：

rockey2020, 2021-01-09:
无线和wan可以用吗

designer-fanfan, 2021-01-09:
> 无线和wan可以用吗

无线不能用，只有5db，ip地址也无法获取。

gsblay, 2021-01-10:
你编译时选择的有线和无线驱动分别是什么？
编译完以后，先刷initramfs-kernel.bin，在此基础上再刷squashfs-sysupgrade.bin试试。
好像2.4G驱动不出来。

wf09, 2021-01-19:
一样的问题哎。直接刷squashfs-sysupgrade.bin无法开机，刷initramfs-kernel.bin后显示存储空间0.00mb。先刷initramfs-kernel.bin，在此基础上再刷squashfs-sysupgrade.bin还是无法开机。可能是新机器，有bug吧。

CYJason2000, 2021-01-26:
我也遇到了同样的问题，不过我的是海凌科的7621模块，现在已经解决。如果保证dts内flash分区无误且刷入sysupgrade后可以启动就说明flash识别是没有问题的，第一次启动需要时间初始化文件系统，我的日志显示100秒左右才挂载上了rootfs

designer-fanfan, 2021-01-28:
> 你编译时选择的有线和无线驱动分别是什么？
> 编译完以后，先刷initramfs-kernel.bin，在此基础上再刷squashfs-sysupgrade.bin试试。
> 好像2.4G驱动不出来。

无线驱动都是默认的，没有做更改， 先刷initramfs再刷squashfs也不行，不能开机

designer-fanfan, 2021-01-28:
> 一样的问题哎。直接刷squashfs-sysupgrade.bin无法开机，刷initramfs-kernel.bin后显示存储空间0.00mb。先刷initramfs-kernel.bin，在此基础上再刷squashfs-sysupgrade.bin还是无法开机。可能是新机器，有bug吧。

问题解决了么？我还在卡着呢

gitbuger, 2021-01-31:
> > 一样的问题哎。直接刷squashfs-sysupgrade.bin无法开机，刷initramfs-kernel.bin后显示存储空间0.00mb。先刷initramfs-kernel.bin，在此基础上再刷squashfs-sysupgrade.bin还是无法开机。可能是新机器，有bug吧。
> 
> 问题解决了么？我还在卡着呢
你的squashfs-sysupgrade.bin多大？

closed 2021-06-20 by coolsnowwolf