coolsnowwolf/lede #6172: K2P 16M启动日志

qjiang888, 2021-01-09:
刷完openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel  正常启动，进web升级openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade后无法启动， 我接上TTL 开机正常，微博里面第一次重启也起来了，第二次重启就卡在日志最后一行了！[    2.586924] mt7621-pci 1e140000.pcie: Parsing DT failed
请大佬看看
日志文件
[bootlog.txt](https://github.com/coolsnowwolf/lede/files/5790394/bootlog.txt)
 没有类似的issue

coolsnowwolf, 2021-01-10:
CPU 死机了，默认超频 1Ghz ，可能你的 CPU 体制太差导致的

qjiang888, 2021-01-10:
谢谢大佬回复，我看看怎么降频

dengxin6, 2021-01-12:
估计不是CPU体质问题   
我的CPU一直用breed超频1.1G在用，刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel能正常使用，但是在“备份/升级”里面没法下载备份，也没有“执行重置”按钮。
web升级openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade后无法启动，Breed里刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel又继续循环。

dengxin6, 2021-01-12:
> CPU 死机了，默认超频 1Ghz ，可能你的 CPU 体制太差导致的

估计不是CPU体质问题
我的CPU一直用breed超频1.1G在用，刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel能正常使用，但是在“备份/升级”里面没法下载备份，也没有“执行重置”按钮。
web升级openwrt-ramips-mt7621-phicomm_k2p-squashfs-sysupgrade后无法启动，Breed里刷openwrt-ramips-mt7621-phicomm_k2p-initramfs-kernel又继续循环。

dengxin6, 2021-01-12:
确实是超频导致的问题，不过很奇怪，breed超频就没问题

PR #6190 (2021-01-13): Revert "ramips: mt7621 default overclock to 1000Mhz"
建议还是用 Breed 来超频更为合适
Fix: #6090 , Fix: #6166 , Fix: #6172
closed 2021-01-13 by coolsnowwolf