coolsnowwolf/lede #6173: 编译失败

Sunnice, 2021-01-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：云编译失败

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志

A：2021-01-08T08:00:58.5971201Z  make[2] -C package/utils/mtd-utils download
2021-01-08T08:00:58.5985994Z  make[2] -C package/utils/util-linux download
2021-01-08T08:31:52.5697713Z make package/download: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
2021-01-08T08:31:52.5762042Z  make[1] target/download
2021-01-08T08:31:54.3006050Z  make[2] -C target/linux download
2021-01-08T08:31:54.3649968Z ##[group]Run cd openwrt
2021-01-08T08:31:54.3650421Z [36;1mcd openwrt[0m
2021-01-08T08:31:54.3650799Z [36;1mecho -e "$(nproc) thread compile"[0m
2021-01-08T08:31:54.3651247Z [36;1mmake -j$(nproc) || make -j1 || make -j1 V=s[0m
2021-01-08T08:31:54.3651718Z [36;1mecho "::set-output name=status::success"[0m
2021-01-08T08:31:54.3652307Z [36;1mgrep '^CONFIG_TARGET.*DEVICE.*=y' .config | sed -r 's/.*DEVICE_(.*)=y/\1/' > DEVICE_NAME[0m
2021-01-08T08:31:54.3652925Z [36;1m[ -s DEVICE_NAME ] && echo "DEVICE_NAME=_$(cat DEVICE_NAME)" >> $GITHUB_ENV[0m
2021-01-08T08:31:54.3653456Z [36;1mecho "FILE_DATE=_$(date +"%Y%m%d%H%M")" >> $GITHUB_ENV[0m
2021-01-08T08:31:54.3815301Z shell: /bin/bash -e {0}
2021-01-08T08:31:54.3815630Z env:
2021-01-08T08:31:54.3816319Z   REPO_URL: https://github.com/coolsnowwolf/lede
2021-01-08T08:31:54.3816923Z   REPO_BRANCH: master
2021-01-08T08:31:54.3817450Z   FEEDS_CONF: feeds.conf.default
2021-01-08T08:31:54.3817953Z   CONFIG_FILE: .config
2021-01-08T08:31:54.3818535Z   DIY_P1_SH: diy-part1.sh
2021-01-08T08:31:54.3819015Z   DIY_P2_SH: diy-part2.sh
2021-01-08T08:31:54.3819453Z   UPLOAD_BIN_DIR: true
2021-01-08T08:31:54.3819896Z   UPLOAD_FIRMWARE: true
2021-01-08T08:31:54.3820363Z   UPLOAD_COWTRANSFER: true
2021-01-08T08:31:54.3820942Z   UPLOAD_WETRANSFER: true
2021-01-08T08:31:54.3821405Z   UPLOAD_RELEASE: true
2021-01-08T08:31:54.3821821Z   TZ: Asia/Shanghai
2021-01-08T08:31:54.3822232Z ##[endgroup]
2021-01-08T08:31:54.3895574Z 2 thread compile
2021-01-08T08:32:01.1063082Z  make[1] world
2021-01-08T08:32:02.8181823Z  make[2] tools/compile
2021-01-08T08:32:02.8957040Z  make[2] package/cleanup
2021-01-08T08:32:04.7128669Z  make[3] -C tools/flock compile
2021-01-08T08:32:04.7699545Z  make[3] -C tools/xz compile
2021-01-08T08:32:19.4191000Z  make[3] -C tools/sed compile
2021-01-08T08:32:44.8336320Z  make[3] -C tools/patch compile
2021-01-08T08:32:44.8343547Z  make[3] -C tools/tar compile
2021-01-08T08:33:21.4966056Z  make[3] -C tools/m4 compile
2021-01-08T08:33:21.4977230Z  make[3] -C tools/pkgconf compile
2021-01-08T08:33:30.3796255Z  make[3] -C tools/expat compile
2021-01-08T08:33:43.4428051Z  make[3] -C tools/autoconf-archive compile
2021-01-08T08:33:45.2154807Z  make[3] -C tools/libressl compile
2021-01-08T08:33:47.7788477Z  make[3] -C tools/sstrip compile
2021-01-08T08:33:48.0231977Z  make[3] -C tools/patch-image compile
2021-01-08T08:33:48.2290384Z  make[3] -C tools/mtools compile
2021-01-08T08:34:01.8821782Z  make[3] -C tools/ucl compile
2021-01-08T08:34:30.5370036Z  make[3] -C tools/lzma compile
2021-01-08T08:34:35.9221041Z  make[3] -C tools/zip compile
2021-01-08T08:34:38.9652676Z  make[3] -C tools/zstd compile
2021-01-08T08:35:31.8370517Z  make[3] -C tools/autoconf compile
2021-01-08T08:35:32.2188540Z  make[3] -C tools/upx compile
[... 528 lines cut ...]
2021-01-08T13:38:23.7844420Z ##[error]The operation was canceled.
2021-01-08T13:38:23.8052312Z Post job cleanup.
2021-01-08T13:38:24.2616662Z [command]/usr/bin/git version
2021-01-08T13:38:24.2687066Z git version 2.29.2
2021-01-08T13:38:24.2747392Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2021-01-08T13:38:24.2807206Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2021-01-08T13:38:24.3309930Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-01-08T13:38:24.3353355Z http.https://github.com/.extraheader
2021-01-08T13:38:24.3370290Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
2021-01-08T13:38:24.3412490Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'http\.https\:\/\/github\.com\/\.extraheader' && git config --local --unset-all 'http.https://github.com/.extraheader' || :
2021-01-08T13:38:24.3842780Z Cleaning up orphan processes
2021-01-08T13:38:24.4082044Z Terminate orphan process: pid (1935) (make)
2021-01-08T13:38:24.4213725Z Terminate orphan process: pid (21557) (perl)
2021-01-08T13:38:24.4254140Z Terminate orphan process: pid (21558) (make)
2021-01-08T13:38:24.4294600Z Terminate orphan process: pid (29824) (make)
2021-01-08T13:38:24.4350565Z Terminate orphan process: pid (29827) (make)
2021-01-08T13:38:24.4399269Z Terminate orphan process: pid (29828) (bash)
2021-01-08T13:38:24.4460973Z Terminate orphan process: pid (29843) (bash)
2021-01-08T13:38:24.4516246Z Terminate orphan process: pid (29844) (make)
2021-01-08T13:38:24.4564810Z Terminate orphan process: pid (29845) (bash)
2021-01-08T13:38:24.4611446Z Terminate orphan process: pid (29996) (bash)
2021-01-08T13:38:24.4657296Z Terminate orphan process: pid (29997) (make)
2021-01-08T13:38:24.4750598Z Terminate orphan process: pid (30068) (arm-openwrt-linux-muslgnueabi-g++)
2021-01-08T13:38:24.4809965Z Terminate orphan process: pid (30070) (cc1plus)
2021-01-08T13:38:24.4867313Z Terminate orphan process: pid (30071) (as)

smallprogram, 2021-01-11:
Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

1607941875, 2021-04-29:
把中文路径改成英文路径试一试

closed 2021-06-20 by coolsnowwolf