coolsnowwolf/lede #6174: 动态DNS有关DNSpod.cn服务商多条同一主机记录更新的问题反馈

feheadls, 2021-01-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
openwrt使用动态DNS，服务商为DNSPOD.CN，在对同一个二级域名例如test.domain.com分别设置一个A记录和一个AAAA记录，用于该域名IPV4/IPV6双栈解析，发现只会更新到服务商端设置的第一条test主机记录上，第二条test主机记录无会被更新。
详细过程：
1、在服务商的域名解析配置里添加两条主机记录，主机记录名均为test，两条记录类型分别为A记录和AAAA记录。
2、在openwrt动态DNS里，添加两条DDNS客户端配置，分别用于更新ipv4地址和ipv6地址，均能成功更新到服务商，但由于服务商端主机记录名相同，均为test，发现ipv4和ipv6地址的更新，只会更新到服务商端设置的第一条test主机记录上，第二条test主机记录无会被更新。
3、提交工单向腾讯反馈，回复我多条解析记录虽然主机名称一样，但记录ID不一样，如要更新指定的解析记录需要按API规范带上recordid参数。
4、通过openwrt动态DNS对应配置项里的日志发现，动态DNS客户端是先通过调用https://dnsapi.cn/Record.List 接口获取记录列表的相关信息，再根据记录列表情况运行对应的更新命令创建/修改/删除解析记录等操作。猜测客户端调用此接口的依据是主机记录test，当主机记录test只有只一记录时，返回的recordid值不会有问题，但当主机记录test存在多个解析记录时，接口会返回test的多个recordid（已证实确实如此），而后发起创建/修改/删除解析记录等操作时，只会以第一条记录的recordid发送给服务端，造成了上述问题。
5、经手动运行更新命令，带入对应的recordid发起更新，然后在服务端检查，发现可以正常更新到指定的解析记录，排除了服务端或API接口相关的问题。

### (1) 具体问题

A：动态DNS功能DNSpod.cn服务商的客户端，当服务端同一主机记录存在多条解析记录时，客户端只能对https://dnsapi.cn/Record.List 接口返回的第一条解析记录进行操作，无法达成IPV4/IPV6双栈解析这样的应用方式。

### (2) 路由器型号和固件版本

A：X86_64 ESXI 7.0U1虚拟机，OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：敏感信息已作了去隐处理
 174831  note : PID '23061' started at 2021-01-09 17:48
 174831       : ddns version  : 2.7.8-3
 174831       : uci configuration:
ddns.dnspod_v6.domain='test.domain.com'
ddns.dnspod_v6.enabled='1'
ddns.dnspod_v6.force_interval='0'
ddns.dnspod_v6.interface='lan'
ddns.dnspod_v6.ip_interface='br-lan'
ddns.dnspod_v6.ip_network='lan'
ddns.dnspod_v6.ip_source='interface'
ddns.dnspod_v6.lookup_host='test.domain.com'
ddns.dnspod_v6.password='***PW***'
ddns.dnspod_v6.service_name='dnspod.cn'
ddns.dnspod_v6.use_ipv6='1'
ddns.dnspod_v6.username='XXXXXX'
ddns.dnspod_v6=service
 174831       : verbose mode  : 0 - run normal, NO console output
 174832       : check interval: 600 seconds
 174832       : force interval: 0 seconds
 174832       : retry interval: 60 seconds
 174832       : retry counter : 0 times
 174832       : No old process
 174832       : last update: 2021-01-09 17:47
 174832       : Detect registered/public IP
 174832       : #> /usr/bin/nslookup test.domain.com  >/var/run/ddns/dnspod_v6.dat 2>/var/run/ddns/dnspod_v6.err
 174832       : Registered IP '240e:330:XXXX:XXXX::1' detected
 174832  info : Starting main loop at 2021-01-09 17:48
 174832       : Detect local IP on 'interface'
 174832       : Local IP '240e:330:XXXX:XXXX::1' detected on network 'lan'
 174832       : Forced Update - L: '240e:0330:XXXX:XXXX:0000:0000:0000:0001' == R: '240e:0330:XXXX:XXXX:0000:0000:0000:0001'
 174833       : parsing script '/usr/lib/ddns/update_dnspod_cn.sh'
 174833       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/dnspod_v6.dat -o /var/run/ddns/dnspod_v6.err --no-proxy --post-data 'login_token=XXXXXX,***PW***&format=json&domain=domain.com&sub_domain=test' https://dnsapi.cn/Record.List
 174833       : Get parsing recordID:[XXXX90862]
 174833       : Address needs to be modified, local address:[240e:0330:XXXX:XXXX:0000:0000:0000:0001]
 174836       : #> /usr/bin/wget-ssl --no-hsts -nv -t 1 -O /var/run/ddns/dnspod_v6.dat -o /var/run/ddns/dnspod_v6.err --no-proxy --post-data 'login_token=XXXXXX,***PW***&format=json&domain=domain.com&sub_domain=test&value=240e:0330:XXXX:XXXX:0000:0000:0000:0001&record_type=AAAA&record_line_id=0&record_id=XXXX90862&ttl=600' https://dnsapi.cn/Record.Modify
 174837       : Modify new parsing record [test.domain.com],[type:AAAA],[ip:240e:0330:XXXX:XXXX:0000:0000:0000:0001],[TTL:600] successfully!
 174837  info : Forced update successful - IP: '240e:0330:XXXX:XXXX:0000:0000:0000:0001' send
 174837       : Waiting 600 seconds (Check Interval)
 175838       : Detect registered/public IP
 175838       : #> /usr/bin/nslookup test.domain.com  >/var/run/ddns/dnspod_v6.dat 2>/var/run/ddns/dnspod_v6.err
 175838       : Registered IP '240e:330:XXXX:XXXX::1' detected
 175838  info : Configured to run once
 175838  note : PID '23061' exit normal at 2021-01-09 17:58

feheadls, 2021-01-14:
经过几天解读代码，通过一种比较暴力的方式解决了IPV4/IPV6分别更新到服务端主机记录同名的两条解析记录上，实现了我的双栈解析需求。

方法就是在修改/usr/lib/ddns/update_dnspod_cn.sh文件，在下面这段代码__POST开头的那一行双引号尾部加入&record_type=$__TYPE，使脚本在获取域名解析记录信息时，指定记录类型来获取解析记录信息，从而获得对应解析记录的record_id。附件已上传修改好的脚本文件，可解压后直接覆盖原脚本使用。
[update_dnspod_cn.zip](https://github.com/coolsnowwolf/lede/files/5815360/update_dnspod_cn.zip)

该方法并不完美，只能解决同一主机记录，每种记录类型只存在一条解析记录的情况，仍然不能解决同一主机记录存在多条A记录的情况，还望有能力的大神继续完善脚本。

describe_domain() {                                                                                                           
        ret=0                                                                                                                 
        __POST="login_token=$username,$password&format=json&domain=$__DOMAIN&sub_domain=$__HOST&record_type=$__TYPE"          
        __POST1="$__POST&value=$__IP&record_type=$__TYPE&record_line_id=0"                                                    
        dnspod_transfer 0                                                                                                     
        __TMP=`jsonfilter -i $DATFILE -e "@.records[@.type!='NS']"`                                                           
        __value=`jsonfilter -s "$__TMP" -e "@.name"`                                                                          
        if [ "$__value" == "" ]; then                                                                                         
                write_log 7 "Parsing record:[${__HOST}.${__DOMAIN}] does not exist"                                           
                ret=1                                                                                                         
        else                                                                                                                  
                __RECID=`jsonfilter -s "$__TMP" -e "@.id"`                                                                    
                __value=`jsonfilter -s "$__TMP" -e "@.type"`                                                                  
                __TTL=`jsonfilter -s "$__TMP" -e "@.ttl"`                                                                     
                write_log 7 "Get parsing recordID:[$__RECID]"                                                                 
                if [ "$__value" != "$__TYPE" ]; then                                                                          
                        write_log 7 "Current parsing [type:$__TYPE], get mismatched [type:$__value]"                          
                        write_log 7 "Address needs to be modified, local address:[$__IP]"                                     
                        ret=2                                                                                                 
                else                                                                                                          
                        __RECIP=`jsonfilter -s "$__TMP" -e "@.value"`                                                         
                        if [ "$__RECIP" != "$__IP" ]; then                                                                    
                                write_log 7 "Address needs to be modified, local address:[$__IP]"                             
                                ret=2                                                                                         
                        fi                                                                                                    
                fi                                                                                                            
        fi                                                                                                                    
}

jdjingdian, 2021-02-09:
@feheadls 感谢方法，终于用上双栈dns解析了

closed 2021-06-20 by coolsnowwolf