coolsnowwolf/lede #6175: 关于luci-app-openvpn-server多用户冲突的问题

a007007007csy, 2021-01-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-openvpn-server 可正常使用，但是无法支持多用户同时在线（证书冲突），搜索过得知需要添加如下参数【duplicate-cn 定义openvpn一个证书在同一时刻是否允许多个客户端接入，默认没有启用。】。但是插件生成的配置文件没有地方可以添加此参数。希望可以解决此问题。

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：None

neavo, 2021-01-10:
同样需求，期望可以有大佬支持

kleisth, 2021-01-11:
openvpn server是可以自己定义配置的啊，我是改成了账户密码验证的，不用客户端证书，如果想要用客户端证书的，可以按用户名自己生成证书用，想只用一个证书的，openvpn服务端配置增加“duplicate-cn”参数即可

coolsnowwolf, 2021-01-11:
正解

qwgaan, 2021-01-14:
还可以ssh进去，直接修改/etc/config/openvpn，增加这个参数也可以，openvpn的参数一般我都是通过修改配置文件改比较快也明确，
        option duplicate_cn '1'

a007007007csy, 2021-01-21:
> 还可以ssh进去，直接修改/etc/config/openvpn，增加这个参数也可以，openvpn的参数一般我都是通过修改配置文件改比较快也明确，
> option duplicate_cn '1'

实测此方法适用。感谢~

closed 2021-01-21 by a007007007csy