coolsnowwolf/lede #6178: 编译5.4.86 ipq40xx 开始内核补丁就过不去帮忙看看

stock169, 2021-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译错误

### (2) 路由器型号和固件版本

A：竞斗云

### (3) 详细日志

A：Applying /home/pchome/lede/target/linux/ipq40xx/patches-5.4/0013-v5.6-crypto-qce-allow-building-only-hashes-ciphers.patch using plaintext: 
patching file drivers/crypto/Kconfig
patching file drivers/crypto/qce/Makefile
Hunk #1 FAILED at 2.
1 out of 1 hunk FAILED -- saving rejects to file drivers/crypto/qce/Makefile.rej
patching file drivers/crypto/qce/common.c
Hunk #3 succeeded at 225 with fuzz 2.
Hunk #4 FAILED at 387.
1 out of 4 hunks FAILED -- saving rejects to file drivers/crypto/qce/common.c.rej
patching file drivers/crypto/qce/core.c
Hunk #1 FAILED at 22.
1 out of 1 hunk FAILED -- saving rejects to file drivers/crypto/qce/core.c.rej
Patch failed!  Please fix /home/pchome/lede/target/linux/ipq40xx/patches-5.4/0013-v5.6-crypto-qce-allow-building-only-hashes-ciphers.patch!
Makefile:25: recipe for target '/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.86/.prepared_917f808ff71b0862b9088a232d1b3996' failed
make[4]: *** [/home/pchome/lede/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/linux-ipq40xx_generic/linux-5.4.86/.prepared_917f808ff71b0862b9088a232d1b3996] Error 1
make[4]: Leaving directory '/home/pchome/lede/target/linux/ipq40xx'
Makefile:13: recipe for target 'compile' failed
make[3]: *** [compile] Error 2
make[3]: Leaving directory '/home/pchome/lede/target/linux'
time: target/linux/compile#15.75#9.25#23.61
target/Makefile:23: recipe for target 'target/linux/compile' failed
make[2]: *** [target/linux/compile] Error 2
make[2]: Leaving directory '/home/pchome/lede'
target/Makefile:18: recipe for target '/home/pchome/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile' failed
make[1]: *** [/home/pchome/lede/staging_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/pchome/lede'
/home/pchome/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

kqbcarl, 2021-01-10:
我也一样，折腾一下午了，开始以为我多选了几个插件的原因

xiaobaishu1, 2021-01-11:
一般补丁打不上的基本都是上游已经改进了 你把它删了就行了

PR #6185 (2021-01-11): ipq40xx/5.4: remove duplicate patches
fix #6178
closed 2021-01-11 by coolsnowwolf