coolsnowwolf/lede #6181: 以容器运行时，作为旁路由网关无法转发路由

FUjr, 2021-01-10:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
在--privileged参数下运行的是用树莓派3b docker版配置文件编译的版本，通过macvlan对外做旁路由

现在遇到的问题是：
容器路由表正常的情况下，局域网设备可以访问网页，在容器的shell中ping外网也能通，但是当手动设置手机网关为容器ip时，手机却无法正常路由，局域网内tcpdump可抓到手机和容器发出的arp包
 
但是神奇的是，在容器中开启tcpdump时，居然一切正常了。
结束tcpdump进程后又恢复原来的样子

我是技术小白，我猜测是什么模块没开起来，而这个模块恰是tcpdump需要调用的，因此能够成功路由

### (2) 路由器型号和固件版本

A：
nanopi neo2（H5）刷的 armbian

宿主机系统：Debian GNU/Linux 10 (buster) with Linux 5.9.14-sunxi64

OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

用的原生网卡，网卡是RTL8211E-VB-CG

### (3) 详细日志

A：
tcpdump抓包能抓到双方arp广播

FUjr, 2021-01-17:
写了个脚本放到 /etc/rc.local开tcpdump，虽然治标不治本，但又不是不能用！！``

`#!/bin/bash
while true 
do
d=`date '+%F %T'`;
num=`ps  | grep 'tcpdump' | egrep -v 'grep|echo|rpm|moni|guard' | wc -l`;
if [ $num -lt 1 ];then

 echo "[$d] tcpdump is dead...restarting" >> /root/tcpdump/tcpdump.log ;
 echo "[$d] tcpdump is dead...restarting";
 nohup tcpdump host 192.168.4.1 ;
 sleep 5
fi
sleep 10
done
`
写好脚本，放在任意目录，再把脚本写入/etc/rc.local就好啦～
如果你网段是192.168.4.1，那我建议你还是小改一下，以免nohup.out体积爆炸

另外就是希望大佬给点解决思路，感谢！

LGA1150, 2021-01-29:
> 但是神奇的是，在容器中开启tcpdump时，居然一切正常了。
结束tcpdump进程后又恢复原来的样子

tcpdump 不加 `-p` 参数时会开启网卡混杂模式
你可以手动打开试试
`ip link set 网卡名称 promisc on`

FUjr, 2021-01-31:
> > 但是神奇的是，在容器中开启tcpdump时，居然一切正常了。
> 结束tcpdump进程后又恢复原来的样子
> 
> tcpdump 不加 `-p` 参数时会开启网卡混杂模式
> 你可以手动打开试试
> `ip link set 网卡名称 promisc on`

感谢帮助，不过我宿主机确实开了混杂模式,而且如果不开混杂模式的话，我应该是连ssh、web管理都上不去的，所以暂时排除这个引起的了。

[image]

privilege特权也是给了的。

1049066154, 2021-02-09:
你是不是在防火墙中加了什么代码？

FUjr, 2021-02-22:
> 你是不是在防火墙中加了什么代码？

好像都是默认的呀

fyxw, 2021-03-17:
> > > 但是神奇的是，在容器中开启tcpdump时，居然一切正常了。
> > > 结束tcpdump进程后又恢复原来的样子
> > 
> > 
> > tcpdump 不加 `-p` 参数时会开启网卡混杂模式
> > 你可以手动打开试试
> > `ip link set 网卡名称 promisc on`
> 
> 感谢帮助，不过我宿主机确实开了混杂模式,而且如果不开混杂模式的话，我应该是连ssh、web管理都上不去的，所以暂时排除这个引起的了。
> 
> [image]
> 
> privilege特权也是给了的。

从你发的截图来看，是没开混杂的，`Multicast`是多播，不是混杂，从我使用中发现，不开混杂时有几率能连上容器内的web和ssh，但是路由不稳定的。

 `ip link set 网卡名称 promisc on`要放在宿主机开机脚本里，不然重启就失效了。

建议不要给`privilege`特权，反而会导致内核不稳定，而且重启容器会导致宿主机也重启。用`--cap-add NET_ADMIN --cap-add NET_RAW`替代就可以了，需要更多权限就换成`--cap-add ALL`，需要加载的内核模块，直接在宿主机里加载，然后重启容器

closed 2021-06-20 by coolsnowwolf