coolsnowwolf/lede #6183: x86 固件网速跑不满问题请教

Guxing0321, 2021-01-11:
Hi Team,

我编译的x86固件无法跑满网速，500M只能跑300左右

[image]

编译的配置文件如下，我不确定是不是我哪个组件没加上，导致驱动问题什么的
[x86.txt](https://github.com/coolsnowwolf/lede/files/5793649/x86.txt)

BR
Robin

minlang112, 2021-01-11:
什么CPU 什么网卡，说下机器配置呢-。-

Guxing0321, 2021-01-11:
Hi minlang112,

D525的cpu，网卡应该是i211的
[image]

或者，有什么命令可以查看，可以提供下，我来操作

BR
Robin

qjiang888, 2021-01-11:
D525 只能300多M

NagaseKouichi, 2021-01-11:
D525 太老了吧，性能太弱了

star-o0O, 2021-01-11:
应该是CPU性能瓶颈了，开SFE可以千兆

sxx1314, 2021-01-11:
远古架构  高功低能   不开sfe 200出头就差不多了   开了sfe也不会改善太多

Guxing0321, 2021-01-11:
Hi 大佬们，

我这边准备上J900 或者 3865U，这两个应该是妥妥的吧

我的那个配置文件需要做修改吗？

BR
Robin

qjiang888, 2021-01-11:
i3 打底，其他都是垃圾

Guxing0321, 2021-01-11:
Hi  @sxx1314 @qjiang888 @NagaseKouichi @star-o0O  ，

我这边试过了，sfe打开最高就360Mb的样子。
我这边会来换上好的cpu，已经联系好了，3865U

谢谢大佬们的解答

BR
Robin

closed 2021-01-11 by Guxing0321