coolsnowwolf/lede #6184: R2S使用“系统”菜单中的“重启”或“定时重启”死机

alulux, 2021-01-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：使用系统菜单的中定时重启或重启后，前面板sys指示灯常亮，wan和lan指示灯不亮，网口侧lan口指示灯闪烁，wan口指示灯不亮，系统web管理无法登录，设备无法上网。系统断电重启后恢复正常

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志

A：
Mon Jan 11 03:16:18 2021 daemon.info dnsmasq[3137]: Connected to system UBus
Mon Jan 11 03:16:25 2021 daemon.info procd: - init complete -
Mon Jan 11 03:16:27 2021 daemon.info pppd[1588]: System time change detected.
Mon Jan 11 03:16:29 2021 cron.err crond[3477]: crond (busybox 1.31.1) started, log level 5
Mon Jan 11 03:16:29 2021 daemon.err uhttpd[1014]: sh: write error: Broken pipe
Mon Jan 11 03:16:39 2021 user.notice firewall: Reloading firewall due to ifup of wan6 (eth0)
Mon Jan 11 03:16:40 2021 cron.err crond[3708]: crond (busybox 1.31.1) started, log level 5
Mon Jan 11 03:16:40 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:16:40 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:16:43 2021 daemon.info dnsmasq[3819]: Connected to system UBus
Mon Jan 11 03:16:43 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:16:43 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:16:46 2021 daemon.info dnsmasq[3970]: Connected to system UBus
Mon Jan 11 03:16:57 2021 user.notice firewall: Reloading firewall due to ifup of wan (pppoe-wan)
Mon Jan 11 03:16:57 2021 cron.err crond[4244]: crond (busybox 1.31.1) started, log level 5
Mon Jan 11 03:16:57 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:16:57 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:17:00 2021 daemon.info dnsmasq[4345]: Connected to system UBus
Mon Jan 11 03:17:00 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:17:00 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:17:03 2021 daemon.info dnsmasq[4495]: Connected to system UBus
Mon Jan 11 03:17:14 2021 user.notice firewall: Reloading firewall due to ifup of wan_6 (pppoe-wan)
Mon Jan 11 03:17:14 2021 cron.err crond[4773]: crond (busybox 1.31.1) started, log level 5
Mon Jan 11 03:17:14 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:17:14 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:17:17 2021 daemon.info dnsmasq[4873]: Connected to system UBus
Mon Jan 11 03:17:17 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 03:17:17 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 03:17:21 2021 daemon.info dnsmasq[5028]: Connected to system UBus
Mon Jan 11 04:00:00 2021 cron.err crond[4773]: USER root pid 10561 cmd /etc/coremark.sh
Mon Jan 11 06:00:00 2021 cron.err crond[4773]: USER root pid 14317 cmd /usr/share/adbyby/adblock.sh > /tmp/adupdate.log 2>&1
Mon Jan 11 06:00:22 2021 cron.err crond[14407]: crond (busybox 1.31.1) started, log level 5
Mon Jan 11 06:00:22 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 06:00:22 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 06:00:25 2021 daemon.info dnsmasq[14502]: Connected to system UBus
Mon Jan 11 06:00:25 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Mon Jan 11 06:00:25 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Mon Jan 11 06:00:28 2021 daemon.info dnsmasq[14698]: Connected to system UBus

feheadls, 2021-01-11:
不只是R2S，R4S也是这样。软重启就无法重启成功，即使终端命令行reboot也不行，必须断电启动。

alulux, 2021-01-11:
> 不只是R2S，R4S也是这样。软重启就无法重启成功，即使终端命令行reboot也不行，必须断电启动。

我是想放进弱电箱使用的，这样的话重启会很麻烦，不知是否是系统适配问题，能否解决

feheadls, 2021-01-11:
我也没解决，只有等大佬修复

wushenghn, 2021-01-11:
bootrom有问题有时候重启启动的时候找不的系统进bootrom,连ttl reset就能正常启动，不连ttl就无限卡boot

xiaobaishu1, 2021-01-11:
表示可用30 4 * * * sleep 70 && touch /etc/banner && reboot ！

minlang112, 2021-01-12:
还是移除R4S这鬼东西的支持吧……

alulux, 2021-01-12:
> 表示可用30 4 * * * sleep 70 && touch /etc/banner && reboot ！

这个我可以回去试试

1715173329, 2021-01-13:
rk3328 / rk3399 通病
来自 UHS SD 卡问题，换卡即可

或者参考 Kwiboo/linux-rockchip@0d329112c709d6cfedf0fffb19f0cc6b19043f6b

alulux, 2021-01-14:
> rk3328 / rk3399 通病
> 来自 UHS SD 卡问题，换卡即可
> 
> 或者参考 [Kwiboo/linux-rockchip@0d32911](https://github.com/Kwiboo/linux-rockchip/commit/0d329112c709d6cfedf0fffb19f0cc6b19043f6b)

那应该用什么型号的卡？我现在是闪迪16GB U1 C10 A1的？

spypie, 2021-01-14:
> > rk3328 / rk3399 通病
> > 来自 UHS SD 卡问题，换卡即可
> > 或者参考 [Kwiboo/linux-rockchip@0d32911](https://github.com/Kwiboo/linux-rockchip/commit/0d329112c709d6cfedf0fffb19f0cc6b19043f6b)
> 
> 那应该用什么型号的卡？我现在是闪迪16GB U1 C10 A1的？

U2或者U3的，应该都行

alulux, 2021-01-14:
友善官方似乎并未推荐U2或U3的卡啊
[image]
https://www.yuque.com/friendlyelec/r2s/wr5gll
或许这个并不是官方发的

feheadls, 2021-01-14:
> > > rk3328 / rk3399 通病
> > > 来自 UHS SD 卡问题，换卡即可
> > > 或者参考 [Kwiboo/linux-rockchip@0d32911](https://github.com/Kwiboo/linux-rockchip/commit/0d329112c709d6cfedf0fffb19f0cc6b19043f6b)
> > 
> > 
> > 那应该用什么型号的卡？我现在是闪迪16GB U1 C10 A1的？
> 
> U2或者U3的，应该都行

我就是老版的633X创见 32G U3 MLC颗粒的卡，同样存在这个问题。

DanielChen008, 2021-01-16:
update:
根据那篇文档的建议，我昨天下单了京东朗科联名的Class10/SDHC/A1的卡，R4S 还是无法软重启的（主要我的2台 R2S 都没有重启问题，一直都用的 SanDisk 卡，很稳；R4S之前也用的同款，但无法重启）。

1715173329, 2021-01-16:
> https://www.yuque.com/friendlyelec/r2s/wr5gll
> 或许这个并不是官方发的

这份 wiki 是由 NanoPi R2S Club 成员共同打造，主要编订者是 @nicksun98，
被某些傻逼商家复制粘贴后当售后卖钱去了

1715173329, 2021-01-16:
> update:
> 根据那篇文档的建议，我昨天下单了京东朗科联名的Class10/SDHC/A1的卡，R4S 还是无法软重启的（主要我的2台 R2S 都没有重启问题，一直都用的 SanDisk 卡，很稳；R4S之前也用的同款，但无法重启）。

参考 Kwiboo/linux-rockchip@0d32911，
亦可试试 [t.me/nanopi_r2s](https://t.me/nanopi_r2s) 的固件.

OomelodyoO, 2021-03-05:
nanopi r4s
安装的是ext4版本
1.在公司用reboot是有效的，在家做主路由并拨号使用，发现执行reboot时，出现直接关机问题
2.在家openwrt管理后台 也经常不能打开了

shikkoku, 2021-06-19:
我今天用脚本编译R21.6.14 ，基于J5005处理器的Intel NUC也不能软重启(SSH reboot指令或者是luci界面点重启)，电源指示灯常亮，显示无网络，也进不了后台，只能按电源键关机然后再开机。不过尝试插HDMI线连接显示器进入控制台，接键盘输入reboot或者是poweroff测试却可以正常重启关机，很奇怪。

closed 2021-06-20 by coolsnowwolf

Calmact, 2022-02-20:
> 我今天用脚本编译R21.6.14 EFI版，基于J5005处理器的Intel NUC也不能软重启(SSH reboot指令或者是luci界面点重启)，电源指示灯常亮，显示无网络，也进不了后台，只能按电源键关机然后再开机。不过尝试插HDMI线连接显示器进入控制台，接键盘输入reboot或者是poweroff测试却可以正常重启关机，很奇怪。

我的J1900工控机也是这么个状况，reboot命令重启就死机。

server-yang, 2022-04-20:
可以自定义命令 reboot -f命令强制重启

siyang814, 2022-10-24:
解决了吗？ 我现在也是这问题， 导致人不在家不敢软重启