coolsnowwolf/lede #6193: 新3运行UU加速器失败

EnnawYang, 2021-01-13:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：新3运行UU加速器失败

### (2) 路由器型号和固件版本

A：Newifi-D2 新三

### (3) 详细日志

A：
```
root@Newifi-D2:~# cat /tmp/uugamebooster.log
/usr/share/uugamebooster/uuplugin: line 1: syntax error: unexpected "("
```
```
root@Newifi-D2:~# /usr/share/uugamebooster/uuplugin
/usr/share/uugamebooster/uuplugin: line 1: syntax error: unexpected "("
```

AmadeusGhost, 2021-01-13:
> modprobe: failed to find a module named tun

Maybe need kmod-tun?

EnnawYang, 2021-01-13:
> > modprobe: failed to find a module named tun
> 
> Maybe need kmod-tun?

是的 加上就可以了

EnnawYang, 2021-01-13:
不过似乎有冲突？
```
Wed Jan 13 16:31:53 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
Wed Jan 13 16:31:53 2021 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:53 2021 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
Wed Jan 13 16:31:55 2021 user.err ssr-rules[2395]: Can't redirect, please check the iptables.
Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: ssr-rules[2395]: Can't redirect, please check the iptables.
```

AmadeusGhost, 2021-01-13:
> 新3运行UU加速器失败，手动下载官网程序替换之后可以运行

是不是之前编译了其它架构的 UU 插件?

EnnawYang, 2021-01-13:
> > 新3运行UU加速器失败，手动下载官网程序替换之后可以运行
> 
> 是不是之前编译了其它架构的 UU 插件?

先编译了软路由的

EnnawYang, 2021-01-13:
> > 新3运行UU加速器失败，手动下载官网程序替换之后可以运行
> 
> 是不是之前编译了其它架构的 UU 插件?

所以是要先删除之前下载的包吗（

AmadeusGhost, 2021-01-13:
>先编译了软路由的

果然 hhh，这个 dl 名字有毒，不同架构的包名字都一样
> 所以是要先删除之前下载的包吗（

对，我之前让 @hyird 大佬帮忙备份了 dl 包并重命名了，正好改下下载地址

EnnawYang, 2021-01-13:
> > 先编译了软路由的
> 
> 果然 hhh，这个 dl 名字有毒，不同架构的包名字都一样
> 
> > 所以是要先删除之前下载的包吗（
> 
> 对，我之前让 @hyird 大佬帮忙备份了 dl 包并重命名了，正好改下下载地址

好吧，那上面的iptables错误大佬也帮忙看看吧（）

KFERMercer, 2021-01-13:
> > 先编译了软路由的
> 
> 果然 hhh，这个 dl 名字有毒，不同架构的包名字都一样
> 
> > 所以是要先删除之前下载的包吗（
> 
> 对，我之前让 @hyird 大佬帮忙备份了 dl 包并重命名了，正好改下下载地址

操, 疏忽了

EnnawYang, 2021-01-13:
> > > 先编译了软路由的
> > 
> > 
> > 果然 hhh，这个 dl 名字有毒，不同架构的包名字都一样
> > > 所以是要先删除之前下载的包吗（
> > 
> > 
> > 对，我之前让 @hyird 大佬帮忙备份了 dl 包并重命名了，正好改下下载地址
> 
> 操, 疏忽了

对了dalao，更新固件之后好像要解绑再重新绑定才能用么，这个有办法解决么？

KFERMercer, 2021-01-13:
> 对了dalao，更新固件之后好像要解绑再重新绑定才能用么，这个有办法解决么？

这得问网易了

NagaseKouichi, 2021-01-14:
> 不过似乎有冲突？
>
> ```
> Wed Jan 13 16:31:53 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
> Wed Jan 13 16:31:53 2021 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Another app is [...]

我这里好像没这问题

EnnawYang, 2021-01-14:
> > 不过似乎有冲突？
> > ```
> > Wed Jan 13 16:31:53 2021 user.notice firewall: Reloading firewall due to ifup of wan (eth0.2)
> > Wed Jan 13 16:31:53 2021 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:53 2021 daemon.notice procd: /etc/rc.d/S91wrtbwmon: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: Another app is currently holding the xtables lock. Perhaps you want to use the -w option?
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: iptables: No chain/target/match by that name.
> > Wed Jan 13 16:31:55 2021 user.err ssr-rules[2395]: Can't redirect, please check the iptables.
> > Wed Jan 13 16:31:55 2021 daemon.notice procd: /etc/rc.d/S95shadowsocksr: ssr-rules[2395]: Can't redirect, please check the iptables.
> > ```
> 
> 我这里好像没这问题，X86软路由

其实我不确定是不是UU加速器造成的 卸载了之后好像也报错

closed 2021-01-14 by coolsnowwolf