coolsnowwolf/lede #6194: execvp: /usr/bin/env: Argument list too long any fix ??

isslam95, 2021-01-13:
`lede/bin/packages/x86_64/luci/*.ipk /home/isso/lede/staging_dir/packages/x86; ln -s /home/isso/lede/bin/packages/x86_64/packages/*.ipk /home/isso/lede/staging_dir/packages/x86; ln -s /home/isso/lede/bin/packages/x86_64/routing/*.ipk /home/isso/lede/staging_dir/packages/x86; ln -s /home/isso/lede/bin/packages/x86_64/telephony/*.ipk /home/isso/lede/staging_dir/packages/x86;
find /home/isso/lede/staging_dir/target-x86_64_musl/root-x86 -type d | xargs -r chmod 0755
rm -rf /home/isso/lede/build_dir/target-x86_64_musl/root-x86 /home/isso/lede/build_dir/target-x86_64_musl/root.orig-x86
mkdir -p /home/isso/lede/build_dir/target-x86_64_musl/root-x86/tmp
IPKG_NO_SCRIPT=1 IPKG_INSTROOT=/home/isso/lede/build_dir/target-x86_64_musl/root-x86 TMPDIR=/home/isso/lede/build_dir/target-x86_64_musl/root-x86/tmp /home/isso/lede/staging_dir/host/bin/opkg --offline-root /home/isso/lede/build_dir/target-x86_64_musl/root-x86 --force-postinstall --add-dest root:/ --add-arch all:100 --add-arch x86_64:200 install \
	/home/isso/lede/bin/targets/x86/64/packages/base-files_224-r3003-622af09a7_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/grub2_2.04-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-antfs_5.4.85+2020-02-10-b41ba529-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-ksmbd_5.4.85+3.2.2-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-netatop_5.4.85+3.1-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-siit_5.4.85+1.2-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-compat-xtables_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-nathelper-rtsp_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-account_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-chaos_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-condition_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-delude_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-dhcpmac_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-dnetmap_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-fuzzy_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-geoip_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-iface_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-ipmark_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-ipp2p_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-ipv4options_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-length2_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-logmark_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-lscan_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-lua_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-proto_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-psd_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-quota2_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-sysrq_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-tarpit_5.4.85+3.13-3_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-ipt-rtpengine_5.4.85+mr8.5.2.7-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-cryptodev_5.4.85+1.10-x86-2_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-exfat_5.4.85+5.10.1-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-sched-cake-virtual_5.4.85+2020-07-24-48979385-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kernel_5.4.85-1-a6bedb988a71a1fa0a7c1006fae70be1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-dax_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-dm_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-scsi-core_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-acompress_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-aead_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-arc4_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-authenc_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-cbc_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-ccm_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-cmac_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-crc32_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-crc32c_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-ctr_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-cts_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-deflate_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-des_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-ecb_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-echainiv_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-gcm_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-gf128_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-ghash_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-hash_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-hmac_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-manager_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-md4_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-md5_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-null_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-pcompress_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-rng_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-seqiv_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-sha1_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-sha256_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-sha512_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-crypto-user_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-btrfs_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-cifs_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-exportfs_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-ext4_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-f2fs_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-nfs_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-nfs-common_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-nfs-common-rpcsec_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-nfs-v4_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-nfsd_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-fs-vfat_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-hwmon-core_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-i2c-core_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-i2c-algo-bit_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-hid_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-hid-generic_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-input-core_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-input-evdev_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-lib-crc-ccitt_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-lib-crc-itu-t_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-lib-crc7_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/kmod-lib-crc16_5.4.85-1_x86_64.ipk /home/isso/lede/bin/targets/x86/64/packages/ [... cut ...] sso/lede/bin/packages/x86_64/routing/luci-lib-dracula_6_x86_64.ipk
make[2]: execvp: /usr/bin/env: Argument list too long
make[2]: *** [package/Makefile:68: package/install] Error 127
make[2]: Leaving directory '/home/isso/lede'
make[1]: *** [package/Makefile:108: /home/isso/lede/staging_dir/target-x86_64_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/isso/lede'
make: *** [/home/isso/lede/include/toplevel.mk:224: world] Error 2`

i tried ulimit -n 65535 and ulimit -s 65535

programmejava, 2021-01-13:
典型linux参数太长错误，现在要么用减少编译的ipk，要么找到对应脚本，用xarg代替原来的。

hzjnet, 2021-01-30:
> 典型linux参数太长错误，现在要么用减少编译的ipk，要么找到对应脚本，用xarg代替原来的。

应该是正解
   我遇见有点奇怪：
rm -f /openwrt/openwrt-nss/lede-17.01-qca-nss-d210119-1/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/host/node-v8.17.0/out/Release/obj.target/deps/openssl/libopenssl.a ar-file-list; mkdir -p `dirname /openwrt/openwrt-nss/lede-17.01-qca-nss-d210119-1/build_dir/target-arm_cortex-a15+neon-vfpv4_musl-1.1.20_eabi/host/node-v8.17.0/out/Release/obj.target/deps/openssl/libopenssl.a`
make[5]: execvp: /bin/sh: Argument list too long
把编译的os由centos 7换成ubuntu后就正常了，其它啥都没变！

Dimitri2020007, 2021-04-14:
我在实体机编译最后的阶段经常会遇到Argument list too long的错误，不知道怎么解决。

只好使用github actions在线编译了。

closed 2021-06-20 by coolsnowwolf