coolsnowwolf/lede #6197: 想知道代码和官方代码的对应关系，网上搜了好几天了，没找到答案啊

teapethu, 2021-01-13:
想知道代码和官方代码的对应关系，网上搜了好几天了，没找到答案啊

这边的代码是怎么和官方同步的呢？同步到官方的master还是什么分支呢？
靠人工同步吗？官方的都会同步过来？还是挑选的呢？

谢谢。

EnnawYang, 2021-01-14:
看大佬们心情更新
[image]

ihuaer, 2021-01-14:
看大佬们心情更新
新的代码可能水土不服

teapethu, 2021-01-14:
好吧，我就想知道一个对应关系。或者说能看到更新到官方的某个commit了吗？

Aes64X, 2021-01-14:
> 好吧，我就想知道一个对应关系。或者说能看到更新到官方的某个commit了吗？

并不和主线同步，不要尝试拿任何主线补丁或更新加在这里使用，你会后悔的。
也不要尝试直接拿主线的设备支持加在这里使用，你会后悔的。
不过不想后悔，请重新生成补丁或修改设备描述的写法。
但。。。相信我，纠结这个你会后悔的。

teapethu, 2021-01-14:
> > 好吧，我就想知道一个对应关系。或者说能看到更新到官方的某个commit了吗？
> 
> 并不和主线同步，不要尝试拿任何主线补丁或更新加在这里使用，你会后悔的。
> 也不要尝试直接拿主线的设备支持加在这里使用，你会后悔的。
> 如果不想后悔，请重新生成补丁或修改设备描述的写法。
> 但。。。相信我，纠结这个你会后悔的。

那。。。这边已经支持的设备，编译出来可以用吧？
感觉这个大家多是编译X86吧，不知道对普通路由支持如何啊？谢谢

Ozzy-hzy, 2021-01-17:
> > > 好吧，我就想知道一个对应关系。或者说能看到更新到官方的某个commit了吗？
> > 
> > 
> > 并不和主线同步，不要尝试拿任何主线补丁或更新加在这里使用，你会后悔的。
> > 也不要尝试直接拿主线的设备支持加在这里使用，你会后悔的。
> > 如果不想后悔，请重新生成补丁或修改设备描述的写法。
> > 但。。。相信我，纠结这个你会后悔的。
> 
> 那。。。这边已经支持的设备，编译出来可以用吧？
> 感觉这个大家多是编译X86吧，不知道对普通路由支持如何啊？谢谢

我就一直编译的硬路由（新三，猴米2100，k2p），挺稳定的

coolsnowwolf, 2021-01-18:
大体上是官方的 Trunk + 稳定版的 软件仓库 + 自定义补丁和驱动 + 19.07 Pre 版本的魔改高速 luci ，什么也不对应，好用就行

teapethu, 2021-01-19:
> 大体上是官方的 Trunk + 稳定版的 软件仓库 + 自定义补丁和驱动 + 19.07 Pre 版本的魔改高速 luci ，什么也不对应，好用就行

谢谢大神。。。刚开始学习，版本有点儿懵，哈哈，总以为这边应该是对标官方的，然后再加上中国习惯app

closed 2021-01-19 by teapethu