coolsnowwolf/lede #6201: 源码出问题了好像

x444222, 2021-01-15:
反馈bug/问题模板，提建议请删除
./scripts/feeds install -a
WARNING: Makefile 'package/lean/
## 1.关于你要提交的问题
./scripts/feeds install -a
出错
Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
./scripts/feeds install -a

WARNING: Makefile 'package/lean/此目录下的全部错误警告
### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：

EnnawYang, 2021-01-15:
https://github.com/EnnawYang/Actions-OpenWrt/runs/1672753874?check_suite_focus=true#step:7:20

EnnawYang, 2021-01-15:
https://github.com/EnnawYang/Actions-OpenWrt/runs/1672753874?check_suite_focus=true#step:7:20
你指的是这个？ 正常无需理会

x444222, 2021-01-16:
https://github.com/x444222/openwrt/blob/main/20210116223249.png

截图出来了。新手我也不是很懂，前几天还没有这样的，四天前我还能编译成功现在就这样，重装系统重新来过都不行

CYJason2000, 2021-01-18:
Update的时候有没有超时或失败？可能是网络问题，开了代理再执行这两条比较好。

jjm2473, 2021-01-19:
试试 `./script/feeds update -a && ./script/feeds install-a`

x444222, 2021-01-21:
解决了重复使用  ./script/feeds install-a

closed 2021-06-20 by coolsnowwolf