coolsnowwolf/lede #6204: 新加入UU加速器怎么用? 丁真都能看懂的说明→

KFERMercer, 2021-01-15:
# 怎么用？

### 1. 首先确定你的固件里已经编译进了`uugamebooster`

### 2. 手机下载安装[这个应用](https://adl.netease.com/d/g/uu/c/uu_router)

### 3. 打开应用选择安装路由器插件

### 4. 无脑下一步直到显示绑定成功

### 5. 用就完了

---

# 各种各样的问题:

> ### Q: 为什么我在'luci→applications'里找不到UU加速器?
> 
> ### A: 插件路径: `network → uugamebooster`
&nbsp;
> ### Q: 为什么编译进去后在路由器后台→服务里找不到?
> 
> ### A: 不需要有. 编译安装后直接默认开启, 我也不推荐使用其他任何带有 luci 的UU加速器插件, 会出现问题
&nbsp;
> ### Q: 那我想关闭/重启/禁止开机自启加速器咋办?
> 
> ### A: 路由器后台→系统→启动项→找到`uugamebooster`→关闭
&nbsp;
> ### Q: 这个软件包是网易官方出的吗? 是不是破解版? 和用官方的安装脚本安装的有啥区别?
> ### A1: 这个软件包是根据网易官方的安装脚本打包的, 优化了运行机制, 推荐首选使用本库而不是官方脚本
> ### A2: 不是破解版, 强烈反对任何破解行为
> ### A3: 与使用官方的安装脚本安装的在结果上没有任何区别, 但是比官方脚本安装运行更稳定
&nbsp;
&nbsp;

## 声明:

1. 本插件没有任何对网易私有的bin文件的解包/修改行为, 也未使用任何网易私有代码, 相关文件全部从网易官方网站动态加载
2. 本人与网易公司无任何关系, 如有任何运行问题请联系网易客服

NagaseKouichi, 2021-01-16:
已经用上了，玩NS比用PC版的UU加速器爽多了，感谢大佬

MLST10019, 2021-01-16:
楼上的，用openclash，然后其他流量选择国内流量，switch的测速贼快。NAT A，你试试

MLST10019, 2021-01-16:
感谢大佬的编译！！！

NagaseKouichi, 2021-01-16:
今天出现openwrt上的网易UU无法联网的情况，APP上显示openwrt路由器无法连接互联网，但是网络其实是好的，其他设备上网没问题，不知道是不是我之前pppoe重新拨过号的原因。在启动项里面重启网易UU也不行。但是重启路由器就好了。

NagaseKouichi, 2021-01-16:
> 楼上的，用openclash，然后其他流量选择国内流量，switch的测速贼快。NAT A，你试试

没明白，用了UU加速，为什么还要用openclash去代理switch。我现在是直接用UU给switch加速，NAT A，测速下载能100多M，上传50M

ihuaer, 2021-01-17:
感谢大佬的编译！！！

MLST10019, 2021-01-17:
> > 楼上的，用openclash，然后其他流量选择国内流量，switch的测速贼快。NAT A，你试试
> 
> 没明白，用了UU加速，为什么还要用openclash去代理switch。我现在是直接用UU给switch加速，NAT A，测速下载能100多M，上传50M

随便找个包流量的套餐，不限时带clash订阅的。大概30不到那种。在clash管理里的”其他流量，选择，国内流量。这样switch就是走运营商的流量，速度也有30M以上，省钱实惠

EnnawYang, 2021-01-17:
> > 楼上的，用openclash，然后其他流量选择国内流量，switch的测速贼快。NAT A，你试试
> 
> 没明白，用了UU加速，为什么还要用openclash去代理switch。我现在是直接用UU给switch加速，NAT A，测速下载能100多M，上传50M

他的意思就是让我们不要用UU加速器，直接用代理

skydevil88, 2021-01-18:
IOS装那个应用？

mjyhj, 2021-01-19:
识别不到WIN和安卓 客户端，IOS没测

hwd1118, 2021-01-19:
> 识别不到WIN和安卓客户端，IOS没测

一样，安卓客户端显示

设备型号openwrt router
该路由型号暂不支持加速

KFERMercer, 2021-01-19:
> > 识别不到WIN和安卓客户端，IOS没测
> 
> 一样，安卓客户端显示
> 
> 设备型号openwrt router
> 该路由型号暂不支持加速

截图, 路由器型号?

hdydy, 2021-01-20:
> > 识别不到WIN和安卓客户端，IOS没测
> 
> 一样，安卓客户端显示
> 
> 设备型号openwrt router
> 该路由型号暂不支持加速

你这个是因为进程没有启动，要么是你程序没安装对，要么就是你绑定后又解绑了。如果是后者，重启就好了。另外显示OpenWrt说明进程运行，显示openwrt router则是进程没有启动。

njf1993, 2021-01-20:
安卓客户端，显示该路由器型号暂不支持加速。
luci：1.0-3和v2.7.2-5两个ipk已装好
但启动项中没有找到uugamebooster。
求大佬指教~

Myoko, 2021-01-21:
不玩游戏！这个东西有什么用？能YT？

EnnawYang, 2021-01-21:
> 不玩游戏！这个东西有什么用？能YT？

玩游戏用的，谢谢

zhouwei07618, 2021-01-21:
什么时候开发腾讯加速器

ZhxChen, 2021-01-23:
ios客户端  找不到符合的路由器。看到openwrt里uu的进程已经在跑

15108428326, 2021-01-26:
无法登陆路由怎么弄？？
[image]

zzzop, 2021-01-28:
上了迅游的 贼船羡慕UU

greennyreborn, 2021-01-28:
ps4 可以正常识别，但是switch 识别不到，界面上跳出了一下，然后就再也没了

NagaseKouichi, 2021-01-29:
想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网

greennyreborn, 2021-01-29:
> 想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网

我也在想同样的事情，我把 iptables, ip rule, ip route 都撸了一遍。
感觉应该是 ip rule 里面加上指定ip 走 163那个路由表就可以了。
但实际测试后，发现数据并没有走出去

NagaseKouichi, 2021-01-29:
> > 想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网
> 
> 我也在想同样的事情，我把 iptables, ip rule, ip route 都撸了一遍。
> 感觉应该是 ip rule 里面加上指定ip 走 163那个路由表就可以了。
> 但实际测试后，发现数据并没有走出去

会不会是网易UU的客户端限制了可以连的服务器，比如NS只能连任天堂的官方服务器

greennyreborn, 2021-01-29:
> > > 想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网
> > 
> > 
> > 我也在想同样的事情，我把 iptables, ip rule, ip route 都撸了一遍。
> > 感觉应该是 ip rule 里面加上指定ip 走 163那个路由表就可以了。
> > 但实际测试后，发现数据并没有走出去
> 
> 会不会是网易UU的客户端限制了可以连的服务器，比如NS只能连任天堂的官方服务器

那就不清楚了，我继续用回我的8lag了

Lrvineshaw, 2021-01-31:
这个需要先充值UU会员吗？

iPixelOldC, 2021-01-31:
> 想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网

想啥呢，别人都是白名单的。像QQ邮箱自带的Gmail代理也是，都是白名单的

koast18, 2021-01-31:
> > 想知道有没有办法把某个设备的流量伪装成NS或者PS4主机的流量，然后走网易UU的隧道进行加速或者科学上网
> 
> 想啥呢，别人都是白名单的。像QQ邮箱自带的Gmail代理也是，都是白名单的

之前没被滥用的时候基本就是一个透明代理，没有白名单限制，速度还快，后来嘛，就GG了。

lyx32, 2021-02-03:
我想问，我是R2S下面接wifi路由器。上面是我路由器不支持加速，是不是无解？

KFERMercer, 2021-02-03:
> 我想问，我是R2S下面接wifi路由器。上面是我路由器不支持加速，是不是无解？

路由器要桥接

lyx32, 2021-02-04:
> 路由器要桥接

我搜了一下路由器桥接，不是很明白。
桥接的目的是不是让手机和r2s在一个网段？

closed 2021-02-06 by KFERMercer

NiriK074, 2021-11-08:
> 楼上的，用openclash，然后其他流量选择国内流量，switch的测速贼快。NAT A，你试试

openclash的这个设置在哪儿啊找半天找不到

mariohide, 2021-11-15:
您好，请问 uu 加速器插件是不是仅支持 x86 架构的路由器？我的路由器是 ATH79 的，在 make config 中找不到 uugamebooster，切换成 x86 才有。

n-WN, 2021-12-21:
> # 怎么用？
> ### 1. 首先确定你的固件里已经编译进了`luci-app-uugamebooster`或`uugamebooster`
> ### 2. 手机下载安装这个应用 →[`鸿蒙OS/Android`](https://adl.netease.com/d/g/uu/c/uu_router) →[`iOS`](https://apps.apple.com/cn/app/%E7%BD%91%E6%98%93uu%E4%B8%BB%E6%9C%BA%E5%8A%A0%E9%80%9F/id1440256574)
> ### 3. 打开应用选择`安装路由器插件`
> ### 4. 无脑下一步直到显示绑定成功
> ### 5. 用就完了
> # 各种各样的问题:
>    
> 
> ### 更新: 大雕已经加入了`luci-app-uugamebooster`, 以下问题仅针对没有luci界面的用户:
>    
> 
> > ### Q: 为什么我在`make menuconfig`时在`luci → applications`里找不到UU加速器?
> > ### A: 插件路径: `network → uugamebooster`
> 
>  
> 
> > ### Q: 为什么编译进去后在路由器后台→服务里找不到?
> > ### A: 因为不需要有. 编译安装后直接默认开启, 我也不推荐使用其他任何带有 luci 的UU加速器插件, 会出现问题
> 
>  
> 
> > ### Q: 那我想关闭/重启/禁止开机自启加速器咋办?
> > ### A: 路由器后台→系统→启动项→找到`uugamebooster`→关闭
> 
>  
> 
> > ### Q: 这个软件包是网易官方出的吗? 是不是破解版? 和用官方的安装脚本安装的有啥区别?
> > ### A1: 此插件是根据网易官方的安装脚本打包的, 同时优化了官方的粪运行机制, 所以推荐首选使用本插件而不是官方脚本
> > ### A2: 不是破解版, 强烈反对任何破解行为
> > ### A3: 与使用官方的安装脚本安装在加速上没有任何区别, 但是由于优化了官方的粪运行机制, 所以比官方脚本安装运行更稳定
> 
>    
> 
> ## 声明:
> 1. 本插件没有任何对网易私有的bin文件的解包/修改行为, 也未使用任何网易私有代码, 相关文件全部从网易官方网站动态加载
> 2. 本人与网易公司无任何关系, 如有任何运行问题请联系网易客服
> 3. 鸿蒙OS™与AOSP为完全不同的两种事物, 请勿搞混

怎么才能删除官方脚本安装的那个呢

katayaburi, 2022-05-06:
> > 识别不到WIN和安卓客户端，IOS没测
> 
> 一样，安卓客户端显示
> 
> 设备型号openwrt router 该路由型号暂不支持加速

有可能是你的Lan口并不是桥接的，改成桥接就可以使用了。

tech-novice01, 2023-02-10:
openwrt的uu插件，可以正常识别xbox,但是无法识别到win 和 安卓可能是什么原因呢
[image]

zzzop, 2023-02-10:
> 

应该是版本问题，openwrt上的UU，包括从官方安装，我用过的都只能加速主机，买的小米路由器都可以加速全平台。

NV3S, 2023-04-11:
咱这个能和openclash共存并开吗？

mariohide, 2023-04-11:
> 咱这个能和openclash共存并开吗？

可以，我就这么用

NV3S, 2023-04-11:
> 

ok 我编译进去，之前都在用openclash带PS玩游戏，ping到50

1sm23, 2023-04-12:
> 咱这个能和openclash共存并开吗？

我是openclash，然后直接装uu的openwrt插件在手机app上就可以加速，不用lede

NV3S, 2023-04-12:
ok 用的官方脚本是吧发自我的 iPhone

1sm23, 2023-04-12:
是的

zzzop, 2023-07-22:
> 

OpenWrt官方版本mispel停留在3.9.2，也不知道什么时候能更新一下。有没有大佬测试过x86的版本是更新过的吗，可以加速pc吗

StayHunFoo, 2023-10-31:
> 可以

我这个openclash开着，然后uu加速器主机端一个都检测不到设备