coolsnowwolf/lede #6208: r2s总是不定时出现断网，不知道是网口软中断还是重启了

Benjamin23333, 2021-01-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：r2s总是不定时出现断网，不知道是网口软终端还是重启了。我试过很多，基于lean大佬的源码和project openwrt项目的
snapshot分支和18.06分支都有这个问题，而https://github.com/QiuSimons/R2S-OpenWrt这个基于官方openwrt源码的项目却没有这个问题。
      发生问题的时候会先断网，然后路由管理界面无法访问，过一会儿才能进入管理界面，进入之后概况里面可以看到ipv6连接变成问号。在netdata里面看到看到cpu刚刚从一个高负载状态回落，而且只能看回落的过程，前面的部分也看不到了。在系统日志中看到之前的系统日志消失，新的系统日志差不多是从断网那一刻开始。因为我这方面知识还非常浅薄，不知道这是软中断还是重启了，还是什么其他的问题。还有一点就是在断网的时间点之前我并没有进行高负载的工作。

### (2) 路由器型号和固件版本

A：nanopi r2s       OpenWrt R20.12.12 / LuCI Master (git-20.343.54716-6fc079f)

### (3) 详细日志

A：因为一整页日志我并不知道那些部分是有用的，直接贴出来可能对看问题造成不便，如果以上的描述还不足以提供详细的信息，我会把日志贴出来

hxlls, 2021-01-16:
这个之前我碰到过，我查看日志好久，发现好像是掉网卡了，目前找不到解决办法

binge8, 2021-01-17:
目测日志消失应该是重启了，高负载回落也是因为刚重启所以高，慢慢回落，可以参考这两个帖子https://github.com/coolsnowwolf/lede/issues/6055  https://github.com/coolsnowwolf/lede/issues/5349  关闭SFE 试试吧

Benjamin23333, 2021-01-17:
> 目测日志消失应该是重启了，高负载回落也是因为刚重启所以高，慢慢回落，可以参考这两个帖子#6055 #5349 关闭SFE 试试吧

我试一下

llzzll, 2021-01-17:
我也是昨天发现这个问题，BT下载个60多G的东西，每下十几G的时候会断一次，断网，后台也进不去，然后日志不是重启后重新写的那种，但也没有什么有用的内容没见着重启或者掉网卡，然后看了issue编译了个只有flowoffload的固件，目前还没出现问题

Benjamin23333, 2021-01-18:
> > 目测日志消失应该是重启了，高负载回落也是因为刚重启所以高，慢慢回落，可以参考这两个帖子#6055 #5349 关闭SFE 试试吧
> 
> 我试一下

不行啊，今天早上六点五十多又出现了这个问题，我看运行时间归零了，看来是重启了，可是sfe已经关了

aueu, 2021-01-19:
一定要贴日志的，否则wan还是lan口出问题都不知道，都是瞎猜

https://github.com/fanck0605/openwrt-nanopi-r2s/issues/10

如果你的内核日志跟这个一样，都是usb网卡出问题，可以尝试ssh进系统，运行 `ethtool 对应网卡 rx off tx off`

但是我记得上游后续的不知道哪个commit已经解决该问题了呀

Benjamin23333, 2021-01-19:
> 一定要贴日志的，否则wan还是lan口出问题都不知道，都是瞎猜
> 
> [fanck0605/openwrt-nanopi-r2s#10](https://github.com/fanck0605/openwrt-nanopi-r2s/issues/10)
> 
> 如果你的内核日志跟这个一样，都是usb网卡出问题，可以尝试ssh进系统，运行 `ethtool 对应网卡 rx off tx off`
> 
> 但是我记得上游后续的不知道哪个commit已经解决该问题了呀

Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:             [ table TABLE_ID ] [ vrf NAME ] [ proto RTPROTO ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:             [ type TYPE ] [ scope SCOPE ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: ROUTE := NODE_SPEC [ INFO_SPEC ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: NODE_SPEC := [ TYPE ] PREFIX [ tos TOS ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:              [ table TABLE_ID ] [ proto RTPROTO ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:              [ scope SCOPE ] [ metric METRIC ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:              [ ttl-propagate { enabled | disabled } ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: INFO_SPEC := { NH | nhid ID } OPTIONS FLAGS [ nexthop NH ]...
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: NH := [ encap ENCAPTYPE ENCAPHDR ] [ via [ FAMILY ] ADDRESS ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: 	    [ dev STRING ] [ weight NUMBER ] NHFLAGS
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: FAMILY := [ inet | inet6 | mpls | bridge | link ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: OPTIONS := FLAGS [ mtu NUMBER ] [ advmss NUMBER ] [ as [ to ] ADDRESS ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ rtt TIME ] [ rttvar TIME ] [ reordering NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ window NUMBER ] [ cwnd NUMBER ] [ initcwnd NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ ssthresh NUMBER ] [ realms REALM ] [ src ADDRESS ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ rto_min TIME ] [ hoplimit NUMBER ] [ initrwnd NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ features FEATURES ] [ quickack BOOL ] [ congctl NAME ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:            [ pref PREF ] [ expires TIME ] [ fastopen_no_cookie BOOL ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: TYPE := { unicast | local | broadcast | multicast | throw |
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools:           unreachable | prohibit | blackhole | nat }
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: TABLE_ID := [ local | main | default | all | NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: SCOPE := [ host | link | global | NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: NHFLAGS := [ onlink | pervasive ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: RTPROTO := [ kernel | boot | static | NUMBER ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: PREF := [ low | medium | high ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: TIME := NUMBER[s|ms]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: BOOL := [1|0]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: FEATURES := ecn
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: ENCAPTYPE := [ mpls | ip | ip6 | seg6 | seg6local ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: ENCAPHDR := [ MPLSLABEL | SEG6HDR ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: SEG6HDR := [ mode SEGMODE ] segs ADDR1,ADDRi,ADDRn [hmac HMACKEYID] [cleanup]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: SEGMODE := [ encap | inline ]
Sat Jan 16 20:40:56 2021 daemon.notice procd: /etc/rc.d/S98udptools: ROUTE_GET_FLAGS := [ fibmatch ]
Sat Jan 16 20:40:57 2021 daemon.info aria2: Instance "main" disabled.
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER':No such file or directory
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: Try `iptables -h' or 'iptables --help' for more information.
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables: No chain/target/match by that name.
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman: iptables v1.8.4 (legacy): Couldn't load target `DOCKER-USER':No such file or directory
Sat Jan 16 20:40:57 2021 daemon.notice procd: /etc/rc.d/S99dockerman:
[... cut ...]
Sat Jan 16 20:44:21 2021 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Sat Jan 16 20:44:21 2021 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Sat Jan 16 20:44:21 2021 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Sat Jan 16 20:44:36 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!

这是我的日志，不知道哪些有用。和他不一样

aueu, 2021-01-19:
只有netdata与docker错误，netdata的日志错误没有影响
没装过docker，不清楚docker是否会造成防火墙崩溃

Benjamin23333, 2021-01-19:
> 只有netdata与docker错误，netdata的日志错误没有影响
> 没装过docker，不清楚docker是否会造成防火墙崩溃

那我试一下吧docker关掉

Benjamin23333, 2021-01-19:
> 只有netdata与docker错误，netdata的日志错误没有影响
> 没装过docker，不清楚docker是否会造成防火墙崩溃

docker关掉后还是出现了一样的问题，这次的日志是这样的
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: Bug report to http://code.google.com/p/mentohust/issues/list
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust:
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: !! 打开libnotify失败，请检查是否已安装该库文件。
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[1]:	br-lan
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[2]:	eth1
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[3]:	eth2
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[6]:	eth0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[7]:	sit0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[8]:	dummy0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[9]:	gre0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[10]:	gretap0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[11]:	erspan0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ** 网卡[12]:	bond0
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ?? 请选择网卡[1-12]: ** 您选择了第[12]块网卡。
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: ?? 请输入用户名: ?? 请输入密码: ?? 请选择组播地址(0标准 1锐捷私有 2赛尔): ?? 请选择DHCP方式(0不使用 1二次认证 2认证后 3认证前): !! 没有MentoHUST正在运行！
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mentohust: >> 认证已退出。
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90mjpg-streamer: /etc/rc.d/S90mjpg-streamer: device '/dev/video0' does not exist
Tue Jan 19 12:54:15 2021 daemon.info avahi-daemon[3468]: Server startup complete. Host name is OpenWrt.local. Local service cookie is 780168042.
Tue Jan 19 12:54:15 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:15 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 192.168.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:15 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to [fd7e:b535:7a1c::1]:80 failed (98: Address in use)
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90sfe: uci: Entry not found
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90sfe: uci: Entry not found
Tue Jan 19 12:54:15 2021 daemon.err rmmod: module is not loaded
Tue Jan 19 12:54:15 2021 daemon.notice procd: /etc/rc.d/S90sfe: net.ipv4.tcp_congestion_control = cubic
Tue Jan 19 12:54:15 2021 user.notice dnsmasq: DNS rebinding protection is active, will discard upstream RFC1918 responses!
Tue Jan 19 12:54:16 2021 user.notice dnsmasq: Allowing 127.0.0.0/8 responses
Tue Jan 19 12:54:16 2021 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: started, v1.31.1
Tue Jan 19 12:54:16 2021 daemon.notice procd: /etc/rc.d/S90sfe: udhcpc: sending discover
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 192.168.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to [fd7e:b535:7a1c::1]:80 failed (98: Address in use)
Tue Jan 19 12:54:16 2021 daemon.notice netifd: Network device 'eth0' link is up
Tue Jan 19 12:54:16 2021 daemon.notice netifd: Interface 'wan' has link connectivity
Tue Jan 19 12:54:16 2021 daemon.notice netifd: Interface 'wan' is setting up now
Tue Jan 19 12:54:16 2021 kern.info kernel: [   23.005556] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Tue Jan 19 12:54:16 2021 kern.info kernel: [   23.006393] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
Tue Jan 19 12:54:16 2021 daemon.notice netifd: Interface 'wan6' has link connectivity
Tue Jan 19 12:54:16 2021 daemon.notice netifd: Interface 'wan6' is setting up now
Tue Jan 19 12:54:16 2021 daemon.err insmod: module is already loaded - slhc
Tue Jan 19 12:54:16 2021 daemon.err odhcp6c[3778]: Failed to send RS (Address not available)
Tue Jan 19 12:54:16 2021 daemon.err insmod: module is already loaded - ppp_generic
Tue Jan 19 12:54:16 2021 daemon.err insmod: module is already loaded - pppox
Tue Jan 19 12:54:16 2021 daemon.err insmod: module is already loaded - pppoe
Tue Jan 19 12:54:16 2021 daemon.info pppd[3794]: Plugin rp-pppoe.so loaded.
Tue Jan 19 12:54:16 2021 daemon.info pppd[3794]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Tue Jan 19 12:54:16 2021 daemon.err odhcp6c[3778]: Failed to send DHCPV6 message to ff02::1:2 (Address not available)
Tue Jan 19 12:54:16 2021 daemon.notice pppd[3794]: pppd 2.4.8 started by root, uid 0
Tue Jan 19 12:54:16 2021 daemon.info dnsmasq[3826]: Connected to system UBus
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 127.0.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to 192.168.0.1:80 failed (98: Address in use)
Tue Jan 19 12:54:16 2021 daemon.err nginx[3527]: 2021/01/19 12:54:15 [emerg] 3527#0: bind() to [fd7e:b535:7a1c::1]:80 failed (98: Address in use)
[... cut ...]
Tue Jan 19 13:17:49 2021 user.notice nft-qos-monitor: ACTION=add, MACADDR=00:0a:f5:40:dc:84, IPADDR=192.168.0.165, HOSTNAME=RedmiNote4X-RedmiNot
Tue Jan 19 13:17:49 2021 user.notice nft-qos-dynamic: ACTION=add, MACADDR=00:0a:f5:40:dc:84, IPADDR=192.168.0.165, HOSTNAME=RedmiNote4X-RedmiNot
Tue Jan 19 13:23:14 2021 user.info mwan3track[6455]: Check (ping) failed for target "114.114.115.115" on interface wan (pppoe-wan)
Tue Jan 19 13:23:19 2021 user.info mwan3track[6455]: Lost 1 ping(s) on interface wan (pppoe-wan)
Tue Jan 19 13:30:47 2021 daemon.info watchcat[5671]: no internet connectivity for 2160 seconds. Reseting when reaching 21600

Benjamin23333, 2021-01-20:
我把我日志的关于pppoe的部分百度了一下，也有其他人遇到了，说是mwan3的问题，有没有大佬来帮我看下是不是😂

AmadeusGhost, 2021-01-21:
不用 mwan3 的话直接进软件包卸掉就好了，这个包的默认策略容易导致断网。

DanielChen008, 2021-01-21:
分享一个经历，也许是玄学，也许是某种潜在原因，仅供参考。前不久给一个哥们配置的 R2S ，他放进弱电箱，过去一个月频繁的说断网，然后登录界面也打不开，我是帮他各种折腾，最后问题依旧。他问我会不会是卡的问题，毕竟系统运行在卡上，我让他死马当活马医，重新找了张16G TF卡，现在反馈弄完再没断网过了。

Benjamin23333, 2021-01-21:
> 分享一个经历，也许是玄学，也许是某种潜在原因，仅供参考。前不久给一个哥们配置的 R2S ，他放进弱电箱，过去一个月频繁的说断网，然后登录界面也打不开，我是帮他各种折腾，最后问题依旧。他问我会不会是卡的问题，毕竟系统运行在卡上，我让他死马当活马医，重新找了张16G TF卡，现在反馈弄完再没断网过了。

啊这。。。。我这卡才不到一年，不能坏的那么快吧😂

Benjamin23333, 2021-01-21:
> 分享一个经历，也许是玄学，也许是某种潜在原因，仅供参考。前不久给一个哥们配置的 R2S ，他放进弱电箱，过去一个月频繁的说断网，然后登录界面也打不开，我是帮他各种折腾，最后问题依旧。他问我会不会是卡的问题，毕竟系统运行在卡上，我让他死马当活马医，重新找了张16G TF卡，现在反馈弄完再没断网过了。

[image]
我把卡拔下来检测了下，也没坏块啊😂

quintus-lab, 2021-01-22:
Cannot open file '/etc/netdata/apps_groups.conf'
这说明读卡已经有问题……
有没有108-rfc-mmc-core-set-initial-signal-voltage-on-power-off.patch啊？

Benjamin23333, 2021-01-23:
我用过 https://github.com/IvanSolis1989/OpenWrt-DIY 和https://github.com/SuLingGG/OpenWrt-Rpi 这个，最开始以为是太高大全，有插件冲突了，又换了https://github.com/Gabrielxzx/Actions-OpenWrt 这个精简一点的，都有那个问题，以我的知识还看不出来有没有108-rfc-mmc-core-set-initial-signal-voltage-on-power-off.patch，这个好像都是直接拉lean大的源码？

1715173329, 2021-01-23:
> Cannot open file '/etc/netdata/apps_groups.conf'

这个纯属垃圾 netdata 问题

y9858, 2021-02-09:
同样的问题 你解决了吗

Benjamin23333, 2021-02-09:
> 同样的问题 你解决了吗

没有，所以我现在用的https://github.com/QiuSimons/R2S-R4S-X86-OpenWrt 这个项目里的固件

huah0235, 2021-02-13:
有可能和mwan有关系

y9858, 2021-02-14:
[image]
[image]
我把SFE里面全关了 固件本身没有mwan

Benjamin23333, 2021-02-14:
> [image]
> [image]
> 我把SFE里面全关了 固件本身没有mwan

可以把固件的链接分享一下吗？

y9858, 2021-02-15:
> > [image]
> > [image]
> > 我把SFE里面全关了 固件本身没有mwan
> 
> 可以把固件的链接分享一下吗？

你自己云编译吧 别选sfe和mwan3就行了

Benjamin23333, 2021-02-24:
我终于搞清楚原因了，感谢Nicholas大佬的指点，是因为watchcat，我发现我用过的会发生这个问题的固件都有watchcat，这个插件的默认策略是ping 8.8.8.8主机，ping不通累计到一定次数就会重启，因为我有时候会把ssr+之类关掉或者设置的默认dns为8.8.4.4，就会出现ping不通的情况，然后累计到一定次数就重启了

closed 2021-02-24 by Benjamin23333

binge8, 2021-02-24:
> 我终于搞清楚原因了，感谢Nicholas大佬的指点，是因为watchcat，我发现我用过的会发生这个问题的固件都有watchcat，这个插件的默认策略是ping 8.8.8.8主机，ping不通累计到一定次数就会重启，因为我有时候会把ssr+之类关掉或者设置的默认dns为8.8.4.4，就会出现ping不通的情况，然后累计到一定次数就重启了。
> 解决办法有几种，要不把watchcat整个app扬了，或者吧ping的主机改成国内dns或者其他能ping通的，或者在luci界面里吧watchcat的配置删掉。但是以上的方法执行后一定要去系统进程里看看有没有watchcat的进程，把watchcat的进程删掉，不然他还是会继续去ping8.8.8.8，或者执行以上方法后手动重启路由器

很明显不是你说的问题。建议你把SFE关了，开着watchcat试试吧

Benjamin23333, 2021-02-24:
> 很明显不是你说的问题。建议你把SFE关了，开着watchcat试试吧

真的是这样，，我把系统日志设置成写入tf卡了，然后再里面看到了这个
[image]

binge8, 2021-02-24:
> > > 我终于搞清楚原因了，感谢Nicholas大佬的指点，是因为watchcat，我发现我用过的会发生这个问题的固件都有watchcat，这个插件的默认策略是ping 8.8.8.8主机，ping不通累计到一定次数就会重启，因为我有时候会把ssr+之类关掉或者设置的默认dns为8.8.4.4，就会出现ping不通的情况，然后累计到一定次数就重启了。
> > > 解决办法有几种，要不把watchcat整个app扬了，或者吧ping的主机改成国内dns或者其他能ping通的，或者在luci界面里吧watchcat的配置删掉。但是以上的方法执行后一定要去系统进程里看看有没有watchcat的进程，把watchcat的进程删掉，不然他还是会继续去ping8.8.8.8，或者执行以上方法后手动重启路由器
> > 
> > 
> > 很明显不是你说的问题。建议你把SFE关了，开着watchcat试试吧
> 
> 真的是这样，，我把系统日志设置成写入tf卡了，然后再里面看到了这个
> [image]
> 在ping 8.8.8.8不通达到21600次之后，shutdown了😂

第一：我都没有安装这个，一样重启，关了SFE才OK。第二：还用得着TF卡吗？把日志改到其他路径重启一样可以保存

Benjamin23333, 2021-02-24:
> 第一：我都没有安装这个，一样重启，关了SFE才OK。第二：还用得着TF卡吗？把日志改到其他路径重启一样可以保存

首先回答你的第一，我把这个关了确实不重启了（到现在为止），至于你的问题为什么是那种解决办法，可能你的问题和我的不一样。
接下来会打你的第二，r2s的系统就是在tf卡上啊，我是把日志文件改成输出到/mnt/目录下面，本质上不就是写在tf卡上吗😂，可能是我的表述不够清晰，让你产生了误解