coolsnowwolf/lede #6210: k2p硬改32m 1月16日版本无法软重启

ltactor, 2021-01-16:
硬改32mk2p版，1月16日最新版全新glong编译，无法软起动。软起直接亮红灯，断电后可以启动。

使用AmadeusGhost@7658518 版可以软重启。

最新版没有32m选项，我看源码好像是自动识别，所以选择k2p进行编译。

sxx1314, 2021-01-17:
看了一下 commits   这几个版本关于mtk的改动也就是 取消了超频  你可以自己改回 [#6190](https://github.com/coolsnowwolf/lede/pull/6190) 之前一个版本  再测试看看

AmadeusGhost, 2021-01-17:
32M 的支持去掉了，需要的自行加上。

ltactor, 2021-01-18:
> 32M 的支持去掉了，需要的自行加上。

请问怎样加上对32M的支持？ 麻烦了。

我尝试使用原来的方法，完全git clong 编译两次报同样的错误。

原方法：
sed -i '/spi-max-frequency/a\\t\tbroken-flash-reset;' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/<0xa0000 0xf60000>/<0xa0000 0x1fb0000>/g' target/linux/ramips/dts/mt7621_phicomm_k2p.dts
sed -i 's/15744k/32448k/g' target/linux/ramips/image/mt7621.mk

报错：
rm -f /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_netgear_wndr3700-v5.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/arch/mips/boot/dts -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/arch/mips/boot/dts/include -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/include/ -undef -D__DTS__  -o /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_netis_wf2881.dtb.tmp ../dts/mt7621_netis_wf2881.dts
/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_netis_wf2881.dtb /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_netis_wf2881.dtb.tmp
rm -f /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_netis_wf2881.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/arch/mips/boot/dts -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/arch/mips/boot/dts/include -I/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/include/ -undef -D__DTS__  -o /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_phicomm_k2p.dtb.tmp ../dts/mt7621_phicomm_k2p.dts
/home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/linux-5.4.86/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_phicomm_k2p.dtb /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_phicomm_k2p.dtb.tmp
../dts/mt7621_phicomm_k2p.dts:60.3-22: ERROR (duplicate_property_names): /palmbus@1E000000/spi@b00/flash@0:broken-flash-reset: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:202: /home/song/k2ptest3/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_phicomm_k2p.dtb] Error 2
make[5]: Leaving directory '/home/song/k2ptest3/target/linux/ramips/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/song/k2ptest3/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/song/k2ptest3/target/linux'
time: target/linux/install#1001.98#101.04#1108.94
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/song/k2ptest3'
make[1]: *** [target/Makefile:19: /home/song/k2ptest3/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/song/k2ptest3'
make: *** [/home/song/k2ptest3/include/toplevel.mk:224：world] 错误 2

AmadeusGhost, 2021-01-18:
> 请问怎样加上对32M的支持？ 麻烦了。

打个 patch 就好：
```
diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
index fa02521ba6..f403b6d7e5 100644
--- a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
+++ b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
@@ -57,6 +57,7 @@
 		compatible = "jedec,spi-nor";
 		reg = <0>;
 		spi-max-frequency = <10000000>;
+		broken-flash-reset;
 
 		partitions {
 			compatible = "fixed-partitions";
@@ -90,7 +91,7 @@
 			partition@a0000 {
 				compatible = "denx,uimage";
 				label = "firmware";
-				reg = <0xa0000 0xf60000>;
+				reg = <0xa0000 0x1f60000>;
 			};
 		};
 	};
diff --git a/target/linux/ramips/image/mt7621.mk b/target/linux/ramips/image/mt7621.mk
index 8c3ef04e35..ce7a8ee6f4 100644
--- a/target/linux/ramips/image/mt7621.mk
+++ b/target/linux/ramips/image/mt7621.mk
@@ -821,7 +821,7 @@ endef
 TARGET_DEVICES += netis_wf2881
 
 define Device/phicomm_k2p
-  IMAGE_SIZE := 15744k
+  IMAGE_SIZE := 32128k
   DEVICE_VENDOR := Phicomm
   DEVICE_MODEL := K2P
   SUPPORTED_DEVICES += k2p
```

lphgor, 2021-01-18:
经过我这几天的测试，我觉得k2p的无法重启问题并不是加入32M版本引起的，而是更新内核造成的。
目前我这边的测试结果是：5.4.83、5.4.84、5.4.86内核有问题，不能软重启，5.4.85是正常的，5.4.83以前的也都是正常的。
我用今天1月18日最新的代码，编译出的固件又不能软重启了。然后单独revert掉5.4.86的内核更新(a8ddd988b1d5c93b85fce780fa05c7e1755c2e13 + 28f854eb5f7931dba5828503cfc8e285f1672eaf)，回滚到5.4.85内核，其他代码保持不变，编译出来的就没有问题了。
```sh
git revert -n a8ddd988b1d5c93b85fce780fa05c7e1755c2e13
git revert -n 28f854eb5f7931dba5828503cfc8e285f1672eaf
```
我用的GitHub Actions云编译，所以可以保证每次代码都是全新的。

coolsnowwolf, 2021-01-18:
@AmadeusGhost 有测试过这个软重启问题么

AmadeusGhost, 2021-01-18:
> 经过我这几天的测试，我觉得k2p的无法重启问题并不是加入32M版本引起的，而是更新内核造成的。

u1s1，我个人觉得跟更新内核没啥关系，之前遇到过很多次别人说的内核问题都是由其他 commit 引起的
> 有测试过这个软重启问题么

按照他们的测试之前的改法能正常使用，主要是 16M 有问题（后面终于发现 16M 出现的问题是由默认超频引起的，也有部分其它机型一样的情况）（那 32M 版本还要加回来嘛

1715173329, 2021-01-18:
ramips: drop support for phicomm k2p 
以上将会成为解决该问题的最佳方案

SilverHunterLib, 2021-01-18:
> > 经过我这几天的测试，我觉得k2p的无法重启问题并不是加入32M版本引起的，而是更新内核造成的。
> 
> u1s1，我个人觉得跟更新内核没啥关系，之前遇到过很多次别人说的内核问题都是由其他 commit 引起的
> 
> > 有测试过这个软重启问题么
> 
> 按照他们的测试之前的改法能正常使用，主要是 16M 有问题（后面终于发现 16M 出现的问题是由默认超频引起的，也有部分其它机型一样的情况）（那 32M 版本还要加回来嘛

大佬我来反馈一下，我16号编译的固件，软重启没问题，内核版本5.4.86，型号K2P  16M

lphgor, 2021-01-18:
没办法只能拆机ttll，又刷了5.4.86，大佬们看看是什么原因重启失败
```log
[  112.574888] reboot: Restarting system

Boot and Recovery Environment for Embedded Devices
Copyright (C) 2020 HackPascal <hackpascal@gmail.com>
Build date 2020-10-09 [git-676bfd4]
Version 1.1 (r1286)

DRAM: 128MB
Platform: MediaTek MT7621A ver 1, eco 3
Board: Phicomm K2P
Clocks: CPU: 880MHz, DDR: 1200MHz, Bus: 220MHz, Ref: 40MHz
Flash: GigaDevice GD25Q128 (16MB) on mt7621-spi.0
rt2880-eth: MAC address from EEPROM is invalid, using default settings.
rt2880-eth: Using MAC address 00:0c:43:00:00:01
eth0: MediaTek MT7530 Gigabit switch

Network started on eth0, inet addr 192.168.1.1, netmask 255.255.255.0

Press any key to interrupt autoboot ... 0   

Trying to boot firmware from 0x000a0000 in flash bank 0 ...
Reading data into memory ...
U-Boot firmware image header detected.
    Image Name:   MIPS OpenWrt Linux-5.4.86
    Data Size:    2394918 Bytes
    Load Address: 80001000
    Entry Point:  80001000
Uncompressing data (LZMA) ... done.
Flushing cache ... done.

Starting kernel at 0x80001000...

[    0.000000] Linux version 5.4.86 (runner@fv-az168-836) (gcc version 8.4.0 (OpenWrt GCC 8.4.0 r3023-3fedff661)) #0 SMP Mon Jan 18 07:53:45 2021
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:3
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is Phicomm K2P
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000]   HighMem  [mem 0x0000000010000000-0x0000000023ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   node   0: [mem 0x0000000020000000-0x0000000023ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000023ffffff]
[    0.000000] percpu: Embedded 14 pages/cpu s27184 r8192 d21968 u57344
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 130496
[    0.000000] Kernel command line: console=ttyS0,57600 rootfstype=squashfs,jffs2
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=0003e080
[    0.000000] Readback ErrCtl register=0003e080
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 509864K/524288K available (5723K kernel code, 252K rwdata, 1236K rodata, 1288K init, 238K bss, 14424K reserved, 0K cma-reserved, 262144K highmem)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] random: get_random_bytes called from start_kernel+0x340/0x550 with crng_init=0
[    0.000000] CPU Clock: 880MHz
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000000] clocksource: MIPS: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 4343773742 ns
[    0.000009] sched_clock: 32 bits at 440MHz, resolution 2ns, wraps every 4880645118ns
[... 88 lines cut ...]
[    2.726168] Hi    : 00000000
[    2.731910] Lo    : 77359400
[    2.737692] epc   : ffff00fe 0xffff00fe
[    2.745336] ra    : ffff00fe 0xffff00fe
[    2.752977] Status: 11008402	KERNEL EXL 
[    2.760807] Cause : 50800010 (ExcCode 04)
[    2.768795] BadVA : ffff00fe
[    2.774537] PrId  : 0001992f (MIPS 1004Kc)
[    2.782697] Modules linked in:
[    2.788803] Process swapper/0 (pid: 0, threadinfo=(ptrval), task=(ptrval), tls=00000000)
[    2.804915] Stack : 11007c00 1000001e 00000000 00000249 00000000 00000402 806e70b0 00000040
[    2.821574]         11007c02 00000510 00000020 00000000 8fc0ded8 fffffffb bfbf8000 00000000
[    2.838231]         00000000 8059614c 8000001f 00000000 8fc64000 8fc0de90 00000000 8001e638
[    2.854889]         11007c02 00000000 0015c000 00000510 40800408 80019164 00000008 00000000
[    2.871549]         80850000 87ffb000 87feb000 800a90b0 00000002 00000000 11007c02 00000510
[    2.888208]         ...
[    2.893117] Call Trace:
[    2.893125] 
[    2.901099] [<8059614c>] r4k_wait_irqoff+0x0/0x24
[    2.910488] [<8001e638>] tlb_do_page_fault_0+0x118/0x120
[    2.921162] [<80019164>] do_page_fault+0x54/0x4b8
[    2.930683] [<800a90b0>] tick_periodic+0xbc/0xe0
[    2.939883] [<8001e638>] tlb_do_page_fault_0+0x118/0x120
[    2.950489] [<803dc71c>] gic_compare_interrupt+0x7c/0x9c
```

ltactor, 2021-01-18:
> 打个 patch 就好：
>
> ```
> diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
> index fa02521ba6..f403b6d7e5 100644
> --- [...]

感谢回复，今早已经成功编译完成。
今晚可以刷机测试。

感谢。

ltactor, 2021-01-19:
> 打个 patch 就好：
>
> ```
> diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
> index fa02521ba6..f403b6d7e5 100644
> --- [...]

感谢回复。

k2p 硬改32M   标准配置，+iphelp +wireguard+ helloword
使用这个patch

刷机后重启没有问题。
重启后恢复配置文件重启也没有问题。

感谢修复。

jjm2473, 2021-01-19:
请问一下，为何删除32M K2P的支持，应该不会对其他设备造成影响啊

ltactor, 2021-01-19:
> 请问一下，为何删除32M K2P的支持，应该不会对其他设备造成影响啊

同问。
k2p  32M 修复后，很好用。  稳定运行。
希望还是增加对32M的支持。

jjm2473, 2021-01-19:
> > 请问一下，为何删除32M K2P的支持，应该不会对其他设备造成影响啊
> 
> 同问。
> k2p 32M 修复后，很好用。 稳定运行。
> 希望还是增加对32M的支持。

我在一个多月前增加的32M支持，因为我自己就只有一台32M版本的K2P，本来今天打算更新代码编译的，发现32M的支持被移除了。看提交记录应该是这段时间引入了什么bug，然后修复不了，于是就删了。

@AmadeusGhost 我看你的提交有个修复32M版重启问题的，这个问题是不是已经解决了，或者跟32M版本其实没什么关系，如果不是32M版的代码引起的问题，希望把这个设备加回来。

AmadeusGhost, 2021-01-22:
@lphgor 麻烦用最新代码再试一次看看

ltactor, 2021-01-22:
@AmadeusGhost 我可以测试32M，希望能把32M选项加回来。

AmadeusGhost, 2021-01-22:
> 我可以测试32M，希望能把32M选项加回来。

https://github.com/AmadeusGhost/lede/commit/df6db9110c490e29a1e83cf80a9627a916fd9a96

ltactor, 2021-01-22:
> > 我可以测试32M，希望能把32M选项加回来。
> 
> [AmadeusGhost@df6db91](https://github.com/AmadeusGhost/lede/commit/df6db9110c490e29a1e83cf80a9627a916fd9a96)

多谢，今晚编译，明天刷机测试。

lphgor, 2021-01-22:
> @lphgor 麻烦用最新代码再试一次看看

5.4.91内核，试了几次，重启都是正常的

ltactor, 2021-01-23:
> > 我可以测试32M，希望能把32M选项加回来。
> 
> [AmadeusGhost@df6db91](https://github.com/AmadeusGhost/lede/commit/df6db9110c490e29a1e83cf80a9627a916fd9a96)

今早已经刷机。
k2p 硬改32 默认配置+iphelp+wiregruad+helloworld+ssmpt+curl
恢复配置文件自动重启一次，手动重启一次，断电重启一次都正常启动。
最近版本wifi信号超好，感觉提高50%。

感谢修复。

closed 2021-01-23 by coolsnowwolf

nobk, 2021-01-23:
@coolsnowwolf 我有两台k2p 32MB 闪存的，今天的32M补丁试过了用-F强刷，不能用，又按我以前的代码进行了修改，能用了，附上进一步的补丁。请注意闪存布局从原厂布局0xa0000改为了公版布局0x50000。

```
diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p-32m.dts b/target/linux/ramips/dts/mt7621_phicomm_k2p-32m.dts
index e828b069a..0969077f6 100644
--- a/target/linux/ramips/dts/mt7621_phicomm_k2p-32m.dts
+++ b/target/linux/ramips/dts/mt7621_phicomm_k2p-32m.dts
@@ -8,9 +8,6 @@
 };

 &firmware {
-       reg = <0xa0000 0x1f60000>;
+       reg = <0x50000 0x1fb0000>;
 };

-&flash {
-       broken-flash-reset;
-};
diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
index 1a973a2a8..836efc6e1 100644
--- a/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
+++ b/target/linux/ramips/dts/mt7621_phicomm_k2p.dts
@@ -8,5 +8,5 @@
 };

 &firmware {
-       reg = <0xa0000 0xf60000>;
+       reg = <0x50000 0xfb0000>;
 };
diff --git a/target/linux/ramips/dts/mt7621_phicomm_k2p.dtsi b/target/linux/ramips/dts/mt7621_phicomm_k2p.dtsi
index 904580505..2158cb19e 100644
--- a/target/linux/ramips/dts/mt7621_phicomm_k2p.dtsi
+++ b/target/linux/ramips/dts/mt7621_phicomm_k2p.dtsi
@@ -49,9 +49,10 @@
        status = "okay";

        flash: flash@0 {
-               compatible = "jedec,spi-nor";
+               compatible = "mx25l25635f", "jedec,spi-nor";
                reg = <0>;
-               spi-max-frequency = <10000000>;
+               spi-max-frequency = <25000000>;
+               m25p,fast-read;

                partitions {
                        compatible = "fixed-partitions";
@@ -76,13 +77,7 @@
                                read-only;
                        };

-                       partition@50000 {
-                               label = "permanent_config";
-                               reg = <0x50000 0x50000>;
-                               read-only;
-                       };
-
-                       firmware: partition@a0000 {
+                       firmware: partition@50000 {
                                compatible = "denx,uimage";
                                label = "firmware";
                        };
diff --git a/target/linux/ramips/image/mt7621.mk b/target/linux/ramips/image/mt7621.mk
index b043d14bd..580252451 100644
--- a/target/linux/ramips/image/mt7621.mk
+++ b/target/linux/ramips/image/mt7621.mk
@@ -830,7 +830,7 @@ endef
 TARGET_DEVICES += phicomm_k2p

 define Device/phicomm_k2p-32m
-  IMAGE_SIZE := 32128k
+  IMAGE_SIZE := 32448k
   DEVICE_VENDOR := Phicomm
   DEVICE_MODEL := K2P
   DEVICE_VARIANT := 32M

```

nobk, 2021-01-23:
这样可能会对16M原厂闪存的版本有影响，我手里没16M的k2p了，最好还是把整个flash段从dtsi分离到k2p.dts和k2p-32m.dts里面。

AmadeusGhost, 2021-01-23:
u1s1，这是你们硬改用户自己的破事。本身不同 spi flash 在 dts 的写法就不一样，鬼知道你们上的是哪个型号的，也不可能有办法兼容每个人的情况。另外说一句，改布局自己想怎么改都行，不要来破坏大家约定俗成的习惯。
接下来直接移除掉 32M 版本的支持好了，我 TM 再支持类似的硬改型号我就是狗。再让这个库出现 32M 版本，
> ramips: drop support for phicomm k2p
> 以上将会成为解决该问题的最佳方案

这就是终极解决方案。

jjm2473, 2021-01-23:
别啊，为啥改布局，就几百k的空间

nobk, 2021-01-23:
0x50000才是约定俗成的习惯，Openwrt rampis下面上百个设备都是这个布局，0xa0000是个异端。

```
$ grep 50000 * | wc -l
772
$ grep 50000 *
mt7620a_alfa-network_ac1200rm.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_alfa-network_ac1200rm.dts:                      partition@50000 {
mt7620a_alfa-network_ac1200rm.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_alfa-network_r36m-e4g.dts:              spi-max-frequency = <50000000>;
mt7620a_alfa-network_r36m-e4g.dts:                      partition@50000 {
mt7620a_alfa-network_r36m-e4g.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_alfa-network_tube-e4g.dts:              spi-max-frequency = <50000000>;
mt7620a_alfa-network_tube-e4g.dts:                      partition@50000 {
mt7620a_alfa-network_tube-e4g.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_asus_rp-n53.dts:                        partition@50000 {
mt7620a_asus_rp-n53.dts:                                reg = <0x50000 0x7b0000>;
mt7620a_asus_rt-ac54u.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_asus_rt-ac5x.dtsi:              spi-max-frequency = <50000000>;
mt7620a_asus_rt-ac5x.dtsi:                      partition@50000 {
mt7620a_asus_rt-ac5x.dtsi:                              reg = <0x50000 0xfb0000>;
mt7620a_bdcom_wap2100-sk.dts:                   partition@50000 {
mt7620a_bdcom_wap2100-sk.dts:                           reg = <0x50000 0xf70000>;
mt7620a_bdcom_wap2100-sk.dts:           ieee80211-freq-limit = <5000000 6000000>;
mt7620a_buffalo_whr-1166d.dts:                  partition@50000 {
mt7620a_buffalo_whr-1166d.dts:                          reg = <0x50000 0xfb0000>;
mt7620a_buffalo_whr-1166d.dts:          ieee80211-freq-limit = <5000000 6000000>;
mt7620a_buffalo_whr-300hp2.dts:                 partition@50000 {
mt7620a_buffalo_whr-300hp2.dts:                         reg = <0x50000 0x7b0000>;
mt7620a_buffalo_whr-600d.dts:                   partition@50000 {
mt7620a_buffalo_whr-600d.dts:                           reg = <0x50000 0x7b0000>;
mt7620a_cameo_810.dtsi:         spi-max-frequency = <50000000>;
mt7620a_cameo_810.dtsi:                 factory5g: partition@50000 {
mt7620a_cameo_810.dtsi:                         reg = <0x50000 0x10000>;
mt7620a_cameo_810.dtsi:         ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dlink_dch-m225.dts:                     partition@50000 {
mt7620a_dlink_dch-m225.dts:                             reg = <0x50000 0x100000>;
mt7620a_dlink_dch-m225.dts:                     partition@150000 {
mt7620a_dlink_dch-m225.dts:                             reg = <0x150000 0x6b0000>;
mt7620a_dlink_dir-510l.dts:             spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-118-a2.dts:           spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-118-a2.dts:           ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dlink_dwr-960.dts:              spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-960.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dovado_tiny-ac.dts:                     partition@50000 {
mt7620a_dovado_tiny-ac.dts:                             reg = <0x50000 0x7b0000>;
mt7620a_dovado_tiny-ac.dts:             ieee80211-freq-limit = <5000000 6000000>;
mt7620a_edimax_br-6478ac-v2.dts:                        partition@50000 {
mt7620a_edimax_br-6478ac-v2.dts:                                reg = <0x50000 0x20000>;
mt7620a_edimax_ew-7478apc.dts:                  partition@50000 {
mt7620a_edimax_ew-7478apc.dts:                          reg = <0x50000 0x20000>;
mt7620a_edimax_ew-747x.dtsi:                    partition@50000 {
mt7620a_edimax_ew-747x.dtsi:                            reg = <0x50000 0x20000>;
mt7620a_engenius_esr600.dts:            spi-max-frequency = <50000000>;
mt7620a_engenius_esr600.dts:                    iNIC_rf: partition@50000 {
mt7620a_engenius_esr600.dts:                            reg = <0x50000 0x10000>;
mt7620a_engenius_esr600.dts:                            reg = <0xfb0000 0x50000>;
mt7620a_fon_fon2601.dts:                        partition@50000 {
mt7620a_fon_fon2601.dts:                                reg = <0x50000 0xf90000>;
mt7620a_fon_fon2601.dts:                ieee80211-freq-limit = <5000000 6000000>;
mt7620a_glinet_gl-mt300a.dts:                   partition@50000 {
mt7620a_glinet_gl-mt300a.dts:                           reg = <0x50000 0xf80000>;
mt7620a_glinet_gl-mt300n.dts:                   partition@50000 {
mt7620a_glinet_gl-mt300n.dts:                           reg = <0x50000 0xf80000>;
mt7620a_glinet_gl-mt750.dts:                    partition@50000 {
mt7620a_glinet_gl-mt750.dts:                            reg = <0x50000 0xf80000>;
mt7620a_head-weblink_hdrm200.dts:                       firmware: partition@50000 {
mt7620a_head-weblink_hdrm200.dts:                               reg = <0x50000 0xfb0000>;
mt7620a_head-weblink_hdrm200.dts:               ieee80211-freq-limit = <5000000 6000000>;
mt7620a_hiwifi_hc5761.dts:              ieee80211-freq-limit = <5000000 6000000>;
[... 685 lines cut ...]
rt5350_poray_m4-8m.dts:                         reg = <0x50000 0x7b0000>;
rt5350_poray_x5.dts:                    partition@50000 {
rt5350_poray_x5.dts:                            reg = <0x50000 0x7b0000>;
rt5350_poray_x8.dts:                    partition@50000 {
rt5350_poray_x8.dts:                            reg = <0x50000 0x7b0000>;
rt5350_tenda_3g150b.dts:                        partition@50000 {
[... cut ...]
rt5350_wansview_ncs601w.dts:                            reg = <0x50000 0x7b0000>;
rt5350_wiznet_wizfi630a.dts:                    partition@50000 {
rt5350_wiznet_wizfi630a.dts:                            reg = <0x50000 0xfb0000>;
rt5350_zorlik_zl5900v2.dts:                     partition@50000 {
rt5350_zorlik_zl5900v2.dts:                             reg = <0x50000 0x7b0000>;
rt5350_zyxel_keenetic-start.dts:                        partition@50000 {
rt5350_zyxel_keenetic-start.dts:                                reg = <0x50000 0x3b0000>;

```

nobk, 2021-01-23:
@AmadeusGhost  @jjm2473  0x50000才是约定俗成的习惯，Openwrt rampis下面上百个设备都是这个布局，0xa0000是个异端。

```
$ grep 50000 * | wc -l
772
$ grep 50000 *
mt7620a_alfa-network_ac1200rm.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_alfa-network_ac1200rm.dts:                      partition@50000 {
mt7620a_alfa-network_ac1200rm.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_alfa-network_r36m-e4g.dts:              spi-max-frequency = <50000000>;
mt7620a_alfa-network_r36m-e4g.dts:                      partition@50000 {
mt7620a_alfa-network_r36m-e4g.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_alfa-network_tube-e4g.dts:              spi-max-frequency = <50000000>;
mt7620a_alfa-network_tube-e4g.dts:                      partition@50000 {
mt7620a_alfa-network_tube-e4g.dts:                              reg = <0x50000 0xfb0000>;
mt7620a_asus_rp-n53.dts:                        partition@50000 {
mt7620a_asus_rp-n53.dts:                                reg = <0x50000 0x7b0000>;
mt7620a_asus_rt-ac54u.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_asus_rt-ac5x.dtsi:              spi-max-frequency = <50000000>;
mt7620a_asus_rt-ac5x.dtsi:                      partition@50000 {
mt7620a_asus_rt-ac5x.dtsi:                              reg = <0x50000 0xfb0000>;
mt7620a_bdcom_wap2100-sk.dts:                   partition@50000 {
mt7620a_bdcom_wap2100-sk.dts:                           reg = <0x50000 0xf70000>;
mt7620a_bdcom_wap2100-sk.dts:           ieee80211-freq-limit = <5000000 6000000>;
mt7620a_buffalo_whr-1166d.dts:                  partition@50000 {
mt7620a_buffalo_whr-1166d.dts:                          reg = <0x50000 0xfb0000>;
mt7620a_buffalo_whr-1166d.dts:          ieee80211-freq-limit = <5000000 6000000>;
mt7620a_buffalo_whr-300hp2.dts:                 partition@50000 {
mt7620a_buffalo_whr-300hp2.dts:                         reg = <0x50000 0x7b0000>;
mt7620a_buffalo_whr-600d.dts:                   partition@50000 {
mt7620a_buffalo_whr-600d.dts:                           reg = <0x50000 0x7b0000>;
mt7620a_cameo_810.dtsi:         spi-max-frequency = <50000000>;
mt7620a_cameo_810.dtsi:                 factory5g: partition@50000 {
mt7620a_cameo_810.dtsi:                         reg = <0x50000 0x10000>;
mt7620a_cameo_810.dtsi:         ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dlink_dch-m225.dts:                     partition@50000 {
mt7620a_dlink_dch-m225.dts:                             reg = <0x50000 0x100000>;
mt7620a_dlink_dch-m225.dts:                     partition@150000 {
mt7620a_dlink_dch-m225.dts:                             reg = <0x150000 0x6b0000>;
mt7620a_dlink_dir-510l.dts:             spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-118-a2.dts:           spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-118-a2.dts:           ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dlink_dwr-960.dts:              spi-max-frequency = <50000000>;
mt7620a_dlink_dwr-960.dts:              ieee80211-freq-limit = <5000000 6000000>;
mt7620a_dovado_tiny-ac.dts:                     partition@50000 {
mt7620a_dovado_tiny-ac.dts:                             reg = <0x50000 0x7b0000>;
mt7620a_dovado_tiny-ac.dts:             ieee80211-freq-limit = <5000000 6000000>;
mt7620a_edimax_br-6478ac-v2.dts:                        partition@50000 {
mt7620a_edimax_br-6478ac-v2.dts:                                reg = <0x50000 0x20000>;
mt7620a_edimax_ew-7478apc.dts:                  partition@50000 {
mt7620a_edimax_ew-7478apc.dts:                          reg = <0x50000 0x20000>;
mt7620a_edimax_ew-747x.dtsi:                    partition@50000 {
mt7620a_edimax_ew-747x.dtsi:                            reg = <0x50000 0x20000>;
mt7620a_engenius_esr600.dts:            spi-max-frequency = <50000000>;
mt7620a_engenius_esr600.dts:                    iNIC_rf: partition@50000 {
mt7620a_engenius_esr600.dts:                            reg = <0x50000 0x10000>;
mt7620a_engenius_esr600.dts:                            reg = <0xfb0000 0x50000>;
mt7620a_fon_fon2601.dts:                        partition@50000 {
mt7620a_fon_fon2601.dts:                                reg = <0x50000 0xf90000>;
mt7620a_fon_fon2601.dts:                ieee80211-freq-limit = <5000000 6000000>;
mt7620a_glinet_gl-mt300a.dts:                   partition@50000 {
mt7620a_glinet_gl-mt300a.dts:                           reg = <0x50000 0xf80000>;
mt7620a_glinet_gl-mt300n.dts:                   partition@50000 {
mt7620a_glinet_gl-mt300n.dts:                           reg = <0x50000 0xf80000>;
mt7620a_glinet_gl-mt750.dts:                    partition@50000 {
mt7620a_glinet_gl-mt750.dts:                            reg = <0x50000 0xf80000>;
mt7620a_head-weblink_hdrm200.dts:                       firmware: partition@50000 {
mt7620a_head-weblink_hdrm200.dts:                               reg = <0x50000 0xfb0000>;
mt7620a_head-weblink_hdrm200.dts:               ieee80211-freq-limit = <5000000 6000000>;
mt7620a_hiwifi_hc5761.dts:              ieee80211-freq-limit = <5000000 6000000>;
[... 685 lines cut ...]
rt5350_poray_m4-8m.dts:                         reg = <0x50000 0x7b0000>;
rt5350_poray_x5.dts:                    partition@50000 {
rt5350_poray_x5.dts:                            reg = <0x50000 0x7b0000>;
rt5350_poray_x8.dts:                    partition@50000 {
rt5350_poray_x8.dts:                            reg = <0x50000 0x7b0000>;
rt5350_tenda_3g150b.dts:                        partition@50000 {
[... cut ...]
rt5350_wansview_ncs601w.dts:                            reg = <0x50000 0x7b0000>;
rt5350_wiznet_wizfi630a.dts:                    partition@50000 {
rt5350_wiznet_wizfi630a.dts:                            reg = <0x50000 0xfb0000>;
rt5350_zorlik_zl5900v2.dts:                     partition@50000 {
rt5350_zorlik_zl5900v2.dts:                             reg = <0x50000 0x7b0000>;
rt5350_zyxel_keenetic-start.dts:                        partition@50000 {
rt5350_zyxel_keenetic-start.dts:                                reg = <0x50000 0x3b0000>;

```

AmadeusGhost, 2021-01-23:
@nobk 你理解错了，openwrt 都是用的原厂布局，异端的是 k2p 本身（原厂分区）。

1715173329, 2021-01-23:
@nobk 其实很好办，K2P 本身就是个异端，直接砍了就不会出现打破规矩。

nobk, 2021-01-23:
出厂k2p我吹下flash读过了，它的0x50000 "permanent_config"分区出厂时是空的，所以刷OpenWRT去掉这个分区完全没影响，留着又用不到，完全是浪费空间。

nobk, 2021-01-23:
@1715173329 你咋不把自己砍掉，我两台k2p用得好好的。

jjm2473, 2021-01-23:
可以参考下上游openwrt的k2p的dts布局，然后32M的应该以16M为准。
https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_phicomm_k2p.dts

nobk, 2021-01-23:
只能说当年提交mt7621_phicomm_k2p.dts这个commit的人没过脑子。你意思是以没过脑子的人的行为为准？

jjm2473, 2021-01-23:
如果你不以上游为准，那只能自己维护了。跟上游的差距越大，维护成本越高。

1715173329, 2021-01-23:
你确实挺有脑子的，smart ass

nobk, 2021-01-23:
我自己改的32M FLASH，自己换的512MB RAM，当然我自己一直在维护，一个bash脚本几行sed命令的事，没啥成本。

AmadeusGhost, 2021-01-23:
> 只能说当年提交mt7621_phicomm_k2p.dts这个commit的人没过脑子。你意思是以没过脑子的人的行为为准？

@nobk 你他妈的太过分了，你以为你是谁啊？有种你去像官方提交 pr 啊，就说这个分区是空的，看他们收不收。

AmadeusGhost, 2021-01-23:
官方的规定就是要按原厂的分区来写，更别说向官方提交 commit 要耗费大量的时间和精力，连最基本的尊重都没有还好意思叫，真他妈的丢人。

jjm2473, 2021-01-23:
我说的维护是指维护这个仓库中对应设备的代码，不是特定的某几台设备。
除非自己愿意以后负责维护这部分代码，不然还是尽量别给维护者添麻烦。

nobk, 2021-01-23:
@AmadeusGhost 我只关心32MB闪存的，原厂16MB爱怎么弄是原厂的事，32MB闪存的不是原厂，你还在扯原厂怎么着，是不是有病？

AmadeusGhost, 2021-01-23:
> 只能说当年提交mt7621_phicomm_k2p.dts这个commit的人没过脑子。你意思是以没过脑子的人的行为为准？

按您的说法确实应该以您的行为为准，毕竟没脑子的是谁显而易见。

AmadeusGhost, 2021-01-23:
> 我只关心32MB闪存的，原厂16MB爱怎么弄是原厂的事，32MB闪存的不是原厂，你还在扯原厂怎么着，是不是有病？

32M 闪存那是你自己的破事，去骂贡献者才是真的是有病。

nobk, 2021-01-23:
> > 只能说当年提交mt7621_phicomm_k2p.dts这个commit的人没过脑子。你意思是以没过脑子的人的行为为准？
> 
> 按您的说法确实应该以您的行为为准，毕竟没脑子的是谁显而易见。

我怀疑你有轻度阅读障碍，我写的原文是**没过脑子**，就是没仔细思考的意思，你直接给改成**没脑子**，建议看眼科。

1715173329, 2021-01-23:
> @AmadeusGhost 我只关心32MB闪存的，原厂16MB爱怎么弄是原厂的事，32MB闪存的不是原厂，你还在扯原厂怎么着，是不是有病？

我看你确实是有病
一个电子垃圾还硬改，你以为扩容 32M 跑梯子就能上千兆
垃圾就是垃圾，成天拿着电子垃圾瞎鸡巴乱搞的人那就叫傻逼
原厂确实不关你的事，毕竟你自己造的 K2B，斐讯团队直呼内行

你不仅需要看眼科，还需要看精神科
大可不必玩你那点文字游戏

nobk, 2021-01-23:
@1715173329 那你还进这贴干毛，这贴就专门讲32M，爱滚哪滚哪去。

AmadeusGhost, 2021-01-23:
> 我怀疑你有轻度阅读障碍，我写的原文是**没过脑子**，就是没仔细思考的意思，你直接给改成**没脑子**，建议看眼科。

确实，我就一废物。但是这跟你直接开骂贡献者有个 j8 关系？真该回去好好学习做人。

AmadeusGhost, 2021-01-23:
> @1715173329 你咋不把自己砍掉，我两台k2p用得好好的。

我和 1715173329 大佬在这个库加起来有两百多万行的代码改动，有种你 TM 就别用，别让我这个废物的代码污染了您的设备。

nobk, 2021-01-23:
@AmadeusGhost 有种你别提交代码了，提交了我就要用，气不死你。

AmadeusGhost, 2021-01-23:
> 有种你别提交代码了，提交了我就要用，气不死你。

恰恰相反，我提交代码就是为了让没教养的人恶心，你不嫌恶心那你继续，记得多积点得。

nobk, 2021-01-23:
@AmadeusGhost 要知道提交的代码并不都是宝，还有垃圾。

1715173329, 2021-01-23:
> @AmadeusGhost 要知道提交的代码并不都是宝，还有垃圾。

我看你这个屁贡献没有的傻逼哪来的底气叫嚣？

1715173329, 2021-01-23:
https://github.com/nobk/openwrt-tarlz/blob/master/Makefile
写个 Makefile 都能写得这么烂的智障，还敢在这里大放厥词

AmadeusGhost, 2021-01-23:
> 要知道提交的代码并不都是宝，还有垃圾。

确实，我说过了我就一垃圾，代码也是，可是它有两百多万行呢，您有多少呢就敢这么开骂？

AmadeusGhost, 2021-01-23:
> 可以参考下上游openwrt的k2p的dts布局，然后32M的应该以16M为准。
> https://github.com/openwrt/openwrt/blob/master/target/linux/ramips/dts/mt7621_phicomm_k2p.dts

事实上硬要支持 32M 版本就是这样，只能以大部分人的情况为准。

AmadeusGhost, 2021-01-23:
另外说一句，个人不建议用 mx25l25635f，这个 flash 直到 5.4 内核才算支持好了，而且读写速度是真的很慢。。

nobk, 2021-01-23:
我这片闪存是八个管脚的MX25L25645G。

1715173329, 2021-01-23:
> 事实上硬要支持 32M 版本就是这样，只能以大部分人的情况为准。

没必要，支持了的下场只是永无止境的骚扰。
不支持才是正确的做法。

1715173329, 2021-01-23:
人生最魔幻的事情之一无非不过被傻逼骂傻逼。

AmadeusGhost, 2021-01-23:
> 今天的32M补丁试过了用-F强刷，不能用，又按我以前的代码进行了修改，能用了，附上进一步的补丁。

说到底还是你的情况特殊，根本就不可能取少舍多去兼容。。
另外 32M spi flash 在 5.4 内核上都是要 `broken-flash-reset` 的，不然有概率重启或者保存配置失败。

nobk, 2021-01-23:
> > 今天的32M补丁试过了用-F强刷，不能用，又按我以前的代码进行了修改，能用了，附上进一步的补丁。
> 
> 说到底还是你的情况特殊，根本就不可能取少舍多去兼容。。
> 另外 32M spi flash 在 5.4 内核上都是要 `broken-flash-reset` 的，不然有概率重启或者保存配置失败。

这个真没注意，搜了一下dts，个别设备加上了，一会我加上之后编译看看效果。

```
grep "m25p,fast-read" *
mt7620a_alfa-network_r36m-e4g.dts:              m25p,fast-read;
mt7620a_alfa-network_tube-e4g.dts:              m25p,fast-read;
mt7620a_asus_rt-ac5x.dtsi:              m25p,fast-read;
mt7620a_dlink_dch-m225.dts:             m25p,fast-read;
mt7620a_engenius_esr600.dts:            m25p,fast-read;
mt7620a_youku_yk1.dts:          m25p,fast-read;
mt7621_alfa-network_quad-e4g.dts:               m25p,fast-read;
mt7621_gehua_ghl-r-001.dts:             m25p,fast-read;
mt7621_glinet_gl-mt1300.dts:            m25p,fast-read;
mt7621_gnubee_gb-pc1.dts:               m25p,fast-read;
mt7621_gnubee_gb-pc2.dts:               m25p,fast-read;
mt7621_iptime_a6ns-m.dts:               m25p,fast-read;
mt7621_iptime_a8004t.dts:               m25p,fast-read;
mt7621_jcg_jhr-ac876m.dts:              m25p,fast-read;
mt7621_jcg_y2.dts:              m25p,fast-read;
mt7621_phicomm_k2p.dtsi:                m25p,fast-read;
mt7621_wevo_w2914ns-v2.dtsi:            m25p,fast-read;
mt7628an_duzun_dm06.dts:                m25p,fast-read;
mt7628an_hiwifi_hc5x61a.dtsi:           m25p,fast-read;
mt7628an_netgear_r6080.dts:             m25p,fast-read;
rt3883_sitecom_wlr-6000.dts:            m25p,fast-read;
rt5350_dlink_dwr-512-b.dts:             m25p,fast-read;

grep broken-flash-reset *
mt7621_d-team_newifi-d2.dts:            broken-flash-reset;
mt7621_d-team_pbr-m1.dts:               broken-flash-reset;
mt7621_gehua_ghl-r-001.dts:             broken-flash-reset;
mt7621_gnubee_gb-pc1.dts:               broken-flash-reset;
mt7621_gnubee_gb-pc2.dts:               broken-flash-reset;
mt7621_lenovo_newifi-d1.dts:            broken-flash-reset;
mt7621_xzwifi_creativebox-v1.dts:               broken-flash-reset;

```

PR #6235 (2021-01-23): ramips: re-added phicomm k2p 32m version
Fix: #6210